openwrt/openwrt #14049: hostapd: excessive log file verbosity, including password details

guidosarducci, 2023-11-23:
### Describe the bug

During startup, `hostapd` will log sensitive debug-level information, including WiFi passwords. This is observed at the default log level, even on a fresh install with configuration wiped. This behaviour wasn't seen on previous releases, and occurs on multiple targets (ath79, ipq40xx).

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ath79/generic

### Device

D-LINK DIR-835 A1

### Image kind

Official downloaded image

### Steps to reproduce

Boot a router with wifi enabled and check boot logs.

### Actual behaviour

During `hostapd` startup, it logs low-level debug information such as passwords. For example:
```
Mon Oct  9 14:46:23 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40] channel=149  interface=phy1-ap0 bssid=<REDACTED> ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=0 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_group_rekey=3600 wpa_passphrase=<REDACTED> wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=<REDACTED> bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=1 wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256 okc=0 disable_pmksa_caching=1 ieee80211w=1 group_mgmt_cipher=AES-128-CMAC dynamic_vlan=0 vlan_naming=1 vlan_no
```

### Expected behaviour

These details should only be captured if debug-level logging is enabled, and not by default.

### Additional info

More detailed redacted log file for startup context:
```
# logread
Mon Oct  9 14:45:48 2023 kern.notice kernel: [    0.000000] Linux version 5.15.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23497-6637af95aa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Mon Oct 9 21:45:35 2023
Mon Oct  9 14:45:48 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Oct  9 14:45:48 2023 kern.info kernel: [    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
Mon Oct  9 14:45:48 2023 kern.info kernel: [    0.000000] MIPS: machine is D-LINK DIR-835 A1
Mon Oct  9 14:45:48 2023 kern.info kernel: [    0.000000] SoC: Atheros AR9344 rev 1

...

Mon Oct  9 14:46:17 2023 daemon.notice hostapd: Set new config for phy phy1:
Mon Oct  9 14:46:17 2023 daemon.notice wpa_supplicant[1718]: Set new config for phy phy1
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.302796] ath: EEPROM regdomain: 0x807c
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.306891] ath: EEPROM indicates we should expect a country code
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.313108] ath: doing EEPROM country->regdmn map search
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.318496] ath: country maps to regdmn code: 0x3a
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.323367] ath: Country alpha2 being used: CA
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.327876] ath: Regpair used: 0x3a
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.331414] ath: regdomain 0x807c dynamically updated by user
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.337326] ath: EEPROM regdomain: 0x807c
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.341393] ath: EEPROM indicates we should expect a country code
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.347589] ath: doing EEPROM country->regdmn map search
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.352988] ath: country maps to regdmn code: 0x3a
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.357849] ath: Country alpha2 being used: CA
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.362354] ath: Regpair used: 0x3a
Mon Oct  9 14:46:18 2023 kern.debug kernel: [   89.365904] ath: regdomain 0x807c dynamically updated by user
Mon Oct  9 14:46:19 2023 daemon.notice hostapd: Set new config for phy phy0:
Mon Oct  9 14:46:19 2023 daemon.notice wpa_supplicant[1718]: Set new config for phy phy0
[... 53 lines cut ...]
Mon Oct  9 14:46:27 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CA ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 chanlist=1 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40] channel=1  interface=phy0-ap0 bssid=<REDACTED> ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=0 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_group_rekey=3600 wpa_passphrase=<REDACTED> wpa_psk_file=/var/run/hostapd-phy0-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=<REDACTED> bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=1 wpa_key_mgmt=WPA-PSK okc=0 disable_pmksa_caching=1 dynamic_vlan=0 vlan_naming=1 vlan_no_bridge=1
Mon Oct  9 14:46:27 2023 kern.info kernel: [   98.933104] br-lan: port 3(phy0-ap0) entered blocking state
Mon Oct  9 14:46:27 2023 kern.info kernel: [   98.938824] br-lan: port 3(phy0-ap0) entered forwarding state
Mon Oct  9 14:46:27 2023 kern.info kernel: [   98.945269] br-lan: port 3(phy0-ap0) entered disabled state
Mon Oct  9 14:46:28 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.104082] ath: EEPROM regdomain: 0x807c
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.108361] ath: EEPROM indicates we should expect a country code
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.114769] ath: doing EEPROM country->regdmn map search
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.120204] ath: country maps to regdmn code: 0x3a
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.125107] ath: Country alpha2 being used: CA
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.129616] ath: Regpair used: 0x3a
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.133296] ath: regdomain 0x807c dynamically updated by user
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.139342] ath: EEPROM regdomain: 0x807c
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.143563] ath: EEPROM indicates we should expect a country code
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.149884] ath: doing EEPROM country->regdmn map search
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.155320] ath: country maps to regdmn code: 0x3a
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.160357] ath: Country alpha2 being used: CA
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.164971] ath: Regpair used: 0x3a
Mon Oct  9 14:46:28 2023 kern.debug kernel: [   99.168616] ath: regdomain 0x807c dynamically updated by user
Mon Oct  9 14:46:28 2023 daemon.notice netifd: Wireless device 'radio1' is now up
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-23:
Yes, happens on error, but not in normal conditions? 23.05.2?

guidosarducci, 2023-11-24:
Nothing to do with errors, as noted, but normal operation. It's most easily seen on boot, or when you enable a wireless interface in Luci. And it happens in all 23.05 including 23.05.2.

silentcreek, 2023-12-08:
I've also stumbled upon this. I briefly browsed the code to find where/when this might have changed. But this should not happen (and didn't on Openwrt 22.03). I'm seeing this on two targets as well: ath79 (TP Link Archer C7v2) and x86 (PC Engines APU2).

It also happens to me always while the devices work fine (used as AccessPoints).

guidosarducci, 2025-01-23:
This problem is still present in the latest `24.10` release candidates and really should be addressed before final release.

I also noticed another user trying to fix this in #17517 , where discussion confirms the problem is self-inflicted and due to an OpenWrt patch, but doesn't occur in upstream code.

@hauke Could you please re-tag this as `release/24.10`?

Neustradamus, 2025-01-26:
Badly for 24.10.x :/

Neustradamus, 2025-01-26:
It can be fixed in 23.05.x and 24.10.x?

guidosarducci, 2025-01-27:
Most likely yes, since I believe the code hasn't changed.

ckorber, 2025-05-23:
My PR is still open and under review.
I tested it today and it still hides the nasty details like the password with the output:
```
Fri May 23 06:09:04 2025 daemon.notice hostapd: Configuration file: <inline>
```
Problem is, that it hasn't got as much attention and approval as I wished.

PR #17718 (2025-08-01): hostapd: fix logging of configuration content
As discussed in #17517, there are contents of hostapd's configuration file logged in syslog. This includes critical information like passphrase`. To circumvent this condition, this commit logs only "inline" if config_fname is inline data.

Otherwise the upstream logic of hostapd applies.

Fixes: #14049