openwrt/openwrt #14053: TP-Link TL-WDR4900 v1 have poor routing performance after switching to DSA.

InsaneKnight, 2023-11-24:
### Describe the bug

By comparison between 22.03.5 (swconfig) and 23.05.2 (DSA), the bandwidth with an IPoE wan without flow offloading falls from 650Mb/s to 400Mb/s, and bandwidth with software flow offloading falls from 900 to 550. The bandwidth with a PPPoE wan is much poorer (260 vs 400 of ath79, seeing https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820159084 ), even with the walkaround in https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820224768 .

It seems that the same performance issue preventing ath79 targets from being switched to DSA also hits this target, as after disabling GRO of all user port as https://github.com/openwrt/openwrt/pull/13589#issuecomment-1766398009 suggests, the bandwidth with an IPoE wan without flow offloading falls to around 270, but bandwidth with software flow offloading becomes 900 again.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mpc85xx/p1010

### Device

TP-Link TL-WDR4900 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-11-24:
can you try modified workaround, adding wan device and  not ppp device to offload list? Do `fw4 check` before restart.

InsaneKnight, 2023-11-24:
> can you try modified workaround, adding wan device and  not ppp device to offload list? Do `fw4 check` before restart.
>
It has no difference on performance with no workaround.

brada4, 2023-11-24:
Lets do some more experiments - from wired client mostly.
Lets fix on one test tool either
https://www.waveform.com/tools/bufferbloat
https://speed.cloudflare.com
Closer one shows better latency results, stay with that.

Now measure ping from router  to:
1/ the ipoe server
2/ your provider's web
You will not get much better latency overall, lets put that aside for now.

Disable qdisc schedulers to save CPU spikes not related:
tc -s qdisc
tc qdisc replace dev eth0 root noqueue
(for all adapters , it will k-log warnings, but dont worry, reboot will restore past status)

Now run following tests:
You need to close and re-open browser tab between tests to close connections on the wrong side of intended offload.
- offload disabled
- offload enabled
- hardware offload enabled
And via workaround plugin file after disabling offload in luci:
- offload disabled on internet uplink but on on ppp device and br-lan
- other way ppp
- offload enabled on ethernet switch interface (like eth0, but not DSA members) and br-lan and best of upstream combination.

You can monitor cpu use with 
while sleep 1 ; do  uptime ; done | tee -a /tmp/test
Anything above 1.0 means it clamps at fully utilized CPU.

InsaneKnight, 2023-11-24:
> Lets do some more experiments - from wired client mostly.
> Lets fix on one test tool either
> https://www.waveform.com/tools/bufferbloat
> https://speed.cloudflare.com
> Closer one shows better latency results, stay with that.
Nether is close enough to me, so I will choose the closest test server of https://speedtest.net
>
> Now measure ping from router  to:
> 1/ the ipoe server
My provider only have pppoe server,  seq=0 ttl=255 time=1.60 ms
> 2/ your provider's web
ttl=249 time=2.40 ms
> You will not get much better latency overall, lets put that aside for now.
>
> Disable qdisc schedulers to save CPU spikes not related:
> tc -s qdisc
> tc qdisc replace dev eth0 root noqueue
> (for all [...]
D229.84, U41.09 (Mbit/s), load average: 0.15
> - offload enabled
devices = { "lan1", "lan2", "lan3", "lan4", "pppoe-wan", "wan" };
D268.72, U40.48, load average: 0.16
> - hardware offload enabled

devices = { "lan1", "lan2", "lan3", "lan4", "pppoe-wan", "wan" };

flags offload;

D268.77, U41.12, load average: 0.16
> And via workaround plugin file after disabling offload in luci:
> - offload disabled on internet uplink but on on ppp device and br-lan

devices = { "br-lan", "pppoe-wan" };

D364.97, U40.60, load average: 0.01

> - other way ppp

devices = { "br-lan", "wan" };

D265.62, U41.74, load average: 0.18

> - offload enabled on ethernet switch interface (like eth0, but not DSA members) and br-lan and best of upstream combination.

devices = { "br-lan", "pppoe-wan", "eth0" };

D367.24, U40.61, load average: 0.08

devices = { "pppoe-wan", "eth0" };

D358.81, U40.63, load average: 0.10

>
> You can monitor cpu use with 
> while sleep 1 ; do  uptime ; done | tee -a /tmp/test
> Anything above 1.0 means it clamps at fully utilized CPU.
>
The above results are all done in a configuration similar to the default one, in which br-lan bridges lan1~4, and pppoe goes through wan.X, but there are other VLANs which I need to bridge to the LAN side. However, if I bridge all user ports together (assuming the bridge is called br-eth0, bridging lan1~4 and wan) and let br-eth0.1 be LAN, and pppoe go through br-eth0.X, the performance will drop to D241.23, U41.16 at best.

brada4, 2023-11-24:
Now check ethtool -S eth0 -> things regarding drops, overruns and flow control? 
Basic conclusions:
no hw offload on your platform, it is a rarity, so fully expected
as soon as real hardware NIC is added to offload group performance drops drastically (see adapter stats for now)

InsaneKnight, 2023-11-24:
Under swconfig, a switch is merely a switch, so VLANs can be directed to different ports regardless of ethernet ports, but on DSA, VLANs seems to be filtered only within bridges of user ports, and extending the bridge to the wan port may lose performance, as the above  tests shows.

brada4, 2023-11-24:
DSA is meant to have a bit more control, it sends broadcasts to CPU port, so that CPU can adjust switching in switch ASIC, but that should not have influence on client to router  to gateway traffic, it is just more control on cable reconnects/MAC changes etc.

InsaneKnight, 2023-11-24:
> Now check ethtool -S eth0 -> things regarding drops, overruns and flow control? 
> Basic conclusions:
> no hw offload on your platform, it is a rarity, so fully expected
> as soon as real hardware NIC is added to offload group performance drops drastically (see adapter stats for now)
>
>
I believe the test is over, so I tore down the test setup and switch back to the "working" setup with my archer c7 v2 as the gateway. The test was done by temporarily disabling  the pppoe on the gateway, and directing the VLAN for pppoe to the TL-WDR4900 v1 for testing, so I cannot afford to keep the test setup too long.

If only you should have noticed me to collect stats earlier...

brada4, 2023-11-24:
Uff sorry....

InsaneKnight, 2023-11-25:
> Uff sorry.... I did quite elaborate test plan, just in hope some combination will get full speed....
> If you are at it - can you check if C7 gets some damage from adding hardware NIC-s to offload pool.

Archer C7 uses swconfig, so it only has eth0, eth1. pppoe goes through eth0.X, and eth1.1 is in br-lan.

With devices = { "br-lan", "pppoe-wan" } download speed can reach 400Mbit/s, but with devices = { "br-lan", "pppoe-wan", "eth0" } download speed will fall to 200Mbit/s, the rest can be seen in https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820159084 and https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820378626 .

brada4, 2023-11-25:
No need to produce no-op duplicate reports thanks.
Conclusion: [kernel doc](https://www.kernel.org/doc/html/v5.12/networking/nf_flowtable.html) is false regarding which ifaces to offload.

InsaneKnight, 2023-11-25:
> No need to produce no-op duplicate reports thanks.
> Conclusion: [kernel doc](https://www.kernel.org/doc/html/v5.12/networking/nf_flowtable.html) is false regarding which ifaces to offload. 
>
This document seems not to imply that adding a lower "physical" ethernet port to the flow table can offload the payload of pppoe going through it. However, what we observed is that adding a lower "physical" ethernet port to the flow table can interfere the offloading of pppoe going through it!

neheb, 2023-11-25:
one thing that has had success on ath79 is disabling GRO to improve performance.

brada4, 2023-11-25:
selectively disabli;g offloads sometimes helps

brada4, 2023-11-25:
Correct document https://www.kernel.org/doc/html/v5.15/networking/nf_flowtable.html#layer-2-encapsulation
Your tests prove opposite that offload should not happen on PPPoE lower interface.

InsaneKnight, 2023-11-25:
> Correct document https://www.kernel.org/doc/html/v5.15/networking/nf_flowtable.html#layer-2-encapsulation
> Your tests prove opposite that offload should not happen on PPPoE lower interface.
>
>
I found some doubtful point in net/netfilter/nf_flow_table_ip.c of linux 5.15:

> static inline __be16 nf_flow_pppoe_proto(const struct sk_buff *skb)
> {
>     __be16 proto;
>
>     proto = *((__be16 *)(skb_mac_header(skb) + ETH_HLEN +
>                  sizeof(struct pppoe_hdr)));
>     switch (proto) {
>     case htons(PPP_IP):
>         return htons(ETH_P_IP);
>     case htons(PPP_IPV6):
>         return htons(ETH_P_IPV6);
>     }
>
>     return 0;
> }
>
> static bool nf_flow_skb_encap_protocol(const struct sk_buff *skb, __be16 proto,
>                        u32 *offset)
> {
>     struct vlan_ethhdr *veth;
>
>     switch (skb->protocol) {
>     case htons(ETH_P_8021Q):
>         veth = (struct vlan_ethhdr *)skb_mac_header(skb);
>         if (veth->h_vlan_encapsulated_proto == proto) {
>             *offset += VLAN_HLEN;
>             return true;
>         }
>         break;
>     case htons(ETH_P_PPP_SES):
>         if (nf_flow_pppoe_proto(skb) == proto) {
>             *offset += PPPOE_SES_HLEN;
>             return true;
>         }
>         break;
>     }
>
>     return false;
> }
>
> static void nf_flow_encap_pop(struct sk_buff *skb,
>                   struct flow_offload_tuple_rhash *tuplehash)
> {
>     struct vlan_hdr *vlan_hdr;
>     int i;
>
>     for (i = 0; i < tuplehash->tuple.encap_num; i++) {
>         if (skb_vlan_tag_present(skb)) {
>             __vlan_hwaccel_clear_tag(skb);
>             continue;
>         }
>         switch (skb->protocol) {
>         case htons(ETH_P_8021Q):
>             vlan_hdr = (struct vlan_hdr *)skb->data;
>             __skb_pull(skb, VLAN_HLEN);
>             vlan_set_encap_proto(skb, vlan_hdr);
>             skb_reset_network_header(skb);
>             break;
>         case htons(ETH_P_PPP_SES):
>             skb->protocol = nf_flow_pppoe_proto(skb);
>             skb_pull(skb, PPPOE_SES_HLEN);
>             skb_reset_network_header(skb);
>             break;
>         }
>     }
> }
These functions seem capable to handle ETH{VLAN{IP}} or ETH{PPPOE{IP}} packets, but unable to handle ETH{VLAN{PPPOE{IP}}} packets.

However, untagging the VLAN for PPPOE to (cpu port) eth0 with the integrated switch on my archer c7 v2 ( it is allowed only with swconfig ), making received packets in ETH{PPPOE{IP}} form, cannot improve the performance with

devices = { "br-lan", "pppoe-wan", "eth0" } or devices = { "br-lan", "eth0" } .

brada4, 2023-11-25:
Also one encap in mainline kernel https://github.com/torvalds/linux/blob/0f5cc96c367f2e780eb492cc9cab84e3b2ca88da/net/netfilter/nf_flow_table_offload.c#L627
Does not contradict doc....

brada4, 2023-11-25:
Is your WAN dsa port or vlan? Try wan@eth0 or eth0.1 + pppoe-wan and br-lan

InsaneKnight, 2023-11-25:
> Is your WAN dsa port or vlan? Try ***@***.*** or eth0.1 + pppoe-wan and br-lan
>
Archer c7 v2 has swconfig instead of DSA, so there is no ***@***.***, only eth0 and eth1, as said in https://github.com/openwrt/openwrt/issues/14053#issuecomment-1826175088 .

InsaneKnight, 2023-11-25:
> eth0.1 + pppoe-wan and br-lan
>
See https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820159084 .

brada4, 2023-11-25:
Device page says it has vlans - are they untagged now?
```
[   22.335597] eth1: link up (1000Mbps/Full duplex)
[   22.340404] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   22.364540] br-lan: port 1(eth1.1) entered blocking state
[   22.370066] br-lan: port 1(eth1.1) entered disabled state
[   22.375868] device eth1.1 entered promiscuous mode
[   22.380773] device eth1 entered promiscuous mode
[   22.429179] br-lan: port 1(eth1.1) entered blocking state
[   22.434664] br-lan: port 1(eth1.1) entered forwarding state
[   22.440491] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   22.483954] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   22.513229] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
[   23.758173] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   23.764989] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   23.853292] eth0: link up (1000Mbps/Full duplex)
[   23.897359] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.980284] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
```

InsaneKnight, 2023-11-25:
> Device page says it has vlans - are they untagged now?
> ```
> [   22.335597] eth1: link up (1000Mbps/Full duplex)
> [   22.340404] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
> [   22.364540] [...]
Yes, I let br-lan cover eth1, pppoe go through eth0, and switch handle the tag for them, to test whether ETH{PPPOE{IP}} can improve. The result is negative, but this setup should save the work for cpu to process tags, so I retained it.

> [   35.088024] eth1: link up (1000Mbps/Full duplex)
> [   35.099437] br-lan: port 1(eth1) entered blocking state
> [   35.104758] br-lan: port 1(eth1) entered disabled state
> [   35.110372] device eth1 entered promiscuous mode
> [   35.137101] br-lan: port 1(eth1) entered blocking state
> [   35.142461] br-lan: port 1(eth1) entered forwarding state
> [   35.236408] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 1 is up
> [   35.242704] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
> [   35.248949] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 3 is up
> [   35.255176] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 4 is up
> [   35.261414] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
> [   35.268906] eth0: link up (1000Mbps/Full duplex)
> [   35.408153] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
> [   36.098996] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

InsaneKnight, 2023-11-25:
DSA loses the capability of untagging for cpu ports.

brada4, 2023-11-25:
Yes, DSA is hidden tagging....

brada4, 2023-11-25:
You should be able to use wan@eth1 (?) as accelerated device?

brada4, 2023-11-25:
```
ethtool -k eth0 | grep an-o
rx-vlan-offload: off [fixed]
tx-vlan-offload: on
```

InsaneKnight, 2023-11-25:
> Yes, DSA is hidden tagging.... On some platforms "wan" is wired directly wo physical port bypassing switch, so you get wild adapter renames on upgrade....
I know, https://github.com/openwrt/openwrt/pull/10238 . That is why I remain hesitating to use them (I have 2, one RB750gr3 and one Unielec U7621-01) for I need to bridge some VLANs on the wan side to lan side, which can be easily done with swconfig, but hard with DSA or a physical port bypassing switch.

brada4, 2023-11-25:
I have wan ripped out from DSA, which makes it ideal "first router" , but if you had 2 eth at DSA you may get gratis no-offload VLAN processing.

InsaneKnight, 2023-11-25:
> You should be able to use ***@***.*** (?) as accelerated device? 
>
> I have wan ripped out from DSA, which makes it ideal "first router" , but if you had 2 eth at DSA you may get gratis no-offload VLAN processing.

Is it allowed now on platforms like WRT1900AC v1 or ones with ipq8064 and ar8327/qca8337, to have ***@***.*** and ***@***.*** ?

brada4, 2023-11-25:
depends, you have to look at dts-es, in some wan is bypassing dsa. 
1) offload can be enabled one level below pppoe-wan ie eth0.1 or wan@eth1 or wan (old hidden eth1 wired directly to port)
2) even 1 phy 1 eth link in DSA will employ likely no-offload VLAN support.

InsaneKnight, 2023-11-26:
I have a VLAN setup similar to https://openwrt.org/docs/guide-user/network/dsa/converting-to-dsa#complex_vlan_configuration_-_vlans_tagged_on_multiple_switch_ports_and_the_router . In order to handle this on DSA, wan should be bridged together with lan1~4. I have had a simplified test on my TP-Link TL-WDR4900 v1 for software flow offload performance, and the result showed that devices = { "br-lan", "pppoe-wan" } can only get D260, like
> devices = { "lan1", "lan2", "lan3", "lan4", "pppoe-wan", "wan" }; D268.72, U40.48, load average: 0.16

in https://github.com/openwrt/openwrt/issues/14053#issuecomment-1825865297 , maybe because wan was covered by br-lan, the overall result is similar , but devices = { "lan1", "lan2", "lan3", "lan4", "pppoe-wan" } can get D373, like

> devices = { "br-lan", "pppoe-wan" }; D364.97, U40.60, load average: 0.01
in https://github.com/openwrt/openwrt/issues/14053#issuecomment-1825865297 , in which case wan is not a part of br-lan.

So excluding the "physical" wan port to transfer pppoe from the flow table mainly works, regardless whether it is bridged along with lan ports.

The test setup was already torn down, so further test request can only be performed during the next test.

brada4, 2023-11-26:
I did a small experiment:
```
        iif "br-lan" meta l4proto { tcp,udp} flow offload @myft counter; 
        iif "dsa" meta l4proto { tcp,udp} flow offload @myft counter; # zeeero #was eth1
        iif "lan2" meta l4proto { tcp,udp} flow offload @myft counter; # zeeero
        iif "wan" meta l4proto { tcp,udp} flow offload @myft counter; #was eth0
        iif "phy0-ap0" meta l4proto { tcp,udp} flow offload @myft counter; 
```
And turns out no states get picked up on either DSA element. 

Rule is passed more often if you force dropback more often 
```
net.netfilter.nf_flowtable_tcp_timeout = 1
net.netfilter.nf_flowtable_udp_timeout = 1
```

Can you do experiment:
iif wan -> counter
iif ppoe -> counter
And check if ppoe or wan picks up any state when that and dsa "almost vlan" has also offload enabled.
Basically all 3 combos....

InsaneKnight, 2023-11-26:
> I did a small experiment:
> ```
>         iif "br-lan" meta l4proto { tcp,udp} flow offload @myft counter;
>         iif "dsa" meta l4proto { tcp,udp} flow offload @myft counter; # zeeero #was [...]
Sorry, I could not understand what you want me to do.

brada4, 2023-11-26:
Spread out the flow offload part by interface in the filter+forward section to identify which adapters actually pick up flows....

InsaneKnight, 2023-11-26:
> ```
>         iif "br-lan" meta l4proto { tcp,udp} flow offload @myft counter;
>         iif "dsa" meta l4proto { tcp,udp} flow offload @myft counter; # zeeero #was eth1
>         iif "lan2" meta l4proto [...]

In what context should I write these?

brada4, 2023-11-26:
then nft list ruleset will tell when offload picks up state:
```
       chain forward {
                type filter hook forward priority filter; policy drop;
                iif "br-lan" meta l4proto { tcp, udp } flow add \@myft counter packets 7266 bytes 1033541
                iif "wan" meta l4proto { tcp, udp } flow add \@myft counter packets 3392 bytes 992878
}
```

InsaneKnight, 2023-11-26:
> then nft list ruleset will tell when offload picks up state:
> ```
>        chain forward {
>                 type filter hook forward priority filter; policy drop;
>                 iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 7266 bytes 1033541
>                 iif "wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 3392 bytes 992878
> }
> ```
>
>
On archer c7 v2:

>     flowtable myft {
>         hook ingress priority filter
>         devices = { br-lan, pppoe-wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "eth1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 108 bytes 10353
>         iif "phy0-ap0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "phy1-ap0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 205 bytes 21559
>         ...
>     }
>     flowtable myft {
>         hook ingress priority filter
>         devices = { br-lan, eth0, pppoe-wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "eth1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 52 bytes 3712
>         iif "phy0-ap0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "phy1-ap0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 45 bytes 4410
>         ...
>     }
On archer c7 v2 with CPU handling VLAN X for pppoe:

>     flowtable myft {
>         hook ingress priority filter
>         devices = { br-lan, pppoe-wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 74 bytes 5809
>         iif "eth0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "eth0.X" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 6 bytes 609
>         ...
>     }

On TP-Link TL-WDR4900 v1

>     flowtable ft {
>         hook ingress priority filter
>         devices = { lan1, lan2, lan3, lan4, pppoe-wan, wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan2" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan3" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan4" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan.1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 65 bytes 6769
>         iif "br-lan.X" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 5 bytes 1148
>         ...
>     }
>     flowtable ft {
>         hook ingress priority filter
>         devices = { lan1, lan2, lan3, lan4, pppoe-wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan2" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan3" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "lan4" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "br-lan.1" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 53 bytes 2776
>         iif "br-lan.X" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add ***@***.*** counter packets 0 bytes 0
>        ...
>     }

brada4, 2023-11-26:
you need to include 'lower' device in offload loop to figure out if its inclusion sabo%ages or widens offload.

brada4, 2023-11-26:
A bit more verbose explanation:
I am trying to count offload pickups on each interface, so I split offload pickup rule into one rule per interface with counter
My finding: no offload pick-ups occur on dsa master interface or dsa physical ports.
My question to you-
if interface under ppoe is dsa port and you enable offload actually none of 2 ports pick up offloadable flow while pppoe alone does pick up.
Also same in regard to eth0 -> eth0.1 -> ppoe-wan (the thing only one encapsulation is handled - if offloading on eth0 as opposed to eth0.1 disables offloading pickups for all interface chain.

InsaneKnight, 2023-11-27:
> My question to you-
> if interface under ppoe is dsa port and you enable offload actually none of 2 ports pick up offloadable flow while pppoe alone does pick up.
> Also same in regard to eth0 -> eth0.1 [...]
What are your questions actually?

brada4, 2023-11-27:
if ppoe+dsa offloads or no

InsaneKnight, 2023-11-27:
> if ppoe+dsa offloads or no
>
In https://github.com/openwrt/openwrt/issues/14053#issuecomment-1825865297 ,

> devices = { "pppoe-wan", "eth0" }; D358.81, U40.63, load average: 0.10
a little poorer than

> devices = { "br-lan", "pppoe-wan" }; D364.97, U40.60, load average: 0.01

brada4, 2023-11-27:
I mean make multiple counter rules to determine which interface actually picks up packet to offload engine.

InsaneKnight, 2023-11-27:
> I mean make multiple counter rules to determine which interface actually picks up packet to offload engine.
>
On TP-Link TL-WDR4900 v1

>     flowtable ft {
>         hook ingress priority filter
>         devices = { eth0, pppoe-wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan1" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan2" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan3" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan4" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "wan" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "br-lan.1" meta l4proto { tcp, udp } flow add @ft counter packets 113 bytes 8717
>         iif "br-lan.X" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add @ft counter packets 9 bytes 2668
>        ...
>     }

brada4, 2023-11-27:
I meant make full devices {} list too, then determine which interfaces actually contribute to offload....

InsaneKnight, 2023-11-27:
We should be performing scientific research instead of riddle solving. If you do not write out what you "mean" explicitly and precisely, I can hardly get them.
> I meant make full devices {} list too, then determine which interfaces actually contribute to offload....
>
On TP-Link TL-WDR4900 v1:

> flowtable ft {
>         hook ingress priority filter
>         devices = { br-lan, br-lan.1, br-lan.X, eth0, lan1, lan2, lan3, lan4, pppoe-wan, wan }
>         counter
>     }
>
>     chain forward {
>         type filter hook forward priority filter; policy drop;
>         iif "eth0" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan1" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan2" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan3" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "lan4" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "wan" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "br-lan" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "br-lan.1" meta l4proto { tcp, udp } flow add @ft counter packets 57 bytes 3024
>         iif "br-lan.X" meta l4proto { tcp, udp } flow add @ft counter packets 0 bytes 0
>         iif "pppoe-wan" meta l4proto { tcp, udp } flow add @ft counter packets 6 bytes 1544
>         ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
>         iifname "br-lan.1" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
>         iifname "pppoe-wan" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
>         jump handle_reject
>     }

brada4, 2023-11-27:
Oki - conclusion - no offload on any DSA port ;-(
Some damage in trying that on wan@dsa port.

brada4, 2023-11-27:
@jow- can you have a look at it:
- dsa host or socket ports do not pick up offload states and reduces forwarding performance a bit if present in offload group
- offload states can handle only one of VLAN or PPPoE encapsulation 1 level deeper from forward rule

InsaneKnight, 2026-04-09:
Currently the throughput of TL-WDR4900 v1 is 650 Mbps (lan->wan) / 600 (wan->lan) without flow offload, and 800 Mbps (lan->wan) / 750 (wan->lan) with software flow offload, when GRO is enabled on DSA user ports by default.

> It seems that the same performance issue preventing ath79 targets from being switched to DSA also hits this target, as after disabling GRO of all user port as #13589 (comment) suggests, the bandwidth with an IPoE wan without flow offloading falls to around 270, but bandwidth with software flow offloading becomes 900 again.

After GRO on DSA user ports is disabled, the throughput becomes 930 Mbps (lan->wan) / 900 (wan->lan) with software flow offload.

neheb, 2026-04-09:
I think the ath79 issue is SGMII being broken.

This uses RGMII.

neheb, 2026-10-08:
Claude says:

```
Why GRO costs throughput on gianfar with DSA

I checked the source.

1. RX: the checksum offload never works on DSA traffic. gianfar marks a packet as checksum-verified only when the controller sets the RXFCB_CIP|RXFCB_CTU flags (gfar_rx_checksum()). The QCA8327 inserts its 2-byte tag_qca header where the ethertype normally sits. The controller's header parser then doesn't recognise the packet as IP, so every packet on lan1–4/wan arrives with no verified checksum, and GRO has to checksum it in software. This is the same thing ag71xx hit before the checksum engine. Unlike ath79, the P1010 has no separate checksum engine to fall back on. As far as I know the controller can't be told to skip a custom tag, but that needs checking in the P1010 reference manual.
2. TX: the DSA ports get no offloads at all. DSA user ports copy their features from conduit->vlan_features (net/dsa/user.c:2716). gianfar sets IP_CSUM | SG in features but never sets vlan_features. So lan1–4 and wan have no scatter-gather and no TX checksum offload. Every GRO-merged packet forwarded out a DSA port gets split back up and checksummed in software, which is the job your ag71xx TX scatter-gather and TX checksum patches took off the CPU.

What could help

- Set vlan_features in gianfar (likely to help). Something like dev->vlan_features = NETIF_F_IP_CSUM | NETIF_F_SG | NETIF_F_HIGHDMA when the chip has checksum offload. This should be safe with the QCA tag: gianfar's TX checksum doesn't parse the frame, because software tells the hardware where the headers are (gfar_tx_checksum() fills in l3os/l4os from skb_network_offset(), which already counts the tag). It needs testing on hardware against the eTSEC checksum errata paths. It's a small change and could go upstream.
- RX checksum: no clear fix. Without hardware RX checksum, GRO on the user ports will always cost something. Turning GRO off on the user ports (the current workaround), or using flow offload, remains the right advice for this board.
```

it gave this patch

[700-net-gianfar-advertise-tx-offloads-to-upper-devices.patch](https://github.com/user-attachments/files/33221530/700-net-gianfar-advertise-tx-offloads-to-upper-devices.patch)

InsaneKnight, 2026-10-09:
> neheb left a comment (openwrt/openwrt#14053)
>
> Claude says:
>
> ```
> Why GRO costs throughput on gianfar with DSA
>
> I checked the source.
>
> 1. RX: the checksum offload never works on DSA traffic. gianfar marks a packet as checksum-verified only when the controller sets the RXFCB_CIP|RXFCB_CTU flags (gfar_rx_checksum()). The QCA8327 inserts its 2-byte tag_qca header where the ethertype normally sits. The controller's header parser then doesn't recognise the packet as IP, so every packet on lan1–4/wan arrives with no verified checksum, and GRO has to checksum it in software. This is the same thing ag71xx hit before the checksum engine. Unlike ath79, the P1010 has no separate checksum engine to fall back on. As far as I know the controller can't be told to skip a custom tag, but that needs checking in the P1010 reference manual.
> 2. TX: the DSA ports get no offloads at all. DSA user ports copy their features from conduit->vlan_features (net/dsa/user.c:2716). gianfar sets IP_CSUM | SG in features but never sets vlan_features. So lan1–4 and wan have no scatter-gather and no TX checksum offload. Every GRO-merged packet forwarded out a DSA port gets split back up and checksummed in software, which is the job your ag71xx TX scatter-gather and TX checksum patches took off the CPU.
>
> What could help
>
> - Set vlan_features in gianfar (likely to help). Something like dev->vlan_features = NETIF_F_IP_CSUM | NETIF_F_SG | NETIF_F_HIGHDMA when the chip has checksum offload. This should be safe with the QCA tag: gianfar's TX checksum doesn't parse the frame, because software tells the hardware where the headers are (gfar_tx_checksum() fills in l3os/l4os from skb_network_offset(), which already counts the tag). It needs testing on hardware against the eTSEC checksum errata paths. It's a small change and could go upstream.
> - RX checksum: no clear fix. Without hardware RX checksum, GRO on the user ports will always cost something. Turning GRO off on the user ports (the current workaround), or using flow offload, remains the right advice for this board.
> ```
>
> it gave this patch
>
> [700-net-gianfar-advertise-tx-offloads-to-upper-devices.patch](https://github.com/user-attachments/files/33221530/700-net-gianfar-advertise-tx-offloads-to-upper-devices.patch)
>
After applying this patch, Throughput of TL-WDR4900 v1 is 600 Mbps without flow offload, and 800 Mbps with software flow offload, no obvious difference with https://github.com/openwrt/openwrt/issues/14053#issuecomment-4211419901 .

neheb, 2026-10-09:
Unfortunate. If I had the hardware I’d hook it up to serial and ask Claude to speed up Ethernet.

ath79 has been fairly successful (58% speed up in one case) but apm821xx not so much.