openwrt/openwrt #14056: DHCPv6 Client includes MAP-T and MAP-E OROs even though the map package isn't installed

detobate, 2023-11-24:
### Describe the bug

The odhcpv6 client sends a DHCPv6 Solicit asking for MAP-T/E (and lw4over6 etc.) even though the packages required to support them (map) are not installed.  This can trigger some ISPs to **only** provide IPv4aaS instead of native DHCPv4, leaving users without IPv4 connectivity until they install the required map package.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

Fresh install, no non-default configuration required

### Actual behaviour

DHCPv6 Solicit contain the MAP-T, MAP-E and other Softwire options as defined by [RFC 7598](https://datatracker.ietf.org/doc/html/rfc7598), within the DHCPv6 Option Request Option.

### Expected behaviour

The DHCPv6 Solicit should only request options within the Option Request Option, for features and functionality the client and system can support.
i.e., if the map package isn't installed, it shouldn't ask for MAP-T/E options

### Additional info

An alternative fix, and one I would prefer, is for the default OpenWRT image to include the map and other IPv4aaS packages, so it can support these protocols out-of-the-box.  

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

detobate, 2024-01-05:
[openwrt/odhcp6c/#81](https://github.com/openwrt/odhcp6c/pull/81) will resolve this ticket

r-pats, 2024-11-11:
Waiting on openwrt/odhcp6c/#89 and openwrt/openwrt/#16209

Headcrabed, 2025-02-09:
https://github.com/openwrt/odhcp6c/pull/89 is now merged, still need to wait for https://github.com/openwrt/openwrt/pull/16209