openwrt/openwrt #14063: This platform requires CONFIG_GRUB_SERIAL be set

very20101, 2023-11-25:
### Describe the bug

Download package
……

make[2] -C feeds/luci/applications/luci-app-firewall download

make[2] -C package/boot/uboot-armsr download

make[2] -C feeds/luci/applications/luci-app-frps download

make[2] -C feeds/luci/applications/luci-app-nft-qos download

ERROR: package/base-files failed to build.

make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

make[1] target/download

make[2] -C target/linux download

=========================================================

Compile the fireware

……
make[3]: Leaving directory '/workdir/openwrt/package/system/usign'

time: package/system/usign/compile#0.19#0.08#0.23

make[3]: Entering directory '/workdir/openwrt/package/base-files'

/workdir/openwrt/target/linux/armsr/base-files.mk:3: *** This platform requires CONFIG_GRUB_SERIAL be set!. Stop.

make[3]: Leaving directory '/workdir/openwrt/package/base-files'

time: package/base-files/compile#0.66#0.05#0.72

ERROR: package/base-files failed to build.

make[2]: *** [package/Makefile:128: package/base-files/compile] Error 1

make[2]: Leaving directory '/workdir/openwrt'

make[1]: *** [package/Makefile:122: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2

make[1]: Leaving directory '/workdir/openwrt'

make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2

Error: Process completed with exit code 2.

### OpenWrt version

OpenWrt version VERSION_INFO:  useVersionInfo: Author: Daniel Golle<br/>date: 2023-11-25<br/>commit: mediatek: remove DTS property added by mistake<br/>commit hash: 1b70025ad1f5d8dd93e8415de7817d7975d4a8d2  DATE: 2023-11-25

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

armsr/armv8

### Device

Phicomm N1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

very20101, 2023-11-25:
Sorry, I made a mistake. 

OpenWrt version: R23.10.24 (kernel 5.15.139)
Device:  s905d_n1

But I can't get the  snapshot info  due to failed build.

hauke, 2023-11-25:
This is probably caused by b10768476f9dc5178d0038e3877b14f95184e46e.
@pprindeville Could you please have a look
@very20101  Please post your build configuration. You can use `./scripts/diffconfig.sh`to export a short version.

very20101, 2023-11-25:
configuration
==========================
CONFIG_TARGET_armsr=y
CONFIG_TARGET_armsr_armv8=y
CONFIG_TARGET_armsr_armv8_DEVICE_generic=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BTRFS_PROGS_ZSTD=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIBSODIUM_MINIMAL=y
CONFIG_LIBSSH2_OPENSSL=y
# CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_TAR_BZIP2=y
CONFIG_PACKAGE_TAR_GZIP=y
CONFIG_PACKAGE_TAR_XZ=y
CONFIG_PACKAGE_TAR_ZSTD=y
CONFIG_PACKAGE_adblock=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_bind-check=y
CONFIG_PACKAGE_bind-client=y
CONFIG_PACKAGE_bind-ddns-confgen=y
CONFIG_PACKAGE_bind-dig=y
CONFIG_PACKAGE_bind-dnssec=y
CONFIG_PACKAGE_bind-host=y
CONFIG_PACKAGE_bind-libs=y
CONFIG_PACKAGE_bind-nslookup=y
CONFIG_PACKAGE_bind-rndc=y
CONFIG_PACKAGE_bind-server=y
CONFIG_PACKAGE_bind-tools=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
CONFIG_PACKAGE_brook=y
CONFIG_PACKAGE_bsdtar=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_PACKAGE_bzip2=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_chattr=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_collectd=y
CONFIG_PACKAGE_collectd-mod-cpu=y
CONFIG_PACKAGE_collectd-mod-interface=y
CONFIG_PACKAGE_collectd-mod-iwinfo=y
CONFIG_PACKAGE_collectd-mod-load=y
CONFIG_PACKAGE_collectd-mod-memory=y
CONFIG_PACKAGE_collectd-mod-network=y
CONFIG_PACKAGE_collectd-mod-rrdtool=y
CONFIG_PACKAGE_coreutils=y
[... 335 lines cut ...]
CONFIG_SINGBOX_WITH_ECH=y
CONFIG_SINGBOX_WITH_GVISOR=y
CONFIG_SINGBOX_WITH_QUIC=y
CONFIG_SINGBOX_WITH_REALITY_SERVER=y
CONFIG_SINGBOX_WITH_UTLS=y
CONFIG_SINGBOX_WITH_WIREGUARD=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
# CONFIG_TARGET_IMAGES_GZIP is not set
# CONFIG_TARGET_ROOTFS_CPIOGZ is not set
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
==================================

The configuration compiled and built successfully 5 days ago.

pprindeville, 2023-11-26:
What serial is the console port on?  And what happens when you set `CONFIG_GRUB_SERIAL=ttyS0` (or whatever) and rebuild?

hauke, 2023-11-26:
These two options are not set:
```
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
```
Then `CONFIG_GRUB_EFI_IMAGES` is not available and you can not set `CONFIG_GRUB_SERIAL`. 

@pprindeville  You should make `CONFIG_GRUB_SERIAL` probably independent of the `GRUB` options. Something like this:
````
	config TARGET_SERIAL
		string "Serial port device"
		depends on TARGET_x86 || TARGET_armsr
		default "ttyS0"
````

hauke, 2023-11-26:
Please try #14078 It should fix your problem.

very20101, 2023-11-28:
The problem is still reproduced as CONFIG_TARGET_ROOTFS_EXT4FS be set

2023-11-28T01:18:01.9848198Z make[2]: Entering directory '/workdir/openwrt/scripts/config'
2023-11-28T01:18:01.9848968Z make[2]: 'conf' is up to date.
2023-11-28T01:18:01.9849732Z make[2]: Leaving directory '/workdir/openwrt/scripts/config'
2023-11-28T01:18:03.1281390Z Checking 'rsync'... ok.
2023-11-28T01:18:04.7371225Z time: target/linux/prereq#0.09#0.04#0.13
2023-11-28T01:18:05.4867410Z  make[1] tools/flock/compile
2023-11-28T01:18:07.4142160Z  make[2] -C tools/flock compile
2023-11-28T01:18:07.6469096Z  make[1] tools/download
2023-11-28T01:18:09.5502201Z  make[2] -C tools/bison download
2023-11-28T01:18:09.5514015Z  make[2] -C tools/autoconf-archive download
2023-11-28T01:18:09.5515978Z  make[2] -C tools/cmake download
2023-11-28T01:18:09.5569399Z  make[2] -C tools/cpio download
2023-11-28T01:18:09.5582013Z  make[2] -C tools/bc download
2023-11-28T01:18:09.5598254Z  make[2] -C tools/autoconf download
2023-11-28T01:18:09.5646257Z  make[2] -C tools/dosfstools download
2023-11-28T01:18:09.5666856Z  make[2] -C tools/automake download
2023-11-28T01:18:10.0643122Z  make[2] -C tools/e2fsprogs download
2023-11-28T01:18:10.1122922Z  make[2] -C tools/elfutils download
2023-11-28T01:18:10.1633019Z  make[2] -C tools/expat download
2023-11-28T01:18:10.2016912Z  make[2] -C tools/fakeroot download
2023-11-28T01:18:10.3476346Z  make[2] -C tools/findutils download
2023-11-28T01:18:10.3989529Z  make[2] -C tools/firmware-utils download
2023-11-28T01:18:10.4129816Z  make[2] -C tools/flex download
2023-11-28T01:18:10.6298640Z  make[2] -C tools/gengetopt download
2023-11-28T01:18:10.6350070Z  make[2] -C tools/gnulib download
2023-11-28T01:18:10.6666426Z  make[2] -C tools/libressl download
2023-11-28T01:18:10.8440353Z  make[2] -C tools/libtool download
2023-11-28T01:18:10.9065149Z  make[2] -C tools/lzma download
2023-11-28T01:18:11.0090135Z  make[2] -C tools/m4 download
2023-11-28T01:18:11.1379048Z  make[2] -C tools/make-ext4fs download
2023-11-28T01:18:11.2641858Z  make[2] -C tools/meson download
2023-11-28T01:18:11.3073574Z  make[2] -C tools/missing-macros download
2023-11-28T01:18:11.3495479Z  make[2] -C tools/mkimage download
2023-11-28T01:18:11.4428915Z  make[2] -C tools/mklibs download
2023-11-28T01:18:11.6140651Z  make[2] -C tools/mtd-utils download
2023-11-28T01:18:11.6257066Z  make[2] -C tools/mtools download
2023-11-28T01:18:11.6581144Z  make[2] -C tools/ninja download
2023-11-28T01:18:11.7291754Z  make[2] -C tools/padjffs2 download
2023-11-28T01:18:11.7728174Z  make[2] -C tools/patch-image download
2023-11-28T01:18:11.8163228Z  make[2] -C tools/patchelf download
2023-11-28T01:18:11.9549039Z  make[2] -C tools/pkgconf download
2023-11-28T01:18:12.0216071Z  make[2] -C tools/quilt download
2023-11-28T01:18:12.1617065Z  make[2] -C tools/squashfs4 download
2023-11-28T01:18:12.3550029Z  make[2] -C tools/sstrip download
2023-11-28T01:18:12.5159861Z  make[2] -C tools/zip download
2023-11-28T01:18:12.7534739Z  make[2] -C tools/zlib download
2023-11-28T01:18:12.7888255Z  make[2] -C tools/zstd download
2023-11-28T01:18:12.9668998Z  make[2] -C tools/gmp download
2023-11-28T01:18:13.1612850Z  make[2] -C tools/mpc download
2023-11-28T01:18:13.1648833Z  make[2] -C tools/mpfr download
2023-11-28T01:18:13.2474316Z  make[2] -C tools/libdeflate download
2023-11-28T01:18:13.4871336Z  make[2] -C tools/patch download
2023-11-28T01:18:13.5343567Z  make[2] -C tools/tar download
2023-11-28T01:18:13.6323736Z  make[2] -C tools/xz download
2023-11-28T01:18:13.6693573Z  make[2] -C tools/flock download
2023-11-28T01:18:13.7159155Z  make[2] -C tools/sed download
2023-11-28T01:18:17.1064871Z  make[1] toolchain/download
2023-11-28T01:18:18.9458580Z  make[2] -C toolchain/kernel-headers download
2023-11-28T01:18:18.9502282Z  make[2] -C toolchain/gcc/initial download
2023-11-28T01:18:18.9515005Z  make[2] -C toolchain/gdb download
2023-11-28T01:18:18.9556742Z  make[2] -C toolchain/binutils download
2023-11-28T01:18:18.9568949Z  make[2] -C toolchain/gcc/final download
2023-11-28T01:18:18.9586434Z  make[2] -C toolchain/musl download
2023-11-28T01:18:18.9636012Z  make[2] -C toolchain/fortify-headers download
2023-11-28T01:18:22.3341315Z  make[1] package/download
2023-11-28T01:18:24.2317346Z  make[2] -C package/boot/grub2 download
2023-11-28T01:18:24.2330155Z  make[2] -C package/base-files download
2023-11-28T01:18:24.2377532Z  make[2] -C feeds/luci/applications/luci-app-adblock download
[... 12 lines cut ...]
================================
config file with CONFIG_TARGET_ROOTFS_EXT4FS be set

# Root filesystem images
#
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_EXT4_RESERVED_PCT=0
CONFIG_TARGET_EXT4_BLOCKSIZE_4K=y
# CONFIG_TARGET_EXT4_BLOCKSIZE_2K is not set
# CONFIG_TARGET_EXT4_BLOCKSIZE_1K is not set
CONFIG_TARGET_EXT4_BLOCKSIZE=4096
# CONFIG_TARGET_EXT4_JOURNAL is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_TARGET_UBIFS_FREE_SPACE_FIXUP=y
CONFIG_TARGET_UBIFS_JOURNAL_SIZE=""
CONFIG_GRUB_EFI_IMAGES=y
CONFIG_GRUB_CONSOLE=y
CONFIG_GRUB_SERIAL="ttyS0"
CONFIG_GRUB_BAUDRATE=115200
# CONFIG_GRUB_FLOWCONTROL is not set
CONFIG_GRUB_BOOTOPTS=""
CONFIG_GRUB_TIMEOUT="5"
CONFIG_GRUB_TITLE="OpenWrt"
# CONFIG_VMDK_IMAGES is not set
# CONFIG_TARGET_IMAGES_GZIP is not set

commit 4211eab0 (2023-11-29): target: Make TARGET_SERIAL independent of GRUB configuration

GRUB_SERIAL is also used for the default serial on the target and not
only in grub. When no grub was build it was not available and the build
fails.

Rename GRUB_SERIAL to TARGET_SERIAL and make it always available on x86
and armsr targets.

Fixes: #14063
Fixes: b10768476f9d ("x86,armsr: interpolate GRUB_SERIAL into /etc/inittab")
PR #14078 (2023-11-29): target: Make TARGET_SERIAL independent of GRUB configuration
GRUB_SERIAL is also used for the default serial on the target and not only in grub. When no grub was build it was not available and the build fails.

Rename GRUB_SERIAL to TARGET_SERIAL and make it always available on x86 and armsr targets.

Fixes: #14063
Fixes: b10768476f9d ("x86,armsr: interpolate GRUB_SERIAL into /etc/inittab")
closed 2023-11-29 by openwrt-bot