openwrt/openwrt #14089: Switch back to ath10k firmware instead of ath10k-ct

clebersantz, 2023-11-28:
### Describe the bug

According to greearb on [Issue 81](https://github.com/greearb/ath10k-ct/issues/81) of ath10k-ct firmware 
`I am not planning to add any new features to ath10k firmware.`

He is not willing to add new features on ath10k-ct firmware, and since this firmware is the default on newer OpenWRT versions, is not better to switch back to old ath10k firmware that has all the features, like Mesh, IBBS, etc ?

We have a forum discussion about the problems with ath10k-ct firmware [Topic 95787](https://forum.openwrt.org/t/state-of-tp-link-archer-c7v2-v5-in-2023/95787/), besides having no Mesh, IBBS on it as well.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

clebersantz, 2023-11-28:
> Does it work stable as of now? Like #6937

The 6937 is a problem reported for -CT version too, as I can see:
`@OpenWrt_LR:~# opkg list-installed`
`ath10k-firmware-qca988x-ct - 2018-10-10-d366b80d-1`
`base-files - 196-r9019-0e8d5ff`

For me and others users on OpenWRT forum, the -CT versions has a lot of issues as well and have no Mesh support and #6937 looks to be one of these issues.

So, is not better to switch back to the non CT version ?

xback, 2023-11-28:
I tried normal ath10k last month and it didn't work for me when using 4addr-wds mode.
Everything connected but I could not send a packet over the link.

IBSS is also simply not present in original ath10k.
(It only works stable using non-encrypted links in ath10k-ct)

clebersantz, 2023-11-28:
> I tried normal ath10k last month and it didn't work for me when using 4addr-wds mode. Everything connected but I could not send a packet over the link.

what hardware do you use ?

xback, 2023-11-28:
Ipq40xx wave2 radio (mikrotik wapac and wallys dr40x9)

brada4, 2023-11-28:
C7 is wave1. ct firmware 10.1 vs 10.4 for wave2. Might not be rvery elated to eachother.

brada4, 2023-11-28:
Also you have -full- firmware which supports more features and less associations.

raenye, 2023-11-29:
Personally I use non-ct drivers on my ath10k devices.

ptpt52, 2023-11-29:
mark.

mrkiko, 2023-11-29:
Do both -CT and stock version support queue flushing?

clebersantz, 2023-11-29:
> firmware 10.1

> C7 is wave1. ct firmware 10.1 vs 10.4 for wave2. Might not be rvery elated to eachother.

Thats right, we are discussing here only the switch back for firmware ath10k 10.1 version, used in Archer C7 QCA988X for example.

ascendbeing, 2023-12-05:
> C7 V5 and V2 are mentioned in radio crash issue. If tthat is solved - lets go for it. problem with htt is it requires ct kmod while nonht works with either.

I haven't seen that one, and I use a v2 wave 1. the main difference as far as I can tell between the two is normal can achieve 200 on 40 rather than 180, and ct is really flaky with changing the coverage class. ct on main will report better rssi because it has a commit from ct that redid rssi reporting. if switching, it would make sense to port the rssi commit to non ct, and btw, that would be cool to have on 23.05 too.

it got better after some changes in the last 6-12 months but the coverage class thing is the biggest single annoyance from my usage of ath10k and ct on a v2 device (wave 1). even without changing coverage class, I often get the same warning lines that you get when you tick off the device or driver. how you know you REALLY tick it off is when it starts saying stuff about expecting 2ns not 9ns, or whatever. once you trigger that (by switching class too many times too quickly, you have to space it out by minutes, and this is ct affecting only, you can do it as much as you want on normal), the only way to remedy is unloading the module and reloading it or power cycling, afaik.

airblag, 2024-03-20:
Hi, 
I'm running a network with 9 TP-Link Archer C7 v5 running openwrt 23.05.2

I was struggling a lot tuning DAWN on it for a month now. Mostly 2.4GHz was way faster than the 5GHz band, and a lot of clients where having different problems, being kicked out of the wifi network.

After finding this ticket, I replaced the ath10k-*-ct packages with the kmod-ath10k and ath10k-firmware-qca988x ones. 

Now, my devices get a far better signal value on 5GHz (read from `iw dev radio0 station dump`). It seems also that roaming works a lot better now, and devices seems to stick to 5GHz when it's good enough, and come back to 5GHz after idling in 2.4GHz.
The download rate over 5GHz with the non-ct firmware is also something like 10 times better than with the -ct firmware:
* -ct firmware : 10-20Mbits
* -non-ct firmware : 100-300Mbits

I would also really encourage to switch the default firmware to the non-ct one since it looks like it solved all my problems at once.

brada4, 2024-03-20:
Speed is capped at CPU, you can set to ath10k_core rawmode=1 and disable switch VLANs and get up from 300-- to 400++Mbps wifi to LAN
Had to uncomment         #option max_neighbor_reports 8 for usteer and -ct to stop crashing

spamsool, 2024-03-24:
COuld you ple

> Hi, I'm running a network with 9 TP-Link Archer C7 v5 running openwrt 23.05.2
>
> I was struggling a lot tuning DAWN on it for a month now. Mostly 2.4GHz was way faster than the 5GHz band, and a lot of [...]

Could you please guide me how to switch the firmware to the non-ct one? I have 3 pieces of Archer C7 v5 with 5Ghz WiFi issues and I would like to fix it, I just need a command list to do it. Thanks :)

brada4, 2024-03-24:
```
opkg list-installed | grep ath10k-
```
You need to keep board package. 
run next on wired, if not at all possible at least in `screen` so that lost connection does not leave you without connection.
```
opkg remove kmod-ath10k-ct ath10k-firmware-qca988x-ct ; opkg install kmod-ath10k ath10k-firmware-qca988x ; echo 255 > /some/useless/led ; sleep 20 ; reboot
```

castillofrancodamian, 2024-03-24:
@brada4 Why is that LED thing?

brada4, 2024-03-24:
If connection is lost you know where it went ;-)

brada4, 2024-03-24:
Say if it is installed  you can just trigger power to reboot.....

spamsool, 2024-03-24:
Thanks a lot for quick reply. I think it runs non -ct fw now.

[   35.721062] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   35.730651] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   35.743358] ath10k_pci 0000:00:00.0: firmware ver 10.2.4-1.0-00047 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 35bd9258
[   36.410727] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08

EricFromCanada, 2024-05-04:
I have an Archer C7 v4 running Gargoyle 1.14 (based on OpenWrt 22.03) which was reporting the same `SWBA overrun on vdev 0, skipped old beacon` log messages. I haven't seen any since running the above command.

brada4, 2024-05-04:
@EricFromCanada worth indicating if it is CT or mainline firmware+driver, anyway it is just point in statistics, gargoyle is not packaged here.

EricFromCanada, 2024-05-04:
I'm aware of that; just pointing out that the C7 v4 is also affected [along with v2 and v5](https://openwrt.org/toh/tp-link/archer_c7#ghz_wi-fi_not_working_as_ap_for_c7_v2_and_v5). It was originally running `ath10k-firmware-qca988x-ct` and `kmod-ath10k-ct`; it's now running `ath10k-firmware-qca988x` and `kmod-ath10k`.

ascendbeing, 2024-05-10:
this is outta far left fyled but any clue why running an ath9k_htc AP on
ea3500 renders the 5ap no longer connectable when i have 2sta+5ap prior to
having connected the ath9k_htc as ap on Kirkwood? result is 5ap pretends
it's connectable but it's not, only 2sta+2ap works w/ ath9k_,htc connected
(usb)

brada4, 2024-05-10:
@ascendbeing better try forums, it is easier to figure things out there.

walmartshopper, 2025-01-17:
This also affects the Unifi UAP-AC-LR running v23.05.5.  Switched to non-ct and it's been stable.

anba8005, 2025-01-28:
Having bunch of Mikrotik cap-ac's, using openwrt since 22.03 release. Switched to ath10k (smallbuffers  version ofc) about same time, roaming (k/v/r + usteer) works , no problems at all. Ant there were weird random crashes with -ct firmware ..,

GeorgeSapkin, 2025-02-01:
Same for Archer C7 v2, with 24.10: tried the default CT firmware, but 5GHz was slow/unstable with it, especially while using WPA3. Switching back to the mainline resolves that.

StifflersMom, 2025-02-10:
Found this because I searched for the kernel warnings

"kernel: [  204.887730] WARNING: CPU: 0 PID: 1081 at backports-6.12.6/net/mac80211/main.c:272 0xbf14c85c [mac80211@9416e071+0x77000]"

at my DLINK DAP2610 APs (IPQ4019) running the latest stable OpenWRT 24.10.0.

I switched back from ath10k-ct to ath10k. I get 650MBit/s on 5GHz/80MHz with ath10k. With ath10k-ct I got ~550MHz with the same devices.
This is the best performance with this AP so far.

loveepy, 2025-03-28:
> 拥有一堆 Mikrotik cap-ac，从 22.03 版开始使用 openwrt。大约在同一时间切换到 ath10k（smallbuffers 版本 ofc），漫游（k/v/r + usteer）可以正常工作，完全没有问题。但是 -ct 固件出现了奇怪的随机崩溃……，

When you use ieee80211rkv, are you in AP mode or STA mode?

akshay1994, 2025-04-13:
Another data point if it helps in the decision to switch. 

I have multiple Archer C7 v2 and v4 devices. With CT firmware I was regularly experiencing wifi crashes and slowness. I had to revert to stock firmware to get things fixed.

Decided to give OpenWRT a try again. Things became much worse after enabling 802.11kvr. Logs consistently showed error messages: `ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon`.

Switched to the non-CT firmware and smallbuffers, and the issues vanished.

```
opkg remove ath10k-firmware-qca988x-ct kmod-ath10k-ct
opkg install ath10k-firmware-qca988x kmod-ath10k-smallbuffers
```

spamsool, 2025-04-14:
> Another data point if it helps in the decision to switch.
>
> I have multiple Archer C7 v2 and v4 devices. With CT firmware I was regularly experiencing wifi crashes and slowness. I had to revert to [...]

Same here with Archer C7 v2 and v5 - non-CT firmware is rock solid.

patrakov, 2025-04-19:
For me, the ct-htt variant of the driver is as stable as the non-ct one, but the non-ct one is more performant.

jabdoa2, 2025-10-31:
Same issue for me. My four Archer C7 v2 worked fine with WPA2 with the -ct-htt-full driver. However, when I switched to WPA3 they started to crash constantly when they were hit by any amount of traffic. Switching to `ath10k-firmware-qca988x kmod-ath10k-smallbuffers` fixed that and I have not seen a single crash since.