openwrt/openwrt #14101: ASUS RT-AX53U - Degraded VLAN performance on 23.05.2

r7eyes, 2023-12-01:
### Describe the bug

My network setup is a Pi 4 running OpenWrt 23.05.2 that's connected to WAN and handling routing, SQM etc, and an ASUS RT-AX53U with OpenWrt configured as a dumb AP/switch that all my devices connect to. I've setup multiple VLANs to separate LAN, Wifi, NAS etc. The VLANs are configured on the Pi 4, and on the AX53U the tags are passed through DSA bridge VLAN filtering. 

I noticed when running 23.05.2 on the AX53U that the network performance between VLANs is slower than when it is running 22.03.5. Running iperf3 between my PC and my NAS, the transfer bitrate on 23.05.2 only averages ~430 Mbps, whereas on 22.03.5 it averages ~905 Mbps. Just for fun, I tested running iperf3 between PCs on the same VLAN, and both versions averages ~940 Mbps. I did not change the OpenWrt version on the Pi 4 while running these tests.

### OpenWrt version

23.05.2

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-01:
What do you have in /etc/config/network ?
Firewall offload?
Only change 22->23 is that WAN port now has separate PHY and NIC from the now 3-port DSA switch hooked to eth0.

Shine-, 2023-12-01:
Possibly a duplicate to #13613.

Are you using both, the WAN and LAN ports for VLAN filtering? If so, then this is by design.

Due to a design change for many MT7621 boards (including yours) in order to achieve 2Gbps offloaded routing performance, VLAN filtering between the WAN port and the LAN ports will be done in software in version 23.05, not in hardware anymore.

If that change affects you negatively, you can either refrain from using the WAN port for VLAN filtering, or you can compile your own 23.05 build with the change from f1c9afd801380a05a91d979b475c76cc0a67caae undone from the DTS file of your device.

r7eyes, 2023-12-02:
> Are you using both, the WAN and LAN ports for VLAN filtering? If so, then this is by design.

Ah, yup that's how I configured it. That's unfortunate then, guess I'll stick to 22.03.5 on this device until the ability to change switch CPU ports is added. I'd rather have 1000 Mbps most of the time than 500 Mbps all of the time.

Side note, would've been nice to have this as a known issue on the wiki's release notes. I wouldn't think bridging WAN and LAN ports together for a dumb AP/switch setup is uncommon.

brada4, 2023-12-02:
Can you elaborate on /etc/config/network? I dont get what you mean by VLAN tagging? iis it like br-lan.4 or lan3@eth0.4 ?
Do you route between eth ports bridges based on VLAN-s?
Is there any difference between lan-lan and lan-wan (the vlan tagging whatever you do)
There is no reason to get 500Mbps at all ever.

r7eyes, 2023-12-03:
@brada4 here you go:
[network-pi.txt](https://github.com/openwrt/openwrt/files/13538032/network-pi.txt)
[network-ax53u.txt](https://github.com/openwrt/openwrt/files/13538034/network-ax53u.txt)

I don't do any custom routing, just firewall zones for each VLAN to isolate them and only allow certain ports/devices to communicate to other VLANs. All this is done on the Pi, the AX53U has firewall disabled. 

I'm assuming because of this, when a device is connected to the AX53U, communicating with a different VLAN goes thru the WAN port and then back again, which causes the speed degradation because I'm bridging the WAN and LAN ports together.

brada4, 2023-12-03:
you can move tagged port to original br-lan port, then only one vlan will need cpu for forwarding.
you can check:
`ethtool -k eth0 |grep vlan` ie if at lest one of offloads is active each direction, because otherwise cpu has to copy packets to fix 4 bytes alignment in the direction offload does not kick in.

same for raspii - if tag offload does not exist you may be frying cpu one way unintentionaly.

r7eyes, 2023-12-03:
> you can move tagged port to original br-lan port, then only one vlan will need cpu for forwarding.

What do you mean by that?

> you can check: `ethtool -k eth0 |grep vlan`

Here's the output on the AX53U:
```
rx-vlan-offload: on
tx-vlan-offload: on
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
```

And on the Pi:
```
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
```

So it seems like it's off on the Pi, but it says [fixed], so does that mean it can't be enabled?

brada4, 2023-12-03:
if you move tagged port to lanX then 2/3 of traffic will cross dsa switch without cpu.
now all traffic to router is forced via cpu.

yes, particular raspberry uses cpu with vlan, keep it untagged if possible, but dont change for compsrison just yet.

r7eyes, 2023-12-03:
> if you move tagged port to lanX then 2/3 of traffic will cross dsa switch without cpu. now all traffic to router is forced via cpu.
> 
> yes, particular raspberry uses cpu with vlan, keep it untagged if possible, but dont change for compsrison just yet.

I don't think that would work for my use case, unless I'm misunderstanding something. The WAN port on AX53U, which connects to eth0 on Pi, is a trunk that carries all 3 VLANs, so both need to always be tagged with all VLANs. I also would rather keep the LAN ports on AX53U untagged so devices can connect without fiddling with OS VLAN settings.

For now, 22.03.5 is working fine, and looking at the CPU usage on AX53U when doing iperf3 between my PC and NAS, I think it's already using hardware routing since CPU usage barely moved. The Pi had one core spike to 100%, but it's probably fine(?).

brada4, 2023-12-04:
Currently all traffic crosses trunk port.
If trunk port is WAN all traffic passes CPU, i.e. all devices are slow and competing for slowness....
If trunk port is LAN then 2 other LAN ports work fast as in v22, only WAN non-no-trunk is using CPU.
Sounds illogical I agree, but no other way if v24+ will not support moving 2nd CPU port away from wan port on the go.