openwrt/openwrt #14102: qnap 301w ipq8072a~~~USB3 low

suyuan168, 2023-12-01:
### Describe the bug

usb3 We can see through htop, every time we run ookla speedtest, the core0 will spike to 100% meanwhile the other 3 cores are like doing nothing which we believe what causes the bottleneck. The speed can't go beyond 700mbps.

### OpenWrt version

main

### OpenWrt release

main

### OpenWrt target/subtarget

ipq8072a

### Device

qnap 301w 

### Image kind

Official downloaded image

### Steps to reproduce

Use USB3 to connect to the hard disk to copy files via dd. Or use ax88179 or rk8125 usb3 for Ipef3 speed measurement

### Actual behaviour

usb3 is extremely slow. CPU0 is particularly high.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

suyuan168, 2023-12-01:
this exists

suyuan168, 2023-12-01:
> fix versions in your report... try irqbalance and verify interrupts land on last 2 cpu-s they are vpe-s like hyperthreads, 2 per core.

How to operate? How can I achieve this?

brada4, 2023-12-01:
install, enable in config file and start, if irqs land on different cpu-s test perf again.

brada4, 2023-12-01:
check /proc/interrupts ...

pwned-pixel, 2023-12-02:
Irqbalance is broken on IPQ807x. Following this: https://github.com/Irqbalance/irqbalance/issues/258
it is kernel related.
There are in almost every IPQ807x related device thread scripts scattered to adjust IRQ's "manually".
While I did that with simple echo commands in the past I've found recently the script from [@fif] on OpenWrt forum.
It is cutting out problematic devices also. Which do not like balancing on every device (but you can still try).

arch*)	;; # Properly balanced
ce*) 	;; # Wifi firmware crashes
edma*)	;; # Hangs wifi on high throughput
xhci*)	;; # Crashes with USB drive

I'Ve modified it a bit: https://forum.openwrt.org/t/dynalink-dl-wrx36-askey-rt5010w-ipq8072a-technical-discussion/110454/2580

While I still don't know how the queues (rx/tx) are set correctly. So like I would do (in reverse order):

`	for i in /sys/class/net/lan*/queues/rx-0/rps_cpus ; do echo "1" > "$i"; done
	for i in /sys/class/net/lan*/queues/rx-1/rps_cpus ; do echo "2" > "$i"; done
	for i in /sys/class/net/lan*/queues/rx-2/rps_cpus ; do echo "4" > "$i"; done
	for i in /sys/class/net/lan*/queues/rx-3/rps_cpus ; do echo "8" > "$i"; done

	for i in /sys/class/net/lan*/queues/tx-0/rps_cpus ; do echo "8" > "$i"; done
	for i in /sys/class/net/lan*/queues/tx-1/rps_cpus ; do echo "4" > "$i"; done
	for i in /sys/class/net/lan*/queues/tx-2/rps_cpus ; do echo "2" > "$i"; done
	for i in /sys/class/net/lan*/queues/tx-3/rps_cpus ; do echo "1" > "$i"; done`

But almost all settting e.g. rx0 and tx0 to irq 1. It would be nice if someone could give a hint here.

Another thing: I think it is good idea to disable packet steering if you've enabled it.

brada4, 2023-12-02:
It is different thing.
Check /proc/interrupts to identify heaviest device interrupts
Then in /prox/irq/55/smp_affinity_list (55 was xhci interrupt) write 2-3 e.g.
`echo 2-3 | tee -a /prox/irq/55/smp_affinity_list`

irqbalance is supposed to do exactly that automatically.

Regarding network - if card has 4 IRQs attached then you can echo zero to those files, and queue will follow irqbalance's IRQ assignments and it will be quite indifferent if you affine queues and IRQ just uses one on same CPU. Its same thing, just pulled off from different sides.

btw kernel does not balance irqs since 3.x series and relies on userland tool like irqbalance or steering script of openwrt.

edrikk, 2023-12-02:
No, irqbalance will not work for ipq807 devices as the architecture does not define the irqs properly for irqbalance to recognize.  

hmyman attempted a patch for irqbalance which allowed recognition however bugs in the architecture ssdk cause race conditions / crashes if some irq groups are separated and/or moved after initial boot / under load.

@pwned-pixel try this script - the above is also linked in the script for awareness.  edma* can be moved, but all have to be moved to the same core.  The script moves it to the last core.

https://forum.openwrt.org/t/potential-memory-leak-introduced-in-snapshot-august-sept-2023/172291/45

edrikk, 2023-12-02:
Correct, but one should not use irqbalance on this target to achieve that goal.

I suggest start with the script, and then there can be further additions if by looking at interrupts there’s large imbalance.  eg I don’t know which interrupts the USB uses as my devices don’t have USB.  After having script run on startup and reviewing the results after a couple of days, one can add the relevant interrupts to move to the appropriate core(s) and then watch for stability issues to ensure the activity didn’t cause issues with the ssdk.

brada4, 2023-12-02:
According to device page botlog irq 109 and 110
```
 echo 2 | tee -a /proc/irq/109/smp_affinity_list
 echo 3 | tee -a /proc/irq/110/smp_affinity_list 
 cat /proc/irq/109/smp_affinity_list /proc/irq/110/smp_affinity_list
```

pwned-pixel, 2023-12-03:
@brada4: Sorry that I caused a bit of confusion. I mostly wanted to make the point that irqbalance is not working on IPQ807x. The side question about the ethernet does not belong in this bugreport.

@edrikk: Thank you that you've made my point more clear than I was able to do. I don't have that insight on this topic. I will try your script.

suyuan168, 2023-12-04:
Help, can anyone give me a script to stop my USB3 from using CPU0. So many CPU0123 can achieve better performance.
I see that qsdk is running fine.

brada4, 2023-12-04:
First paste into command line, if it survives benchmarks then you can append to /etc/rc.local

https://github.com/openwrt/openwrt/issues/14102#issuecomment-1837235227

suyuan168, 2023-12-07:
In fact, the script didn't solve anything.

brada4, 2023-12-07:
Did it change CPU affinity of both USB interrupts? Ca you provide dmesg and/or /proc/interrupts?

pwned-pixel, 2023-12-08:
You would have to either adjust the scripts to your needs or use the rc.local approach @brada4 has described. In first place just execute the below commands and make a few tests if it works (3rd command here you can use to check if interrupts are moved:

` echo 2 | tee -a /proc/irq/109/smp_affinity_list
 echo 2 | tee -a /proc/irq/110/smp_affinity_list 
 cat /proc/irq/109/smp_affinity_list /proc/irq/110/smp_affinity_list`

For xhci in particular you have to assign both usb devices to the same CPU. As soon as you assign them both to a different CPU it will crash. So you can choose CPU 1, 2 or 3 instead of 0.