openwrt/openwrt #14110: Bootloop on Archer C7 v4 and v5

oldium, 2023-12-01:
### Describe the bug

I am usually building my own images for TP-Link Archer C7 v4 and v5, but they started boot-looping today. So I took latest snapshot (Fri Dec 1 04:17:32 2023), flashed it with TFTP and it boot-loops too. I suspect the latest changes to nvmem-layout.

### OpenWrt version

Snapshot (master)

### OpenWrt release

Snapshot (master)

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v4 and v5

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castillofrancodamian, 2023-12-02:
I think they have to do with Felix's latest commits. I've also noticed bootloop on my Archer C60 v1 and it appeared after incorporating those commits into my build.

KosmiK2001, 2023-12-02:
Got this problem too. I Builded it from clean source -> bootloop.
C7 v5

ascendbeing, 2023-12-03:
> @ascendbeing What is going wrong?

when I upgraded the packages on the snapshot prior to the Felix package bumps to the latest package versions of the then current snapshot, boot loop. when I upgraded to the version that contained those updates built-in, boot loop. I'm not sure if the one I tried was the 2nd to last one or the last one from ~2-3h ago. but I didn't see anything addressing the boot loops in the change log so I suspect it wasn't resolved.
I don't know how to do sophisticated debugging or I would provide more but it just infinite reboots. ie instead of normal bootup the lights stay stable eventually only to turn to every light goes on rinse repeat. tried with and without settings (with on the older snapshot, without settings on the 1st or 2nd most recent snapshot)

ascendbeing, 2023-12-03:
> Boot loop with TP-Link TL-WDR6500 too. ar71xx(ath79) generic Latest official snapshot a 2023-11-27 build of [x-wrt](https://github.com/x-wrt/build-release/releases/tag/v23.10_b202311271303) (based on openwrt) still works, so I guess the error started from the snapshot of 2023-12-01.

it started having this problem on this build: https://buildbot.staging.openwrt.org/images/#/builders/30/builds/158
the problem likely relates to the packages as I described having the issues using the build prior with packages updated to the same ones that build comes with (which came out hours apart, less than 24h apart). This set of packages (still the operative set; the builds build every package, its change set is misleading): https://buildbot.staging.openwrt.org/master/packages/#/builders/8/builds/118

oldium, 2023-12-04:
Build with 38eeefd060f70d917a68e82e0dede078446d6d64 reverted boots fine on Archer C7 v5.

oldium, 2023-12-04:
Perfect, thanks. Building...

closed 2023-12-04 by oldium