openwrt/openwrt #14111: File system check not triggered on Extroot even with e2fstools installed and check_fs enabled

Edmondena, 2023-12-02:
### Describe the bug

dmesg: -> Error message after successful mounting of extroot device /dev/sda1
-> e2fstools installed in root-system and extroot on /dev/sda1
-> check_fs set to '1' in fstab global section of root-system
-> maybe obsolete /lib/functions/fsck/e2fsck.sh seems to be never executed anymore
-> /sbin/block is not finding /usr/sbin/e2fsck
        -> maybe because relocated during remounting or relocating of extroot paths?

[   11.541295] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   11.553242]  sda: sda1 sda2 sda3
[   11.561451] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   11.900199] rt3050-esw 10110000.esw: port 4 link up
[   11.905237] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.912175] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.1: link becomes ready
**[   16.339239] block: check_filesystem: /usr/sbin/e2fsck not found**
[   16.603690] EXT4-fs (sda1): recovery complete
[   16.611223] EXT4-fs (sda1): mounted filesystem with writeback data mode. Opts: data=writeback. Quota mode: disabled.
[   16.674310] mount_root: switched to extroot
[   16.693248] urandom-seed: Seeding with /etc/urandom.seed
[   16.839009] procd: - early -
[   16.842385] procd: - watchdog -
[   17.794347] procd: - watchdog -
[   17.800515] procd: - ubus -
[   18.067250] random: ubusd: uninitialized urandom read (4 bytes read)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR902AC v3/v4

### Image kind

Official downloaded image

### Steps to reproduce

1st - install **e2fstools + usb-storage + kmod-fs-ext4 +block-mount** in root (on /dev/mtdblock4) and extroot (on usb stick /dev/sda1 partition)
2nd - **install extroot** as recommended by openwrt manual and boot into extroot to ensure it is properly working 
3rd - configure **option check_fs = '1'** in /rwm/upper/etc/config/fstab 
4th - reboot and check **dmesg** around the mounting of the extroot partition **grepping for e2fsck** and that e2fsck was properly executed
5th - properly reproduced if you find **block: check_filesystem: /usr/sbin/e2fsck not found**

### Actual behaviour

Even as everything is properly installed and can be executed manually in root and in extroot and everything was reinstalled to ensure nothing was lost or misconfigured e2fsck will never be executed.
- In contradiction to earlier Openwrt versions e2fsck today e2fsck seems to be invoked by /sbin/block which seems to be not properly working. The earlier (maybe obsolete) implementation of e2fsck in /lib/functions/fsck/e2fsck.sh is not used anymore
- dmesg/logread error message shows reason why e2fsck could never be executed:
**block: check_filesystem: /usr/sbin/e2fsck not found**

### Expected behaviour

1st - If enabled **check_fs = '1'** in /rwm/upper/config/fstab e2fsck is required to be executed
2nd - FS check required to execute and configured to run as **e2fsck -y -v -f <overlay device>**

### Additional info

**Critical Must Have Reason:** On SD card or USB storage Extroot devices only ext4 without journalling with e2fsck FS are mostly capable of capturing and fixing most errors that are occuring during power outages.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Edmondena, 2023-12-02:
What do you think about the following maybe possible solution:
1st mount /overlay with "-o ro" option to gain access to usr/sbin/xxx_fsck installation
2nd run xxx_fsck as chrooted command on the in advance bind mounded /dev/ directory extroot devices and as configured as required e.g. in /etc/config/ckeckfs
3rd after repair remount extroot with "-o rw" option and continue with extroot booting

Do I overlook here something why that cannot work?

Anyway I'd suggest that **check_fs** option is marked as **not working anymore** till it was fixed and pointing people maybe to this discussion?

obsy, 2023-12-02:
Use and build image with e2fsprogs included.

closed 2023-12-02 by hauke

Edmondena, 2023-12-02:
Sorry but if I understand you correctly and on base of my intensive research this could be a really unfortunate idea as this means to to suggest any "normal" Openwrt user to build his own image when it comes to general extroot solutions involving any NAND based storage device. On the marked NOR is not available since long anymore and mechanical HDDs are mostly no option for most OpenWrt installations. If you carefully read the above linked **official linux's conference article** we will, more than ever before, require to use fs_checks to avoid early bricking and destroying **all our OpenWrt devices** integrated NAND-Storage devices when we are not able to reliably export these silent storage errors to external storage devices. Thus this means every "NAND"-Router will earlier or later face this problem. Accordingly manual and annoying fs checks will be required repeatably if it runs OpenWrt and needs 24/7 operational reliability and 100% guaranteed facing fs problems with power outages.

Anyway thank you very much for your reply and maybe lets wait if we find other followers for this discussion that understand to break down some of their OpenWrt reliability problems to such a solution.

Are the above suggested solutions maybe possible? Then I kindly could offer to support you to continue to search for a workaround on base of the series image?

Edmondena, 2023-12-02:
Thank you for your answer, but as documented above **e2fsck is properly installed**, was reinstalled, tested and can be **guaranteed to be properly working** in either the **root** and the **extroot** environment. 

Obviously ASFAIK the problem turns out to be existing because curiously the Root-MTD-Overlay seems to be not available at anytime during Extroot boot process - which I honestly do not believe because otherwise you could never configure /rwm/upper/etc/config fstab to properly use the Extroot installation. Maybe it could be as simple as provide **/sbin/block mount** with the proper path to point correctly e.g. to /tmp/jffs2/upper/usr/sbin/e2fsck instead of the wrong /usr/sbin/e2fsck?

What is guaranteed not to be working anymore is to use **check_fs** from /etc/config/fstab of the root & extroot installation even when fs errors were detected during boot and that exceed the EXT4 included mounting recovery capabilities that occur exponential increasingly the weared your NAND storage gets and the bigger and newer your extroot NAND storages get in the future.

**Please Note:**
Thus the resulting problem will highly likely increase (as even confirmed by the above linked linux conference investigations paper) in the near future of OpenWrt and with more and more NAND based routers come into OpenWrt's focus. Bricking your routers without extroot storage gets highly reasonable as more and more bits are stored into programming of on single transistor storage cells (please see SLC,MLC,TLC,QLC...NAND technology) higher integration density followed by lower supply voltages and thus the lower and rapidly decreasing noise-signal gap gets - Be Warned!

Edmondena, 2023-12-02:
Sorry, but I cannot confirm that this is not a bug as **fs_check = '1'** as documented here: 
https://openwrt.org/docs/guide-user/storage/fstab#the_global_section
**was working all time in the past** is **now not working anymore** as e2fsck is nowadays and obviously since the last releases invoked by /sbin/block mount **with the *wrong path: /usr/sbin/e2fsck**. This is not working at all anymore during during extroot installation and rules out Extroot to be used in future application where you have mostly no alternative than using todays and future NAND based usb storage Devices like SSDs, DCXCs or any other USB-sticks.

Edmondena, 2023-12-02:
Maybe a more simple suggested solution: Give **/sbin/block mount** when e.g. invoked by /rwm/upper/etc/init.d/fstab a configurable path that simply can be configured to point to e.g. /tmp/jffs2/overlay/upper/sbin/e2fsck instead of to the erroneous path /usr/sbin/e2fsck that gets inaccessable at the moment when you move the internal mtdblock4 from /overlay to /tmp/jffs2/overlay as required during extroot switching.

What do you think about this future safe bugfix?

Edmondena, 2023-12-02:
As suggested by obsy "e2fsck should be buld-in into images" this seems to be an option only for e2fsck. For other FSs like e.g. btrfs, the FS-package itself already consumes about 1,2MB and btrfs-progs even more on top. Thus its very easy to rule out that you will ever be able to include a flexible fs check for different FSs into the base image. Anyway for e2fsck and todays use cases this seems to be a good idea as this is the most robust available Extroot-FS solution that is available today.
To regain a proper working solution for all in Openwrt available FSs I'd suggest to make the path provided in "/sbin/block mount" flexible configurable to properly point to either FS's fsck utillity.

Edmondena, 2023-12-02:
Just a simple reliability calculation for todays NAND Storage devices to make more transparent how critical this problem is for any professional solution:
Assumptions:
- Average 24/7 operating temperature (SDXC/USB) = 50°C (e.g. placed into a tiny usb storage adapter inside the USB device slot
- USB-storage (extroot) SDXC NAND device size = 128GB
- Average daily writes ~10GB (swap enabled and swapiness/vfs_cache_pressure = default)
- Max TB Written = 170TB (as provided by Transcend e.g. for 64GB High Endurance uSDXC
NAND-Flash cards
- NAND Failure Type: Statistical / Gauss
->Arrhenius Law: Thermal stretching factor FT = 2^((50°C- 20°C)/10) = 8
->Gaussian Reliability Faktor with MTBF @ 30% -> MTBF / 3 ~ 90% survival rate

=> USB-Storage Lifetime (90% Survival Rate) = 170.000GB / ( 8 x 3 x 10GB/d) = 708d @ 90%
=> Within this period of operation time the system has somehow to deal with more than 1x10^6 hidden NAND errors with most but by far not all covered by a more or less reliable manufacturers wear leveling and error correction and thus surely thousands of errors to be handled by the fs error detection and fs_check.
Please make your own picture to understand if you expect to be e2fsck as essential as I suggest for Extroot on NAND devices for our Openwrt community or not?
Please also you may search for Problems in the Raspi community that basically and 100% rely on SDXC cards as storage devices and their discussion about these topics since years now.

Edmondena, 2023-12-02:
When we brake down this to error rates I can refer you to this article which states the following with regards to NAND-Flashs silent errors: **That means that a continuous error source could introduce an error as often as every 125 MiB transferred**

https://unix.stackexchange.com/questions/269098/silent-disk-errors-and-reliability-of-linux-swap

With this and my above NAND-Flash (128GB High Endurance SDXC) lifetime estimation you can easily prove yourself the above estimation with regards to the amound of Rest-NAND-Errors.
And again - please be warned: that you could maybe never expect to reliably run any of todays Openwrt 23.xx from a NAND based Extroot installations without having an automated running e2fsck as of today's and hereby documented bug until this is fixed.

Thank you all for your really great work and support and help for fixing this!

Edmondena, 2023-12-02:
As Obsy suggested (Thanks a lot for this idea!) I build an image using the suggested firmware selector with included e2fsprogs packages and what ever was required. But finally I still were not able to fix any problem as you can see from the following captured log:

[    9.250905] mount_root: loading kmods from internal overlay
[    9.292633] kmodloader: loading kernel modules from //etc/modules-boot.d/*
[    9.305969] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
[   10.609895] jffs2: notice: (353) jffs2_build_xattr_subsystem: complete building xattr subsystem, 11 of xdatum (9 unchecked, 1 orphan) and 13 of xref (1 dead, 0 orphan) found.
[   10.626831] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
[   11.021365] rt3050-esw 10110000.esw: port 4 link up
[   11.060650] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.068627] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.1: link becomes ready
[   14.690785] **block: check_filesystem: /usr/sbin/e2fsck returned 4**
[   14.712451] **EXT4-fs (sda1): warning: mounting fs with errors, running e2fsck is recommended**
[   14.731692] **EXT4-fs (sda1): mounted filesystem with writeback data mode. Opts: data=writeback.** Quota mode: disabled.
[   14.819055] **mount_root: switched to extroot**
[   14.828256] urandom-seed: Seeding with /etc/urandom.seed
[   14.982778] procd: - early -
[   14.986153] procd: - watchdog -
[   15.851431] procd: - watchdog -
[   15.857324] procd: - ubus -
[   16.089002] random: ubusd: uninitialized urandom read (4 bytes read)

Even with this method I found no way to configure the /rom/etc/e2fsck.conf which would be required to invoke the e2fsck with verbose kernel logging and e.g. "-D -p -v -f" options or what ever would be required.

On top of that I also tried including this ugly patch into the rom boot script:
eval "`uci show fstab.extroot.uuid | sed -e's/^.*\.uuid=/e2fsck -y -v -f UUID=/'` 2>&1 | logger -t info"

Edmondena, 2023-12-03:
**Final Conclusion & Evaluation Result:**
Relying on Openwrt when installed on an Extroot based NAND-Storage device like SSD/USB/SDCard or similar may be not a good idea (with Owrt 23.05.x / Nov.2023) because of the following reasons:
- e2fsck or other may not be simply invoked to check and reliably fix any Extroot device using a standard release image (simply because e2fsprogs package) is missing in the ROM image at mount time
- The /etc/e2fsck.conf is mostly empty before overlay mounting the Extroot device and thus cannot be used for log configuration or intensive checks and bug fixing before mounting
- Installing e2fsprogs in the internal storage (like suggested above and evaluated here with mtdblock4) does not help because even the internal storage is not mounted in place before mounting the extroot and /sbin/block (which is invoking /usr/sbin/e2fsck) has a hard coded path that is point into nirvana.
- Building your own image using the above suggested firmware-selector (Thanks a lot to Obsy for this hint!) where you're required to comfortably include kmod-fs-ext4, e2fsprogs, usb-storage and optionally block-mount, tune2fs gives you back the basic standard e2fsck test
- You are maybe required using **tune2fs -c 1 /dev/sda1** (extroot NAND device) to make /sbin/block trigger the check on every boot but be aware that this check may often but not allways fix your filesystem errors and thus mounts extroot even with errors.
- Permanently monitoring of logfiles at boot time is required to repeat manually fixing your FS errors maybe and up to 1 or more per day if your using today's and future NAND storage devices.
- Ensure to always have automated backups of your important Data every time before continuing with the following ugly workaround!!
- To automate e2fsck beeing invoked for intensive and automatic filesystem checks e.g. by using the following parameters **e2fsck -p -v -D -f /dev/sda1** you require: 1st cron to check fs errors in logread; 2nd **touch /fsckforce**; 3rd disable extroot booting in /rwm/upper/etc/config/fstab; 4th reboot into internal overlay root system; 4th let rc.local check for /fsckforce and remove it; 5th run e2fsck like suggested above;  6th resore fstab with extroot enabled; 7th reboot

My scripts for this ugly workaround I provide in a following reply if its stable running...
Thanks a lot for helping and best regards!