openwrt/openwrt #14114: 6in4 tunnel is broken

philfr42, 2023-12-02:
### Describe the bug

OpenWRT 23.05.2, basic 6in4 configuration of HE tunnel (working on 19.07.3) never connects, system log shows:
`netifd: hurricane (10595): Command failed: ubus call network.interface notify_proto { "action": 0, "ifname": "6in4-hurricane", "link-up": true, "tunnel": { "mode": "sit", "mtu": 1280, "ttl": 64, "local": "192.168.x.x", "remote": "x.x.x.x" }, "keep": false, "ip6addr": [ { "ipaddr": "2001:x:x::x", "mask": "64" } ], "routes6": [ { "target": "::", "netmask": "0", "source": "2001:x:x::x\/64" }, { "target": "::", "netmask": "0", "source": "2001:x:x::\/48" } ], "ip6prefix": [ "2001:x:x::\/48" ], "interface": "hurricane" } (Unknown error)`

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

LENOVO 20EQS5LA04

### Image kind

Official downloaded image

### Steps to reproduce

create /etc/config/network:
`config interface 'hurricane'
        option proto '6in4'
        option peeraddr 'x.x.x.x'
        option ip6addr '2001:x:x:x:2/64'
        list ip6prefix '2001:x:x::/48'
`

### Actual behaviour

Start interface, Rx/Tx stay on 0, system log repeatedly shows
`netifd: hurricane (10595): Command failed: ubus call network.interface notify_proto { "action": 0, "ifname": "6in4-hurricane", "link-up": true, "tunnel": { "mode": "sit", "mtu": 1280, "ttl": 64, "local": "192.168.x.x", "remote": "x.x.x.x" }, "keep": false, "ip6addr": [ { "ipaddr": "2001:x:x::x", "mask": "64" } ], "routes6": [ { "target": "::", "netmask": "0", "source": "2001:x:x::x\/64" }, { "target": "::", "netmask": "0", "source": "2001:x:x::\/48" } ], "ip6prefix": [ "2001:x:x::\/48" ], "interface": "hurricane" } (Unknown error)` 

### Expected behaviour

Start interface, interface connects and starts 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-12-02:
set additionally: option mtu '1480'

brada4, 2023-12-02:
what do you see in `ip link` ?
btw no need to hide peer address it is publicly known 100 or so henet IP4 addresses.

brada4, 2023-12-02:
'local' address 192.168.x should be globally routable ipv4 address.

philfr42, 2023-12-03:
`option mtu` doesn't change anything
`ip link` doesn't show the interface at all
`option ipaddress` was left empty, setting it to the actual WAN address doesn't help. BTW, where does it take the WAN address when left empty in the config ?

brada4, 2023-12-04:
Probably try to use henet script, then compare if /etc/config/network section is same as entered by luci

philfr42, 2023-12-04:
Configuration is the same as the one from HE, and, as stated in the description, works perfectly on an older router with OpenWRT 19.07.3.
I sincerely doubt the issue comes from the configuration, as the error seems to come well before any connection to HE, and logs show the following sequence 40 times per second:
`Mon Dec  4 12:22:24 2023 daemon.notice netifd: Interface 'hurricane' is setting up now`
`Mon Dec  4 12:22:24 2023 daemon.notice netifd: hurricane (17383): Command failed: ubus call network.interface notify_proto { "action": 0, "ifname": "6in4-hurricane", "link-up": true, "tunnel": { "mode": "sit", "mtu": 1480, "ttl": 64, "local": "109.x.x.x", "remote": "216.66.84.46" }, "keep": false, "ip6addr": [ { "ipaddr": "2001:470:x:x::2", "mask": "64" } ], "routes6": [ { "target": "::", "netmask": "0", "source": "2001:470:x:x::2\/64" }, { "target": "::", "netmask": "0", "source": "2001:470:x::\/48" } ], "ip6prefix": [ "2001:470:x::\/48" ], "interface": "hurricane" } (Unknown error)`
`Mon Dec  4 12:22:24 2023 daemon.notice netifd: Interface 'hurricane' is now down`

brada4, 2023-12-04:
im no wiser, it was 192.168 in initial report.
try clean setup with uci, worked for me 5min ago...