openwrt/openwrt #14117: Enable multicast to unicast by default on qualcommax 5GHz?

leideno, 2023-12-03:
### Describe the bug

Spent about a week trying to set up Avahi / DNS-SD wondering why services weren't showing up on some devices but showing on others fine, worked one reboot then not the next reboot. Tried all sorts of combinations until finally reading about the 5GHz on IPQ807x devices having a well known issue with multicasting and 'Multicast to Unicast' should be enabled.

Everything has been working perfectly since, right from a fresh install.

Why would this option not be enabled by default?

### OpenWrt version

r0-60f03ee

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Arcadyan AW1000

### Image kind

Self-built image

### Steps to reproduce

Do a fresh install with Avahi and something like Samba sharing. The shares and services will display on 2.4GHz Wifi but not if you connect to 5GHz. (2.4GHz and 5GHz separate SSID's)

### Actual behaviour

No multicast or dns service discovery on 5GHz

### Expected behaviour

For it to just work

### Additional info

Build for Arcadyan AW1000 target (Qualcomm AX ipq807x device). Hosted on my github.

### Diffconfig

```text
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq807x=y
CONFIG_TARGET_qualcommax_ipq807x_DEVICE_arcadyan_aw1000=y
# CONFIG_LUA_ECO_MBEDTLS is not set
CONFIG_LUA_ECO_OPENSSL=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_avahi-daemon-service-http=y
CONFIG_PACKAGE_avahi-daemon-service-ssh=y
CONFIG_PACKAGE_avahi-nodbus-daemon=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-md4=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-fs-ksmbd=y
CONFIG_PACKAGE_kmod-fs-msdos=y
CONFIG_PACKAGE_kmod-fs-smbfs-common=y
CONFIG_PACKAGE_kmod-fs-vfat=y
# CONFIG_PACKAGE_kmod-libphy is not set
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-utf8=y
CONFIG_PACKAGE_kmod-oid-registry=y
# CONFIG_PACKAGE_kmod-phy-aquantia is not set
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-serial-qualcomm=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_ksmbd-avahi-service=y
CONFIG_PACKAGE_ksmbd-server=y
CONFIG_PACKAGE_libavahi-nodbus-support=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libdaemon=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmbedtls is not set
CONFIG_PACKAGE_libnl-core=y
CONFIG_PACKAGE_libnl-genl=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-ksmbd=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-qmi=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_mtd is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mesh-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_zlib=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-03:
It is not a supported device. Can you repeat "known issue" you refer to on a supported device?

brada4, 2023-12-06:
Check in TCPDUMP on wifi interface that after re-association IGMP subscription was sent. Then if any MDNS packet reached client.

Likely missing is to enable IGMP snooping on the br-lan (default off)

leideno, 2023-12-07:
If you enable multicast to unicast on the 5GHz Radio broadcast starts working fine, Enabling IGMP snooping didn't make any difference on my setup

brada4, 2023-12-07:
First try firmware flip. And get tcpdump -> was igmp delivered to br-lan? was any mdns crossing cable before that and after. Broadcast ping? v6?
"it does not work" is insufficient to diagnose problem obviously. even telco cable guy will not run to your house to fix internet for that.

termomaks, 2024-02-16:
Here are my latest observations with `Multi to Unicast option`.
I started this [forum thread here](https://forum.openwrt.org/t/ath11k-possible-firmware-bug-traffic-interruptions-when-a-client-disconnects-from-wlan/167400).

Currently I run the latest [@qosmio NSS WiFI branch](https://forum.openwrt.org/t/ipq807x-nss-build/148529/1428?u=sppmaster) with Commits on Feb 16, 2024.

As I was having issues with the File sharing and ping (between some of the WiFi clients) since the previous commits, I've started searching through the OpenWrt issues.
And I found this one that reminded me of [this thread and my last post](https://forum.openwrt.org/t/ath11k-possible-firmware-bug-traffic-interruptions-when-a-client-disconnects-from-wlan/167400/61) there.
I stopped using Multi to Unicast option as I didn't have any issues for some time but obviously something changed the last two weeks and I was unable to access my file shares on a Laptop via WiFi.
Now I turned on `Multi to Unicast option` and I can access file shares and ping the Laptop again.
Running several speedtests I've observed that when Multi to Unicast option was off the speed was anywhere half of the speed when `Multi to Unicast option` was turned on.

brada4, 2024-02-16:
Can you add that hyped branch link?

sppmasterspp, 2024-02-16:
> Can you add that hyped branch link?

https://github.com/qosmio/openwrt-ipq/tree/qualcommax-6.1-nss-wifi

edrikk, 2024-10-06:
Running a Xiaomi AX3600 (which is a IPQ807 device, along with 3 AX6S dumb APs (which are MediaTek.  The family has been complaining about AirPrint not working, and a laptop not seeing the Printer.

The printer sits on Wifi and can roam at times (e.g. if the AP its connected to restarts).

Investigating today I found that I was able to reproduce the issue if:
* The printer was associated to an AP other than the AX6S and the device looking for the printer was connected to the AX3600
* The printer was associated to one band of the AX3600 while the device looking for the printer was connected to another band of the AX3600

I had assumed it was a multicast issue, but stumbled upon this issue once I was able to narrow things down to above.
Turning on "Multicast to Unicast" on the AX3600 immediately resolved the issue.

Running snapshot build from 2024-09-30, and have experienced the issue through many snapshot upgrades.

kyogithub, 2024-11-15:
Same report on Arix FOSS builds for MX4300: https://github.com/arix00/openwrt-mx4300, at the moment I'm in kernel 6.6.58
I noticed that the arp table begin to lost MACs and therefor connectivity, the issue was noted when tried to do "casting" to google home or google tv devices... the "Multicast to unicast" solved the issue. Wired clients worked fine. Maybe when wireless clients renew keys some code to refresh mac o something like that is missing???...

brada4, 2024-11-15:
Chromecast is not related to multicast.

ziddey, 2024-11-15:
> Chromecast is not related to multicast.

[SSDP](https://en.m.wikipedia.org/wiki/Simple_Service_Discovery_Protocol)

BrushXue, 2025-01-15:
Now MX4300 is supported in the snapshot. I also have the same problem. If I don't enable multicast to unicast it will be OK in the beginning. But in a few hours, wireless devices can't ping each other, like having AP isolated.

gtxaspec, 2025-02-25:
Confirming this issue:

On 5ghz radios, IPv6 does not work unless `multicast to unicast` is enabled.  2.4ghz radios work fine.

Tested on Linksys MX4200 and MX4300 units, with both 24.10 and snapshot versions for investigation, same problem on both.

TurtleRecall, 2025-03-25:
Have moved to a Xiaomi AX9000 running openwrt recently (tried 23.05.5 and 24.10.0). Using it as an AP with 3x SSIDs/VLANs with chromecasts on IOT vlan. Was only intermittently able to see/connect to them from LAN vlan despite mdns repeater running on opnsense and all possible necessary firewall rules to allow multicast traffic and the various other ports chromecasts apparently required from IOT to LAN.

Note I was previously using Xiaomi AX6000 on factory firmware as LAN AP and Mercusys MR90X on openwrt 23.05.5 for the other SSIDs/VLANs. With mdns repeater enabled I saw no issues. 

On the AX9000, after weeks of cross-vlan multicast issues and troubleshooting, as soon as I enabled multicast to unicast on the IOT 5ghz SSID I was able to instantly see and connect to all of the chromecasts, so definitely impacts chromecast's ability to advertise using multicast.

gtxaspec, 2025-05-16:
still occurring for me on the MX4200/4300's as of latest master at the time of this post.

kyogithub, 2025-05-16:
I saw the patch on the snapshot https://github.com/openwrt/openwrt/tree/main/package/kernel/mac80211/patches/ath11k , not sure if it is already on master branch.

dwyeraidan, 2025-05-24:
Same issue still on an AW1000. Surprised multicast to unicast isn’t default. Glad my network is working now.

bogardon, 2025-06-01:
@leideno  Where did you originally read about this issue? Thanks for creating this issue though, fixed my mdns issue right away :)