openwrt/openwrt #14118: Starting odhcp6c as a daemon results in a continuously increasing recv queue of raw sockets.

Horbivores, 2023-12-03:
### Describe the bug

When starting odhcp6c as a daemon using the command "odhcp6c -s /etc/odhcp6c.script -p /var/run/eth0_odhcp6c.pid -P 0 -t 120 eth0", the DHCPv6 server sends RA packets every 10 seconds. 
However, it was observed that the raw socket's receive buffer keeps increasing continuously.
After investigating the code, an error was found: the I/O signal was bound before starting the daemon, causing the daemon to not receive the I/O signal. Consequently, the raw socket's recv queue kept growing indefinitely.

I have observed that this bug seems to be persistent.

This patch is my solution, which places the initialization after creating the daemon process.

diff --git a/src/odhcp6c.c b/src/odhcp6c.c
index 4a6d7bd..4d3c31e 100644
--- a/src/odhcp6c.c
+++ b/src/odhcp6c.c
@@ -421,20 +421,12 @@ int main(_unused int argc, char* const argv[])
 	signal(SIGUSR1, sighandler);
 	signal(SIGUSR2, sighandler);
 
-	if ((urandom_fd = open("/dev/urandom", O_CLOEXEC | O_RDONLY)) < 0 ||
-			init_dhcpv6(ifname, client_options, sk_prio, sol_timeout) ||
-			ra_init(ifname, &ifid, ra_options, ra_holdoff_interval) ||
-			script_init(script, ifname)) {
-		syslog(LOG_ERR, "failed to initialize: %s", strerror(errno));
-		return 3;
-	}
-
 	if (daemonize) {
 		openlog("odhcp6c", LOG_PID, LOG_DAEMON); // Disable LOG_PERROR
 		if (daemon(0, 0)) {
 			syslog(LOG_ERR, "Failed to daemonize: %s",
 					strerror(errno));
-			return 4;
+			return 3;
 		}
 
 		if (!pidfile) {
@@ -449,7 +441,15 @@ int main(_unused int argc, char* const argv[])
 		}
 	}
 
-	script_call("started", 0, false);
+    if ((urandom_fd = open("/dev/urandom", O_CLOEXEC | O_RDONLY)) < 0 ||
+            init_dhcpv6(ifname, client_options, sk_prio, sol_timeout) ||
+            ra_init(ifname, &ifid, ra_options, ra_holdoff_interval) ||
+            script_init(script, ifname)) {
+        syslog(LOG_ERR, "failed to initialize: %s", strerror(errno));
+        return 4;
+     }
+
+    script_call("started", 0, false);
 
 	while (!signal_term) { // Main logic
 		odhcp6c_clear_state(STATE_SERVER_ID);

### OpenWrt version

r7258-5eb055306f

### OpenWrt release

18.06.1

### OpenWrt target/subtarget

mvebu/cortexa53

### Device

Globalscale Marvell ESPRESSOBin Board

### Image kind

Official downloaded image

### Steps to reproduce

using the command "odhcp6c -s /etc/odhcp6c.script -p /var/run/eth0_odhcp6c.pid -P 0 -t 120 eth0"

The issue is more noticeable when the server sends RA packets at shorter intervals.

### Actual behaviour

The receive buffer of the raw socket keeps increasing when monitoring with the command "netstat -anp | grep "odhcp"".

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Horbivores, 2023-12-03:
> can you fix the formatting or PRify this? if it's a PR you usually can put .patch at the end of the url and get a patch from it. gh put a bunch of bullets there which is kinda wacky

I submitted a bug for odhcp6c, but it seems that I shouldn't have submitted it here. OpenWrt does not directly include its source code.

closed 2023-12-03 by Horbivores

ascendbeing, 2023-12-03:
> I submitted a bug for odhcp6c, but it seems that I shouldn't have submitted it here. OpenWrt does not directly include its source code.

good on ya but ya only included your proposed patch here not on the issue over there. I am telling you this here so your probably more tailored report can be pristine and we can point this out on the "practice" report.
I would also still suggest doing a PR might make more sense

Horbivores, 2023-12-03:
PR： https://github.com/openwrt/odhcp6c/pull/80