openwrt/openwrt #14120: umdns page fault at startup on MT7621

s3phy, 2023-12-03:
### Describe the bug

Hello,

On MT7621 devices (reproducible on Xiaomi Mi Router 3G and Netgear R6850 at least), umdns does not start properly from a reboot or cold boot.

I see the following in the system log from boot:

```
Sun Dec  3 07:40:29 2023 kern.info kernel: [   30.989939] do_page_fault(): sending SIGSEGV to umdns for invalid write access to 00000004
Sun Dec  3 07:40:29 2023 kern.info kernel: [   30.998392] epc = 77db662b in libubox.so.20230523[77db4000+1f000]
Sun Dec  3 07:40:29 2023 kern.info kernel: [   31.004621] ra  = 77db9a0f in libubox.so.20230523[77db4000+1f000]
```

In this state, the umdns process appears to still be running:

```
root@MiRouter3G:~# ps w | grep umdns
 2669 root      2912 S    {umdns} /sbin/ujail -t 5 -n umdns -S /etc/seccomp/umdns.json -u -l -- /usr/sbin/umdns
 2670 root      1852 S    /usr/sbin/umdns
 3370 root      1376 S    grep umdns
```

But there is no mDNS traffic/communication from the router (sniffed from a computer on the LAN with Wireshark while doing tests)

Restarting the umdns process makes it work fine. I have had uptimes of several days without any problems after manually restarting umdns after startup.

This behaviour did not happen with 23.05.0, it's a new bug in 23.05.2.

umdns version 2023-10-19-d45c443a-5 installed from luci opkg.

It could be specific to the mt7621 target, I cannot reproduce the bug on a Buffalo WZR-HP-G302H A1A0 (ath79/generic).

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Steps to reproduce

1. Clean install of OpenWrt 23.05.2 on a MT7621 device (I have the bug on both a Xiaomi Mi Router 3G and on a Netgear R6850)
2. Install package umdns from opkg. The package installs and runs fine.
3. Restart device, either from a reboot from software or a power cycle.

### Actual behaviour

umdns does not work at startup. Its process appears to be running, but no mdns traffic is seen from the router. The following messages appear in the system log:
```
Sun Dec  3 07:40:29 2023 kern.info kernel: [   30.989939] do_page_fault(): sending SIGSEGV to umdns for invalid write access to 00000004
Sun Dec  3 07:40:29 2023 kern.info kernel: [   30.998392] epc = 77db662b in libubox.so.20230523[77db4000+1f000]
Sun Dec  3 07:40:29 2023 kern.info kernel: [   31.004621] ra  = 77db9a0f in libubox.so.20230523[77db4000+1f000]
```

### Expected behaviour

umdns starts properly.

### Additional info

Before opening this bug, I tried resetting my configuration, starting from a clean, unconfigured OpenWrt 23.05.2.
I use OpenWrt on "dumb AP" devices, after the configuration reset I only configured a static IP address, gateway, and IPv6 as dhcp-v6 client. I then installed umdns from opkg, it started properly on first run following the installation. I then rebooted the router, and it did the page fault on startup, as I had observed previously with my configuration. Restarting it manually would make it work, for as long as the device is not restarted.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ChaitanyaPramod, 2024-01-19:
Hi, I have the same issue on two MT7621 devices (D-Link DIR-2640 A1 and D-Link DIR-2660 A1), both on OpenWrt 23.05.2 r23630-842932a63d.

Similar symptoms as OP, process appears to be running but doesn't discover other hosts. Only restarting the process makes it work again.

```
[   26.981973] do_page_fault(): sending SIGSEGV to umdns for invalid write access to 00000004
[   26.998634] epc = 77d0d62b in libubox.so.20230523[77d0b000+1f000]
[   27.010962] ra  = 77d10a0f in libubox.so.20230523[77d0b000+1f000]
```

Wanted to add another data-point to the bug report.

webknjaz, 2024-02-29:
FTR This is also happening to my TOTOLINK X5000R hostspots under 23.05.2 (was fine on 23.05.0). Same MT7621 chip too.
```console
Thu Feb 29 00:11:38 2024 daemon.warn dawn: Failed to lookup ID!
Thu Feb 29 00:11:38 2024 daemon.warn dawn: Failed to lookup ID!
Thu Feb 29 00:11:38 2024 kern.info kernel: [294081.681885] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068
Thu Feb 29 00:11:38 2024 kern.info kernel: [294081.690249] epc = 0040c11b in dawn[400000+13000]
Thu Feb 29 00:11:38 2024 kern.info kernel: [294081.695028] ra  = 0040c11b in dawn[400000+13000]
```

ynezz, 2024-02-29:
Would it be possible to [test snapshot build](https://openwrt.org/docs/guide-quick-start/developmentinstallation)? That way we can easily test if that issue was fixed already (we would need to backport fixes for 23.05 which we probably should anyway) or this needs to be debugged further.

s3phy, 2024-03-02:
> Would it be possible to [test snapshot build](https://openwrt.org/docs/guide-quick-start/developmentinstallation)?

It appears fixed in snapshot! I've been running snapshot since yesterday on mi Xiaomi Mi Router 3G and umdns works fine immediately after a cold boot or a reboot.

A few weeks ago I also tried going back to 23.05.0 and... I was surprised to see that the bug was there! Same behaviour as with 23.05.2. I guess I never noticed it at the time.

ynezz, 2024-03-08:
@webknjaz  FYI, following:

> do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068

means, that `dawn` is crashing, so not related to `umdns` and this issue:

> do_page_fault(): sending SIGSEGV to umdns for invalid write access to 00000004

webknjaz, 2024-03-08:
Ah, thanks for pointing out! I haven't been able to find any time to try out the snapshot but I'm watching the discussions. For now, reverting to 23.5.0 works.

ynezz, 2024-03-11:
So the fixes landed and [umdns_2024-03-04-7c675979-1_mips_24kc.ipk](https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/base/umdns_2024-03-04-7c675979-1_mips_24kc.ipk) seems to be already available, so can you try to upgrade `umdns` package on your 23.05.2 systems and check if it is really fixed in `umdns` version `2024-03-04` ?

s3phy, 2024-04-07:
> can you try to upgrade `umdns` package on your 23.05.2 systems and check if it is really fixed in `umdns` version `2024-03-04` ?

I moved both my routers back to 23.05.2, clean "sysupgrade" install without keeping my settings (trying to be as neutral as possible), installed umdns on both: works perfectly! No more segfault at startup.

I upgraded both routers to 23.05.3 when it came out about a week later, also with umdns 2024-03-04, it's been around two weeks now, everything has been working flawlessly.

It looks to me like this bug is fixed :)

closed 2024-04-07 by s3phy