openwrt/openwrt #14121: MR8300: LAN ports speed drop to 100 Mb/s

badulesia, 2023-12-03:
### Describe the bug

When connecting a 100 Mb/s device to WAN, linkspeed on LAN ports drops to 100Mb/s whereas it should remain at 1 Gb/s.

### OpenWrt version

r24557-643a4adf9a

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Disconnect WAN port. Perfom a huge file transfer between two computers on LAN. Monitor the transfer speed (110 MB/s expected). Than connect a 100 Mb/s device on WAN. File transfer speed drops to 11 MB/s, indicating that LAN ports are acting at 100 Mb/s. Disconnect WAN port, and than file transfer speed rises again to 110 MB/s. LAN ports are back to 1 Gb/s. Luci status page is still indicating 1Gb/s even when the drop occurs.

### Actual behaviour

LAN ports speed drops to 100 Mb/s when WAN is at 100 Mb/s.

### Expected behaviour

LAN ports should remain at 1 Gb/s

### Additional info

Perform perfectly on 23.05.2 with the same config files. LAN ports speed remains at 1Gb/s even with a 100 Mb/s on WAN.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

badulesia, 2023-12-03:
Can you elaborate please? I'm not sure to understand. Furthermore I don't have acces to the device anymore (it was in a relative's home). I made a test from time to time with a snapshot, than I restore back a stable.

brada4, 2023-12-03:
> page is still indicating 1Gb/s 

vs 

>  linkspeed on LAN ports drops to 100Mb/s 

No idea what you are reporting. So I asked for clarification.

badulesia, 2023-12-04:
Unfortunalty I'm not able to acces on the device anymore.

badulesia, 2023-12-04:
I have check link speed with eth tool before posting : ports are still reported to 1 Gb/s. I can acces the device in the future, by  don't know when (a matter of weeks).

brada4, 2023-12-04:
if somebody else faces similar problem before that they may pick up , no rush though, if it is that bad it will be soon ;-)

badulesia, 2024-02-18:
Hi.
I'm back on this device. I have installed the latest snapshot from today: still same issue.
Using ethtool I checked that LAN ports speed is displayed at 1Gb/s. Nevertheless the bandwidth (LAN to LAN file copy) is that of 100Mb/s. As soon as I disconnect the WAN port, bandwidth returns to expected 1Gb/s.

brada4, 2024-02-18:
Reiterating: 
ethtool X reports good speed
Throughput is low
what does ethtool -S X report with regard to pauses errors fails drops?

brada4, 2024-02-18:
Could be related https://github.com/openwrt/openwrt/issues/14562

badulesia, 2024-02-18:
ethtool -s on LAN1 (main PC) while idling.
```
tx_packets: 2591
     tx_bytes: 12692366
     rx_packets: 4385
     rx_bytes: 404097
     RxBroad: 29
     RxPause: 37
     RxMulti: 221
     RxFcsErr: 0
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 3630
     Rx128Byte: 457
     Rx256Byte: 85
     Rx512Byte: 83
     Rx1024Byte: 95
     Rx1518Byte: 8265
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 12803860
     RxBadByte: 2368
     RxOverFlow: 0
     Filtered: 1
     TxBroad: 20
     TxPause: 0
     TxMulti: 48
     TxUnderRun: 0
     Tx64Byte: 1056
     Tx128Byte: 1234
     Tx256Byte: 236
     Tx512Byte: 164
     Tx1024Byte: 146
     Tx1518Byte: 16755
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 25655122
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 12328
     TXUnicast: 19523

```

and while copying huge file from LAN2

 ```
tx_packets: 139457
     tx_bytes: 217554231
     rx_packets: 51678
     rx_bytes: 2849990
     RxBroad: 30
     RxPause: 37
     RxMulti: 222
     RxFcsErr: 0
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 49340
     Rx128Byte: 647
     Rx256Byte: 1046
     Rx512Byte: 454
     Rx1024Byte: 131
     Rx1518Byte: 8290
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 16101027
     RxBadByte: 2368
     RxOverFlow: 0
     Filtered: 1
     TxBroad: 179
     TxPause: 0
     TxMulti: 394
     TxUnderRun: 0
     Tx64Byte: 47097
     Tx128Byte: 1905
     Tx256Byte: 1868
     Tx512Byte: 907
     Tx1024Byte: 539
     Tx1518Byte: 151715
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 234343777
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 59619
     TXUnicast: 203458
```

now with WAN unplunged
```
 tx_packets: 914904
     tx_bytes: 1391402631
     rx_packets: 221938
     rx_bytes: 10839897
     RxBroad: 30
     RxPause: 37
     RxMulti: 222
     RxFcsErr: 0
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 218440
     Rx128Byte: 673
     Rx256Byte: 2128
     Rx512Byte: 460
     Rx1024Byte: 166
     Rx1518Byte: 8301
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 27155614
     RxBadByte: 2368
     RxOverFlow: 0
     Filtered: 1
     TxBroad: 223
     TxPause: 0
     TxMulti: 401
     TxUnderRun: 0
     Tx64Byte: 216346
     Tx128Byte: 1973
     Tx256Byte: 2971
     Tx512Byte: 918
     Tx1024Byte: 567
     Tx1518Byte: 927092
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 1422418631
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 229879
     TXUnicast: 1149243
```

badulesia, 2024-02-18:
Some extra infos I have experimented.
WAN is connected to a bridged modem with a 100 Mb/s port. So I can't change this. 
Instead of the modem, I put a dummy computer with 1Gb/s link on WAN. File transfers are now good. If I force speed link to 100Mb/s than the bug re-occured.

brada4, 2024-02-18:
nothing wrong in stats, you used tcp with sack for test
there are pause frames but not increasing during test

if you force 100mbos on other port - does it suffer too?
what is logged on other ports - like disconnect or smth?

badulesia, 2024-02-19:
Same thing occures if I force 100Mb/s with the dummy computer.
Other LAN ports are disconnected.

DragonBluep, 2024-02-19:
It sounds like qualcomm phy driver issue. Maybe @Ansuel and @robimarko can help.

robimarko, 2024-02-19:
Can you post your `/etc/config/network`?

Also, post what ethtool reports

badulesia, 2024-02-19:
I used a fresh install : set up password, add luci and luci-app-advanced-reboot manually with ssh.
`/etc/config/network` is the default one. Here it is anyway. MAC adresses are redacted.

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdba:e360:6610::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config device
	option name 'lan1'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan2'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan3'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config device
	option name 'lan4'
	option macaddr 'e8:9f:80:18:5c:2f'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'wan'
	option macaddr 'yy:yy:yy:yy:yy:yy'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'
```

ethtool reports

```
Settings for lan1:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: external
        Auto-negotiation: on
        MDI-X: on (auto)
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes

```
```
Settings for lan2:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Half 1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes

```
```
Settings for wan:
        Supported ports: [ TP MII FIBRE ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                1000baseX/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                1000baseX/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 4
        Transceiver: external
        Auto-negotiation: on
        MDI-X: on (auto)
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes
```

badulesia, 2024-04-17:
Issue is still present with OpenWrt SNAPSHOT r25940-8c3dfdc46b / LuCI Master 24.100.29615~1181915 under kernel 6.1.86
Openwrt is installed without keeping settings. Nothing is configure. WAN is plugged to a 100 Mb/s modem in bridge mode.
Plug 2 computer to LAN with wires and perform a file transfer between them : expected speed is 1 Gb/ (110 MB/s)
When WAN is plugged, traffic drops to 11 MB/s. When WAN is unplugged, traffic raises to 50 MB/s only, with a huge load average (0.80).
Same tests under 23.05.3 are running perfectly (speed about 100MB/s and load average < 0.15).

lukakama, 2024-05-29:
Hi, I have the same or a very similar behavior on a FRITZ!Box 7530 running latest snapshot (r26445-4f078bf377).

I have a PC and a NAS attached to the router. When both are turned on, both are linked at 1Gb/s and I get 1Gb/s throughput on both download and upload between both devices and router (tested using iperf3), but when PC is turned off, its link speed is re-negotiated to 10Mb/s, and when its happen the NAS, still reported as linked at 1Gb/s, gets its download speed limited to 10Mb/s, meanwhile the upload remain at 1Gb/s. 

As soon the PC is turned on or get the NIC disconnected, the NAS download speed returns at 1Gb/s.

badulesia, 2024-05-29:
Hi.
Very interesting ! I thought I would be alone on this one.
In the meantime I have also performed tests with an early kernel 6.6  custom build and still have the issue. Do you also have high CPU consumption while tranfering on LAN? Upload is still high while download drops? Weirder ;)

I managed somehow some kind of workaround by putting a switch between the router and the bridged modem, but this is still a workaround.

lukakama, 2024-05-29:
> Hi. Very interesting ! I thought I would be alone on this one. In the meantime I have also performed tests with an early kernel 6.6 custom build and still have the issue. Do you also have high CPU consumption while tranfering on LAN? Upload is still high while download drops? Weirder ;)

Just checked, and yes, with 1Gb/s throughput I have high CPU usage (2 of 4 CPUs are at ~100%, mostly on system and softirq) both in download and upload.

Instead, when the upload is capped at 10Mb/s, the CPU load is quite low.

Nothing change enabling or disabling software and hardware offloads.

badulesia, 2024-05-30:
With kernel 5.15 (OpenWrt 23.05), LAN to LAN transfers don't consume high CPU usage, and the bandwioth is what to expect.
Since kernel 6.1 (and so 6.6) it is the case. Every kernels use DSA: DSA implies that each packet is analysed. I don't understand why 5.15 don't suffer from the bandwith drop while 6.1 and 6.6 do. Something deep has change since 6.1, that impact the bandwith and the CPU consumption. 

I don't remember the impact of SW offloading with 23.05. HW offloading is not implemented, albeit the CPU is capable.
I'm using kernels 6.1 and 6.6 on others targets (x86, mt7621, ath79à and LAN to LAN tranfers occur correctly. So I guess that the issue is specific to ipq40xx target.

badulesia, 2024-07-04:
Updated status
I have retested using OpenWrt SNAPSHOT r26844 (Kernel 6.6.35) on 4/7/2027.
Speed drop doesn't occur anymore when WAN is at 100 Mb/s. Somehow this has been fixed by a commit.
Meanwhile LAN to LAN transferts does reach 115 MB/s as expected, but only 108MB/s with huge CPU load.

closed 2024-07-04 by badulesia