openwrt/openwrt #14128: CONFIG_PCIE_PTM needs to be enabled

MoetaYuko, 2023-12-04:
### Describe the bug

The igc driver expects PCIe PTM to be enabled https://github.com/torvalds/linux/commit/1b5d73fb862414106cf270a1a7300ce8ae77de83. Currently, this is disabled for openwrt, and a warning is raised on igc probing:
```
[    3.826371] igc 0000:06:10.0: PCIe PTM not supported by PCIe bus/controller
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-04:
I dont think you get any microsecond accuracy inside a VM. Have you tried rebuilding with parameter enabled?

MoetaYuko, 2023-12-05:
No, because I didn't observe any downsides apart from the warning. However, it still makes sense to enable PTM because OpenWrt could run on bare-metal hardware.

raenye, 2023-12-13:
Did you self-compile a version with this flag enabled? does it work?

brada4, 2023-12-13:
openwrt is not popular in places with ptp on the wire and 100gbe netcards to test it.