openwrt/openwrt #14130: New commit for ubox breaks OpenWrt build

Ykidia, 2023-12-04:
### Describe the bug

This commit https://git.openwrt.org/?p=project/ubox.git;a=commit;h=c08709cceb554cba02c935d1442f6a042fe6b2a8
breaks OpenWrt build out of nowhere:
```
[1/5] Building C object CMakeFiles/logd.dir/log/logd.c.o
FAILED: CMakeFiles/logd.dir/log/logd.c.o 
/home/user/sources/openwrt/mir3x/staging_dir/toolchain-mipsel_24kc_gcc-13.2.0_musl/bin/mipsel-openwrt-linux-musl-gcc   -O3 -pipe -march=1004kc -mdsp -mmt -fomit-frame-pointer -mframe-header-opt -fno-caller-saves -fno-plt -fomit-frame-pointer -mframe-header-opt -Wno-error -fhonour-copts -msoft-float -fmacro-prefix-map=/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc=ubox-2023-11-30-c08709cc -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG   -Os -ggdb -Wall -Werror --std=gnu99 -Wmissing-declarations -MD -MT CMakeFiles/logd.dir/log/logd.c.o -MF CMakeFiles/logd.dir/log/logd.c.o.d -o CMakeFiles/logd.dir/log/logd.c.o -c /home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c
In file included from /home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c:30:
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/syslog.h:45:31: error: 'struct udebug_ubus' declared inside parameter list will not be visible outside of this definition or declaration [-Werror]
   45 | void log_udebug_config(struct udebug_ubus *ctx, struct blob_attr *data,
      |                               ^~~~~~~~~~~
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c: In function 'main':
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c:266:9: error: implicit declaration of function 'udebug_ubus_init'; did you mean 'udebug_buf_init'? [-Werror=implicit-function-declaration]
  266 |         udebug_ubus_init(&udebug, &conn.ctx, "log", log_udebug_config);
      |         ^~~~~~~~~~~~~~~~
      |         udebug_buf_init
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c:280:9: error: implicit declaration of function 'udebug_ubus_free'; did you mean 'udebug_buf_free'? [-Werror=implicit-function-declaration]
  280 |         udebug_ubus_free(&udebug);
      |         ^~~~~~~~~~~~~~~~
      |         udebug_buf_free
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c: At top level:
/home/user/sources/openwrt/mir3x/build_dir/target-mipsel_24kc_musl/ubox-2023-11-30-c08709cc/log/logd.c:35:27: error: storage size of 'udebug' isn't known
   35 | static struct udebug_ubus udebug;
      |                           ^~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

Earlier there also were stupid "warnings as errors", but now it is cleanly errors - simply forgotten corresponding includes or defines, I guess. And a follow-up question, if enabling this "udebug support", am I need also selecting corresponding libraries/daemons packages in OpenWrt?

P.S. The same bad with procd by just right the same reason. Is there a big need in this udebug?

### OpenWrt version

current snapshot

### OpenWrt release

current snapshot

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3 Pro, Xiaomi Mi Router 3G

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ykidia, 2023-12-04:
> Is this from a clean repository?
Yes, but I made some own settings in menuconfig, added some 3rd-party packages and also some own patches to existing packages. But these modified/out-of-tree packages compiling after all base packages including ubox and procd.
Please note that I already said that ubox (and procd) builds OK on previous commit with the same my own changes to config, packages set and additions/modifications.
Thank you!

mpratt14, 2023-12-04:
change version to "SNAPSHOT"
change release to the revision string `cat /etc/openwrt_version`

to satisfy the bot

guidosarducci, 2023-12-06:
CC @nbd168

Ykidia, 2023-12-11:
After some investigations I found that dir "libs" exists partially in two places inside OpenWrt root. Don't know how did it happen, but, at last, this is definitely my bad. Sorry!

closed 2023-12-11 by Ykidia

mpratt14, 2023-12-11:
I recommend `make dirclean` before your next build