openwrt/openwrt #14135: OpenWrt Firmware-Selector bootscript start index too late for extroot environment e2fsck

Edmondena, 2023-12-04:
### Describe the bug

For small memory devices and tons of applications that rely on most common USB NAND storage devices using OpenWrt in Extroot is missing any opportunity to run e2fsck with customizable flags.
Openwrt has introduced **/sbin/block mount** in standard OpenWrt releases but possibly may have overlooked the lack of any possibility to run e2fsck or any other FS checks required **before** mounting the extroot NAND storage device. Even the customizable boot script in the firmware selector lacks the option to specify something like:
START=xx
STOP=xx
in the head of the script to be interpreted to e.g. inject an e2fsck command before the final Extroot is mounted even though a post installed /tmp/jffs_data/upper/usr/e2fsck would be available.
As far as I could find out, these circumstances make it mostly uncomfortable/impossible for normal users on standard releases to deal with often FS corruptions that today's and tomorrow NAND device inject by nature of SLC,MLC,TLX...technologies and future's increasing cell densities.

### OpenWrt version

23.05.0

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link TL-WR902AC v3/v4

### Image kind

Official downloaded image

### Steps to reproduce

1. Install Openwrt with kmod-fs-ext4, e2fsprogs, usb-storage, block-mount and tune2fs
2. Copy OpenWrt into Extroot environment on an Usb-Stick or SDXC-Highspeed Adapter and make it run
3. run **tune2fs -o journal_data_writeback,user_xattr,acl -c 1 -o journal_data_writeback /dev/sda1** to enable highest performance swapfile access
4. In /etc/config/fstab set global **option check_fs ='1'** and mount **option options 'noatime,nodiratime,data=writeback'**for extroot's best performance at lower NAND wear
5. Boot into extroot Openwrt and run **dmesg | grep e2fsck** showing **block: /usr/sbin/e2fsck not found**
-> As far as I know this shows you 2 things:
a) e2fsck probably could not never be executed by Openwrt's boot environment because at the time when Extroot NAND-USB-storage-device gets mounted, the internal, high reliable NOR-Flash boot device with the preinstalled e2fsck utility (included in e2fsprogs package) is not accessible.
b) There maybe no solution existing from any of the standard Openwrt's image to run e2fsck at all without building you own OpenWrt image and even when you build your own you get no information what went good or wrong and (because very early invoked you're out of any opportunity to configure this check and provide you with log results.

**First round try to fix** it using Openwrt's great firmware-selector and adapt the package list injecting **e2fsprogs, block-mount, usb-storage, kmod-fs-ext4**, make e2fsck run and inject swap on SDXC usb storage device as continous error source.
BTW: The following may make linux run into "normal" FS corruption errors per daily repetitions or more often either by swap failures or unintended watchdog hardware shutdowns triggered in overload conditions.

6.1 Enable swap like documented in the Openwrt's extroot installation guide e.g. easiest as a swap file e.g. in /overlay/swap
6.2. Ensure to check swap is beeing used by **swapon -s** in extroot environment
6.3. Install e.g. adguardhome and add some filters until in luci-statistics with processes plugin install you see starting a high volatitly in resident memory usage. But take care as if adding too much filters you run adguardhom out of memory and kill it.
6.4 Check heavy increasing storage wear using **awk '{ print $3"\t"$10 / 512 }' /proc/diskstats** and run "logread -f &" to see sda I/O and/or FS errors popping up that you now can try to fix by rebooting and e2fsck'ing resulting in something like "mounting a filesystem with errors"

**Second round try to fix** by using Openwrt's firmware-selector's boot script
6.1 include a command line like e.g. "e2fsck -p -v -f -D /dev/sda | tee -a /tmp/jffs_cfg/e2fsck.log | logread -t debug" to see e2fsck outputs and results in the later debug logging.

-> No progress to see any early invocation of e2fsck what ever I try to make it run. No chance it will never executed.

### Actual behaviour

Openwrt's great Firmware-Selector's so called "early" boot script is executed much too late to run proprietary e2fsck or any other Fscheck in advance of mounting the Extroot environment. This makes commonly use of todays NAND-Usb-storage-devices highly unreliable for any router application that is relying on swap memory or any data storage depending use case like e.g. adguardhome, radicale adress database, caldav calendar database, email server when facing more or less often power outages, overload or extensive swap operation. 

### Expected behaviour

Enable Openwrt for reliably running at 24/7 from Extroot-NAND-usb storage devices.
Ensure FS consistency at every Extroot boot of an Extroot Openwrt installation
Ensure FS consistence even after power outages that are normal in non continuous Offgrid applications with solar power, RV-car battery power, overload conditions of small memory devices and FS-inconsistency kernel crashed.

**Maybe simplest, most flexible Solution**
Write "START=09" on top into the Firmware-Selector's boot script to mount and run "e2fsck ...." from internal root storage device much earlier and right before the Extroot NAND-Usb storage device gets mounted (just before the first and rom invoked /etc/init.d/fstab execution.

**Perfect Solution (without: Firmware-Selektor** All this by simply following these straight forward steps: Install e2fsck and required usb-storage package in internal root storage, configure "check_fs = '1'" and configure /etc/e2fsck.conf respectively /etc/config/e2fsck to invoke "/usr/sbin/e2fsck -p -v -f -D /dev/sda1" to check and forced repair any possible e2fs error.

### Additional info

I call the above error rates "normal" because this is the nature of todays high density NAND devices and "normal" statistic error correction faults that cannot be properly handled by the device manufacturers wear leveling and correction algorithms e.g. in combination with weak Swap-FS that misses any integrity checking and thus makes Openwrt's embedded Linux stability and reliability highly sensible to NAND-Flash errors even when using new and formerly unused devices from stock. 

These here integrated manufacturer correction algos make or inject errors statistically while trying to correct e.g. multiple read or checksum errors erroneously wrong among many other silent bit flip errors and punching holes into the FS-Structure that could only (if ever) be handled by FS-checks.

These Errors are by far rare and throwing away todays high density NAND-devices that show these kind of errors means throwing away new purchased devices. 

As it is impossible to check the Extroot-FS on a live running Openwrt installation it is essential to make it easily possible to check any NAND-Device repeatedly on every boot to:
1. Shorten the error free consistency period to a much shorter period (e.g. 1day)
2. Do not let NAND-FS errors accumulate to multiple errors per file so e2fschk can fix high
reliably fix them
3. Ensure Openwrt to run on a mostly clean Extroot boot environment and to expand
memory restriction on base of a rugged FS base.
4. Enable Openwrt to run 24/7 even on the growing Extroot NAND-Router devices without destroying or early wearing out and bricking the internal memory 

To simulate or enforce generation of high error rates it is an advantage to use a "high performant" U3/10 SDXC card and a card reader that shows around 10-15MB/s (hdparm) read timings (HighSpeed USB) and around 50MB/s cached read timings to see errors and how the standard e2fsck configuration can handle and detect errors and fixes them or not. Sadly about 2-3 times per week the defaul e2fsck can't correct the errors and while writing to about 10-15GB swap data to disc it leaves Openwrt alone with mounting a FS with errors that you most probably find even in every Extroot installation which is running for weeks. 

p.s. The above is by far an abnormal use case while my efforts focus on creating an ultra low power RV router powered 24/7 from my cars battery. This small light device is used with dnscrypt and adguardhome together with radicale for calendar, contacs synchronisation and lighthttpd with dokuwiki for travel planning, documentation and gps tracking.
Thus it seem very similar to any Off-Grid application like e.g. Whether-Stations, Solar-powered relays or many other that rely on a stable Openwrt installation with today's and tomorrows commonly used NAND-devices where you cannot or don't want to maintain errors on a weekly basis.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-04:
Duplicate of https://github.com/openwrt/openwrt/issues/13968

Edmondena, 2023-12-04:
I corrected the version, it is now 23.05.2.
This is not duplicate but all issues stick together, were detected with same router but are either specific to this device like the watchdog issue or are general to Openwrt as follows:
  - The watchdog issue is a bug that caused my new v4 router to completely freeze instead of reboot. I don't know if I have worked around this issue or if you fixed this issue somehow with new 23.05.2? What I did from my side with regards to the watchdog and what may have helped was 1st to increase WDG timeout to 60 and frequency to 9 to reduce load and increase tolerance for more robustness. 2nd I increased many of the high memory and CPU loads like, lighthttpd, samba, dnscrypt-proxy2 and adguardhome and many more to between nice levels of 5-19 and limited adguardhome with GOMEMLIMIT='32MiB' This helped a lot for performance and stability to relatively give usbusd's "0" nice level a higher priority.
  - One critical symptom of the watchdog behavior in combination with unintended HW-freezes and reboots is that it is one main source of FS corruptions. Either when it freezes the router, that you have to do a power cycle. Or when it reset your router it causes incomplete write operations which seem both leading to FS corruptions.
  - But to be honest is that the watchdog is one source among some others with regards to NAND silent errors that introduce additional FS corruptions and "high frequent errors" (some times per week)

- Anyway the key solution to fix any of these FS corruption errors is to have a running e2fsck that everytime you reboot makes a intensive e2fsck and is the most valuable solution in combination with ext4 in writeback mode and for every todays NAND usb storage based Extroot installation.
This shows a general lack of todays Openwrt to have the possibility to do a deep FS check at boot time before you mount the Extroot environment and would be required anyway if the HW watchdog or whatever (e.g. swap memory manager) kills the kernel.
But even with Openwrts great Firmware-Selector it is not possible to have e2fscheck configured for an intensive FS check in advance of mounting Extroot.
BR and thanks for your reply and help!

brada4, 2023-12-04:
If you are testing not yet introduced V4 support you need to report over there at PR in works, 
it is not supported by current V3 image past booting to ssh to keep on porting.

Edmondena, 2023-12-05:
I report this firmware-selector bug as common for any Openwrt (maybe I'd better correct this above) as this report is a general and common behavior for all Extroot Openwrt installations.
BTW I tested this for WR902ACv3, WR902ACv4 and even Fritzbox 7530 based on arm_cortex-a7_neon-vfpv4 but why do you think that this should make any difference?

This is a openwrt.firmware-selector web configuration problem that has nothing to do with any architecture itself and you can easily try to understand the problem repeating it on your any hardware while trying to run Extroot and implement filesystem checks for your NAND usb storage device.

Edmondena, 2023-12-05:
This is clear but not the point. I have successfully setup Extroot some hundred times, did you? The problem is that for extroot you require a storage device and today this is mostly 100% NAND flash based storage devices and they are prone to silent errors with increasing error rates since years. Thus it is absolutely "normal" that they cause errors why you use them even if you buy them from stock. If you're lucky today they can be written some 100 times until cells start to fail and you cannot have Openwrt running reliable at 24/7 like routers require without checking them at least every time you boot Openwrt into the extroot environment.
If you don't believe me you find here all information you need to understand why e2fsck is the only solution for that: https://www.usenix.org/system/files/atc19-jaffer.pdf
Especially e2fsck -p -v -f -D -c /dev/sda1 with badblocks package installed. If you're cannot check extroot before mounting it we are running routers with low reliability especially when swap is required which is the case for all lower end devices that are supported by openwrt.

brada4, 2023-12-05:
probably more beef in throwing buck at more DDR inside router, or at least zram-swap.
extroot is a bit of wilderness outside normal usage.

Edmondena, 2023-12-05:
This is option as this router is only consuming 1W high efficiently and by far the only one around that can be run off-grid and from simple solar powered battery storage.
This is a common problem as openwrt is supporting more and more NAND based router and why its clear why it is urgently required to change from jffs2 to ubifs? Its simply because Ubi-FS has a defect management what is strictly a must have for NAND but for extroot based systems openwrt just because of that has handicrafts reliability :-(

brada4, 2023-12-05:
that is achieved by minimizing flash writes, kind like readyboost does. Usb flash kills that low power rose garden though.

Edmondena, 2023-12-05:
I fully agree to you that minimizing Flash writes is one option, but its not the only one.
Otherwise you would fully accept, that the future of Flash devices which is 100% NAND (NOR is mostly dead for the conumer market) would never again allow embedded off-grid linux storage devices like you require for tons of application like weather stations, earthquake data loggers, solar wifi relays or what ever.

It is possible if just strictly following some simple guidelines:
- You have to carefully calculate the USB-MTBFs for your device estimating average temperature and free Flash storage headroom for your swap device.
- An example calculation calculation I wrote somewhere else these days, I attach the link later.
- The key is to reduce the FS check rate down to around e.g. 1 full check per day so that you have a very high chance for an intensive e2fsck to find all single bit flips and correct them.
- No option is to check the FS only from time to time because these errors accumulate to fast until even e2fsck cannot fix them anymore.

brada4, 2023-12-05:
You need to plug usb stick somewhere else and fsck it, at the point rootfs overlay is mounted only scripts from readonly squashfs are executed. 
Default hardcoded e2fsck -p -v -f -D -c would keep router out of service for another day....

Edmondena, 2023-12-05:
Sorry but your suggestion sounds like you're kidding me as a somehow acceptable solution could be as simple as adding a START-Number to the firmware-selectors boot script to ensure it will be executed from rom and short in advance of mounting extroot by /etc/rc.d/S10fstab. Its just as simple as 1. Reading the head of the provided script 2. change the name of this boot script, bake it into rom and "boom" gives you all the required flexibility you need to solve this for everyone on its own.

To answer to your suggestions:
- The router is physically not accessible and suggesting to replug it somewhere else to check maybe best for handicrafts but not for any professional or even home router use case!
- You're 1d estimation is far away from reality. The above suggested full check on my 128GB SDXC USB-Highspeed storage takes exactly and only 8,5 minutes, can run at night, takes the router not out only 0,5% of duty and is no real problem.
- Obviously I'd better  live with my ugly workaround though this is much comfortable than your suggestion by just rebooting twice into internal root Openwrt for fs_check service and after that return to extroot for full functionality.

Otherwise if someone in the future wants to take care about improving the openwrt firmware-selector would be really great  to have a START=09 Boot-script entry that gives the opportunity to run a very simple bootscript to be injected by rom (every boot) and **before** /etc/rc.d/S10fstab gets executed.

Suggested Bug-Fix solution as bash script for firmware-seletor
sed -e'/^ *START=/!d' -e's/.*=\([0-9][0-9]*\).*/\/etc\/rc.d\/\1_user_boot_script/' /etc/uci-defaults/99-asu-defaults | xargs mv /etc/uci-defaults/99-asu-defaults

Thanks for your interest and for your kind help!

brada4, 2023-12-05:
heard of redhat? XFS thingie? Seriously go ask in forums, you have no intent to report or fix bugs here. It is your system, you tune it to your needs. 
8 minutes a day is not 24x7
that equals 2 days on-site work a year sry, you have to work out your contradictions.

brada4, 2023-12-06:
You need to find what parts of your supposed service maxima can work on your very constrained hardware.