openwrt/openwrt #14137: MF289F: Won't boot from spi on current snapshot

fisch2, 2023-12-04:
### Describe the bug

[mf289f.log](https://github.com/openwrt/openwrt/files/13552500/mf289f.log)
Openwrt gets stuck during boot. See the attached log.

### OpenWrt version

SNAPSHOT

### OpenWrt release

 r24564-78d259e7d2

### OpenWrt target/subtarget

ipq40xx

### Device

MF289F

### Image kind

Official downloaded image

### Steps to reproduce

Install or Upgrade Openwrt to the latest snapshot.

### Actual behaviour

The device gets stuck during boot.

### Expected behaviour

The device boots, a shell can be accessed via serial, the device is reachable from its lan port, etc...

### Additional info

I already attempted to use the current snapshot in the past and I believe I encountered the same issue with the device getting stuck at `Waiting for root device /dev/ubiblock0_1...`, after the target got upgraded to linux 6.1.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chunkeey, 2023-12-05:
Congrats! I think you just found a really juicy bug here. 

Looking at the code, it seems that his patch is the culprit:

https://lore.kernel.org/all/20220420102022.3310970-1-p.yadav@ti.com/

But it's not that easy, since the patch does make sense... so best would be to update the driver.
(I think there are more bugs like this!)

Can you please test the following patch (i.e. put the content into `/target/linux/ipq40xx/patches-6.1/999-gigadevice-fix.patch` and compile again?). 

Note: This patch only touches on that specific function, but there are potentially more.
so if it crashes, the `gd5fxgq4uexxg_ecc_get_status` function() shouldn't show up in the backtrace)

Thank you!
---
```
diff --git a/drivers/mtd/nand/spi/gigadevice.c b/drivers/mtd/nand/spi/gigadevice.c
index 987710e09441..3d4e2e616c44 100644
--- a/drivers/mtd/nand/spi/gigadevice.c
+++ b/drivers/mtd/nand/spi/gigadevice.c
@@ -184,7 +184,7 @@ static const struct mtd_ooblayout_ops gd5fxgq4xc_oob_256_ops = {
 static int gd5fxgq4uexxg_ecc_get_status(struct spinand_device *spinand,
 					u8 status)
 {
-	u8 status2;
+	u8 *status2;
 	struct spi_mem_op op = SPINAND_GET_FEATURE_OP(GD5FXGQXXEXXG_REG_STATUS2,
 						      &status2);
 	int ret;
@@ -208,7 +208,7 @@ static int gd5fxgq4uexxg_ecc_get_status(struct spinand_device *spinand,
 		 */
 		/* bits sorted this way (3...0): ECCS1,ECCS0,ECCSE1,ECCSE0 */
 		return ((status & STATUS_ECC_MASK) >> 2) |
-			((status2 & STATUS_ECC_MASK) >> 4);
+			((*status2 & STATUS_ECC_MASK) >> 4);
 
 	case GD5FXGQ4XA_STATUS_ECC_8_BITFLIPS:
 		return 8;
@@ -226,13 +226,22 @@ static int gd5fxgq4uexxg_ecc_get_status(struct spinand_device *spinand,
 static int gd5fxgq5xexxg_ecc_get_status(struct spinand_device *spinand,
 					u8 status)
 {
-	u8 status2;
-	struct spi_mem_op op = SPINAND_GET_FEATURE_OP(GD5FXGQXXEXXG_REG_STATUS2,
-						      &status2);
+	u8 *status2 = kzalloc(sizeof(*status2), GFP_KERNEL | GFP_DMA);
+	struct spi_mem_op op = SPINAND_GET_FEATURE_OP(GD5FXGQXXEXXG_REG_STATUS2, status2);
 	int ret;
 
+	/*
+	 * this is dangerous, since we already referenced the status2 pointer, that said:
+	 *
+	 * commit a3ca86aea507 ("Add '-fno-delete-null-pointer-checks' to gcc CFLAGS")
+	 * should protect us from damange.
+	 */
+	if (!status2)
+		return -ENOMEM;
+
 	switch (status & STATUS_ECC_MASK) {
 	case STATUS_ECC_NO_BITFLIPS:
+		kfree(status2);
 		return 0;
 
 	case GD5FXGQ5XE_STATUS_ECC_1_4_BITFLIPS:
@@ -241,6 +250,7 @@ static int gd5fxgq5xexxg_ecc_get_status(struct spinand_device *spinand,
 		 * bit error status
 		 */
 		ret = spi_mem_exec_op(spinand->spimem, &op);
+		kfree(status2);
 		if (ret)
 			return ret;
 
@@ -248,12 +258,14 @@ static int gd5fxgq5xexxg_ecc_get_status(struct spinand_device *spinand,
 		 * 1 ... 4 bits are flipped (and corrected)
 		 */
 		/* bits sorted this way (1...0): ECCSE1, ECCSE0 */
-		return ((status2 & STATUS_ECC_MASK) >> 4) + 1;
+		return ((*status2 & STATUS_ECC_MASK) >> 4) + 1;
 
 	case STATUS_ECC_UNCOR_ERROR:
+		kfree(status2);
 		return -EBADMSG;
 
 	default:
+		kfree(status2);
 		break;
 	}
 
``

fisch2, 2023-12-06:
Thanks for your help, I will wait for a new kernel release with this patch included and report back, whether it did indeed fix the issue.

fisch2, 2023-12-06:
I've now tried to build a custom image with the patch from the mailing list and at first glance it seems that with the patch, the issue is now fixed. Here is the new log from the device with the patch applied: 
[mf289f.log](https://github.com/openwrt/openwrt/files/13581633/mf289f.log)
Therefore, I'm closing the issue for now. Thanks a lot!

fisch2, 2024-01-26:
The fix for ```gigadevice.c``` still hasn't made it into the mainline kernel or any of the stable kernels, including the newest 6.1.x release, which at the time of writing is 6.1.75. However, it is part of linux-next. Could this patch either be added to the openwrt repo or could we request upstream to include the fix in the stable releases? It seems kind of unintentional, that the fix isn't present there.

fisch2, 2024-03-07:
Fix got added in 6.1.81. Closing.

closed 2024-03-07 by fisch2