openwrt/openwrt #14138: MF283+ Regression, 22.03 Ethernet, 23.05 Kernel

najdanovicivan, 2023-12-05:
### Describe the bug

There are issues with kernel on the device

21.02.7 - Works Fine 

22.03.0 +  Ethernet not working wont even respond to ping occasional kernel panic.

23.05.0 + - initramfs-kernel  not booting stuck at Starting kernel ...

22.03.0 bootlog 

```
Starting kernel ...

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80000000...

[    0.000000] Linux version 5.10.138 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19685-512e76967f) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Sat Sep 3 02:55:34 2022
[    0.000000] SoC Type: Ralink RT3352 id:1 rev:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001964c (MIPS 24KEc)
[    0.000000] MIPS: machine is ZTE MF283+
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00002577
[    0.000000] Readback ErrCtl register=00002577
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55168K/65536K available (3980K kernel code, 582K rwdata, 572K rodata, 4392K init, 199K bss, 10368K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 400MHz
[    0.000000] timer_probe: no matching timers found
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9556302233 ns
[    0.000018] sched_clock: 32 bits at 200MHz, resolution 5ns, wraps every 10737418237ns
[    0.015730] Calibrating delay loop... 265.42 BogoMIPS (lpj=1327104)
[    0.107965] pid_max: default: 32768 minimum: 301
[    0.117360] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.131742] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.151355] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.171281] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.190693] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.204145] pinctrl core: initialized pinctrl subsystem
[    0.219133] NET: Registered protocol family 16
[    0.348905] rt2880_gpio 10000600.gpio: registering 24 gpios
[    0.359956] rt2880_gpio 10000600.gpio: registering 24 irq handlers
[    0.372924] rt2880_gpio 10000638.gpio: registering 16 gpios
[    0.403214] clocksource: Switched to clocksource MIPS
[    0.415695] NET: Registered protocol family 2
[    0.424833] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.440822] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.457417] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.472516] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.486393] TCP: Hash tables configured (established 1024 bind 1024)
[    0.499253] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.512199] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.526538] NET: Registered protocol family 1
[... 118 lines cut ...]
```

### OpenWrt version

r19685-512e76967f

### OpenWrt release

22.03.0

### OpenWrt target/subtarget

ramips/rt305x

### Device

ZTE MF283+

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mpratt14, 2023-12-12:
@Leo-PL you might as well make a PR with the reset signal changes as well

Leo-PL, 2024-01-02:
@najdanovicivan would you like to test a build? This can possibly speed up merging of both fixes.

najdanovicivan, 2024-01-07:
@Leo-PL I've tested the latest snapshot build and I can confirm that device is now booting properly and that ethernet works without any issues

Leo-PL, 2024-01-07:
Glad to hear that!

closed 2024-12-26 by najdanovicivan