openwrt/openwrt #14139: typo in the 22.03.6 release note

xlighting2017, 2023-12-05:
### Describe the bug

I presume the version number should be **22**.03 , not   23.03, right?
[image]

### OpenWrt version

22.03.6

### OpenWrt release

22.03.6

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-12-05:
The report does not concern openwrt itself but incorrect information on the website: https://openwrt.org/releases/22.03/notes-22.03.6

brada4, 2023-12-05:
ping @hauke as per wiki history

closed 2023-12-09 by Ansuel