openwrt/openwrt #14141: upgrade new file, error code - Xiaomi Miwifi Mini MT7620

InBomBi, 2023-12-05:
### Describe the bug

i build firmware for xiaomi mini wifi Mt7620, when running sysupgrade, error appear as bellow.
/lib/upgrade/platform.sh: line 33: syntax error: unexpected redirection (expecting ")")
If i upgrade from shell, system did not reboot and can not run.
Upgrade the old FW is ok.
Is there something change to the new openwrt enviroment?
Thanks

### OpenWrt version

SNAPSHOT r24579-62277c485e

### OpenWrt release

r24579-62277c485e

### OpenWrt target/subtarget

MT7620

### Device

Xiaomi miwifi mini

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

InBomBi, 2023-12-05:
> Invalid Target/Subtarget reported. `MT7620` Is this from a supported device?
Yes it is, I build it from clean repository? with git clone from openwrt

brada4, 2023-12-05:
it checks for info that is produced from commands ran on the device, not fuzzy approximations.
for sysupgrade from/to version needed and try upgrade clearing config

InBomBi, 2023-12-05:
> it checks for info that is produced from commands ran on the device, not fuzzy approximations. for sysupgrade from/to version needed and try upgrade clearing config

so you mean i need to upgrade from factory reset? right?

DragonBluep, 2023-12-05:
Please check your local commits and patches. I believe some of them have conflicts with https://github.com/openwrt/openwrt/commit/1b091311aafc59db576a51b3d2b5febaac45d522 in `target/linux/ramips/mt7620/base-files/lib/upgrade/platform.sh`

brada4, 2023-12-05:
> factory reset

determine bug trigger, eg release to snapshot and werk old snapshot to latest, and check if errors persist if reset is triggered (which actually means  not packing and restoring old conf files)

InBomBi, 2023-12-06:
I have to upgrade firmware from pbbox mode, then it can run normal.
Thank for all of your support