openwrt/openwrt #14146: EEE buggy, loosing link multiple times a day

philS1, 2023-12-05:
### Describe the bug

I have a cisco core switch with eee enabled. I get multiple Network device 'eth0' link is down and Network device 'eth0' link is up every day, as soon as I disable eee on the core switch, problem is gone

``
Mon Dec  4 20:10:49 2023 daemon.notice netifd: Network device 'eth0' link is down
Mon Dec  4 20:10:49 2023 kern.info kernel: [28550.353274] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Mon Dec  4 20:10:49 2023 kern.info kernel: [28550.354267] br-lan: port 1(eth0) entered disabled state
Mon Dec  4 20:10:53 2023 daemon.notice netifd: Network device 'eth0' link is up
Mon Dec  4 20:10:53 2023 kern.info kernel: [28554.512439] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Mon Dec  4 20:10:53 2023 kern.info kernel: [28554.512490] br-lan: port 1(eth0) entered blocking state
Mon Dec  4 20:10:53 2023 kern.info kernel: [28554.512502] br-lan: port 1(eth0) entered forwarding state
``

### OpenWrt version

24555-4b4c940fbc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ZyXEL NWA50AX Pro

### Image kind

Official downloaded image

### Steps to reproduce

just wait, problem will occur multiple times a day

### Expected behaviour

Device should not loose ethernet connectivity

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-05:
please post
`ethtool --show-eee eth0` 
if it is shown & enabled you can disable it as well

brada4, 2023-12-06:
Then disable on the opposite port.

brada4, 2023-12-11:
@philS1 indeed, this is phy connected directly to mtk soc port without switch. what is the manufacturer of PHY via `mii-tool eth0 -v`  , better post in other issue alongside router make, as symptom is almost same.
@namiltd thanks, another theory that it is not phy but particular unfortunate mii on directly connected phy (no switch here)

tim-seoss, 2024-01-30:
@philS1 Do you know if it does this with the OEM firmware too?  Do you know if the cabling has been tested (or changed with an alterative path)?  What's the cable type (shielded vs unshielded etc. and approx length). Has a different switch port been tried?  Since EEE works by backing-off the TX power on the link, it's possible that one end or another is backing off the tx power too far, or that periodic noise is causing loss of link.

philS1, 2024-02-01:
@tim-seoss Sorry I didn't test with OEM fw. Cabling is shielded and is tested using a LAN tester. I have 3 of those APs and problem occurs on all of them. Only the link is lost, teh  POE power stays active.

TobleMiner, 2024-05-20:
The MaxLinear Ethernet PHY used in the AP has a feature called `Smart-AZ`:

> 3.8 Smart-AZ Feature
The Smart-AZ feature is relevant when the GPY211 is connected to a MAC SoC that does not implement the EEE
feature in its MAC layer. In this case, the MAC SoC cannot initiate a transition to the low-power idle state.
To alleviate the limitation of such a MAC SoC, the GPY211 detects the conditions that may lead to low-power idle
and generates the control messages to enter EEE mode in accordance with the IEEE 802.3az standard.
The Smart-AZ feature is always enabled
[1]

Not saying that has to be the issue here but it could be another source for problems.

[1] https://maxlinear-assets.azureedge.net/web/documents/617810_gpy211b1vc_gpy211c0vc_ds_rev1.4.pdf