openwrt/openwrt #14147: Stop LEDE Project updates to have only one place

Neustradamus, 2023-12-05:
### Describe the bug

Dear @openwrt team,

There is:
- https://github.com/lede-project/source

Now it is:
- https://github.com/openwrt/openwrt

It is possible to stop
- https://github.com/lede-project/source

All people will move on https://github.com/openwrt/openwrt.

When we do a search on GitHub, there are double...

Thanks in advance.

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-06:
check again and close this issue

brada4, 2023-12-06:
and? forks is basic functionality in github, just checked, button still in same place.

stklcode, 2023-12-11:
Technically it’s a mirror, not a fork.
But mirroring is also generally fine, if there’s any need for it (no idea if some downstream projects still reference the LEDE stuff) I guess there are many many more mirrors and forks across the web (I have 2 for myself, GH fork and an on-prem mirror), nothing wrong with that.

One could be slightly more clear that there is no development at all in this repo and also one should disable GH actions which fail in almost any case.