openwrt/openwrt #14148: lede-project.org redirection problems

Neustradamus, 2023-12-05:
### Describe the bug

Dear @openwrt team,

It is not redirect to https://openwrt.org/:
- http://lede-project.org/
- https://lede-project.org/
- http://www.lede-project.org/
- https://www.lede-project.org/

Can you solve it?

Thanks in advance.

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-06:
archive.org says you rwong

Neustradamus, 2023-12-06:
@brada4: I think that one guy has done recently ^^

Thanks to this guy!