openwrt/openwrt #14149: Problem with router xiaomi MI3G v1? firmware 23.05.0 and 23.05.2 - cannot enable 5G wifi network

skif5, 2023-12-05:
### Describe the bug

Hi OpenWRT
install openwrt 23.05.2 release on xiaomi MI3G router. 2.4 network fine, but problem with 5GHz network (can not enable 5G network)
in syslog
-----------------------------------------
Tue Dec  5 20:25:43 2023 daemon.notice wpa_supplicant[1303]: Set new config for phy phy1
Tue Dec  5 20:25:43 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Tue Dec  5 20:25:43 2023 daemon.notice hostapd: Restart interface for phy phy1
Tue Dec  5 20:25:43 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3] channel=36  interface=phy1-ap0 bssid=00:aa:bb:cc:dd:ee ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=HeroeS5!$ wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=103HMWS_5G1 bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK o
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.652243] br-lan: port 4(phy1-ap0) entered blocking state
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.657909] br-lan: port 4(phy1-ap0) entered disabled state
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.663968] device phy1-ap0 entered promiscuous mode
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.669433] br-lan: port 4(phy1-ap0) entered blocking state
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.675056] br-lan: port 4(phy1-ap0) entered forwarding state
Tue Dec  5 20:25:44 2023 kern.info kernel: [ 1405.682257] br-lan: port 4(phy1-ap0) entered disabled state
Tue Dec  5 20:25:44 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->HT_SCAN
Tue Dec  5 20:25:44 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Tue Dec  5 20:25:45 2023 kern.info kernel: [ 1406.800015] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Tue Dec  5 20:25:45 2023 kern.info kernel: [ 1406.807175] br-lan: port 4(phy1-ap0) entered blocking state
Tue Dec  5 20:25:45 2023 kern.info kernel: [ 1406.812841] br-lan: port 4(phy1-ap0) entered forwarding state
Tue Dec  5 20:25:45 2023 daemon.notice netifd: Network device 'phy1-ap0' link is up
Tue Dec  5 20:25:45 2023 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->ENABLED
Tue Dec  5 20:25:45 2023 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Tue Dec  5 20:25:45 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Dec  5 20:25:45 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Dec  5 20:25:45 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Tue Dec  5 20:25:45 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Dec  5 20:25:46 2023 kern.err kernel: [ 1408.006425] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:46 2023 daemon.warn odhcpd[1545]: No default route present, overriding ra_lifetime!
Tue Dec  5 20:25:47 2023 kern.err kernel: [ 1409.246399] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:48 2023 kern.err kernel: [ 1410.266396] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:50 2023 kern.err kernel: [ 1411.496400] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:51 2023 kern.err kernel: [ 1412.736383] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:52 2023 kern.err kernel: [ 1413.966424] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:53 2023 kern.err kernel: [ 1415.196360] mt76x2e 0000:02:00.0: MAC error detected
Tue Dec  5 20:25:55 2023 kern.err kernel: [ 1416.436383] mt76x2e 0000:02:00.0: MAC error detected
--------------------------------------------
if I try to change canal to 120, in syslog 
------------------------------------------- 
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.559557] br-lan: port 4(phy1-ap0) entered blocking state
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.565175] br-lan: port 4(phy1-ap0) entered disabled state
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.571292] device phy1-ap0 entered promiscuous mode
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.576741] br-lan: port 4(phy1-ap0) entered blocking state
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.582387] br-lan: port 4(phy1-ap0) entered forwarding state
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.589238] br-lan: port 4(phy1-ap0) entered disabled state
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: Frequency 5600 (primary) not allowed for AP mode, flags: 0x197b NO-IR RADAR
Tue Dec  5 20:20:58 2023 daemon.err hostapd: Primary frequency not allowed
Tue Dec  5 20:20:58 2023 daemon.warn hostapd: phy1-ap0: IEEE 802.11 Configured channel (120) or frequency (5600) (secondary_channel=-1) not found from the channel list of the current mode (2) IEEE 802.11a
Tue Dec  5 20:20:58 2023 daemon.warn hostapd: phy1-ap0: IEEE 802.11 Hardware does not support configured channel
Tue Dec  5 20:20:58 2023 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->DISABLED
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Tue Dec  5 20:20:58 2023 daemon.err hostapd: phy1-ap0: Unable to setup interface.
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.601947] device phy1-ap0 left promiscuous mode
Tue Dec  5 20:20:58 2023 kern.info kernel: [ 1119.607078] br-lan: port 4(phy1-ap0) entered disabled state
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Tue Dec  5 20:20:58 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Tue Dec  5 20:20:58 2023 daemon.notice hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
Tue Dec  5 20:20:58 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Tue Dec  5 20:20:59 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Dec  5 20:20:59 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Dec  5 20:20:59 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Tue Dec  5 20:20:59 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Dec  5 20:21:00 2023 daemon.warn odhcpd[1545]: No default route present, overriding ra_lifetime!
---------------------------------------------------------
/etc /var/run/hostapd and syslog and dmesg log in archive in /tmp dir
[my_config.tar.gz](https://github.com/openwrt/openwrt/files/13572880/my_config.tar.gz)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

skif5, 2023-12-05:
try change default mac on interface - config wifi-iface 'default_radio1'
option macaddr '00:aa:bb:cc:dd:ee'
but it's not work too.

brada4, 2023-12-05:
SET COUNTRY CODE

skif5, 2023-12-06:
Hi brada4,
it solve a problem. But why it's now documented? Why error is not understandable

brada4, 2023-12-06:
it is common knowledge, just like no CH13 in the US. 
Documented in 3a here: https://openwrt.org/docs/guide-quick-start/basic_wifi like if you red documentation at all could be hard to miss.

skif5, 2023-12-06:
but I monitor system. Not solved at all

Wed Dec  6 14:33:32 2023 daemon.notice hostapd: phy0-ap1: AP-STA-DISCONNECTED f4:12:fa:06:c3:14
Wed Dec  6 14:33:32 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: disassociated
Wed Dec  6 14:33:32 2023 kern.err kernel: [  425.281006] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:32 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: authenticated
Wed Dec  6 14:33:32 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: associated (aid 2)
Wed Dec  6 14:33:32 2023 daemon.notice hostapd: phy0-ap1: AP-STA-CONNECTED f4:12:fa:06:c3:14 auth_alg=open
Wed Dec  6 14:33:32 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 14:33:32 2023 daemon.notice hostapd: phy0-ap1: EAPOL-4WAY-HS-COMPLETED f4:12:fa:06:c3:14
Wed Dec  6 14:33:33 2023 kern.err kernel: [  426.511058] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:34 2023 kern.err kernel: [  427.651005] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:35 2023 kern.err kernel: [  428.690961] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:36 2023 daemon.info hostapd: phy0-ap0: STA 24:a1:60:1d:a1:34 IEEE 802.11: disassociated
Wed Dec  6 14:33:36 2023 kern.err kernel: [  429.700991] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:37 2023 daemon.info hostapd: phy0-ap0: STA 24:a1:60:1d:a1:34 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Dec  6 14:33:38 2023 kern.err kernel: [  430.950983] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:38 2023 daemon.notice hostapd: phy0-ap1: AP-STA-DISCONNECTED 9c:9c:1f:89:8a:65
Wed Dec  6 14:33:38 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 IEEE 802.11: disassociated
Wed Dec  6 14:33:39 2023 kern.err kernel: [  432.190986] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:39 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Dec  6 14:33:40 2023 daemon.info hostapd: phy0-ap0: STA 24:a1:60:1d:a1:34 IEEE 802.11: authenticated
Wed Dec  6 14:33:40 2023 daemon.info hostapd: phy0-ap0: STA 24:a1:60:1d:a1:34 IEEE 802.11: associated (aid 2)
Wed Dec  6 14:33:40 2023 kern.err kernel: [  433.211049] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:40 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 24:a1:60:1d:a1:34 auth_alg=open
Wed Dec  6 14:33:40 2023 daemon.info hostapd: phy0-ap0: STA 24:a1:60:1d:a1:34 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 14:33:40 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 24:a1:60:1d:a1:34
Wed Dec  6 14:33:41 2023 kern.err kernel: [  434.220965] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:42 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 IEEE 802.11: authenticated
Wed Dec  6 14:33:42 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 IEEE 802.11: associated (aid 3)
Wed Dec  6 14:33:42 2023 daemon.notice hostapd: phy0-ap1: AP-STA-CONNECTED 9c:9c:1f:89:8a:65 auth_alg=open
Wed Dec  6 14:33:42 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 14:33:42 2023 daemon.notice hostapd: phy0-ap1: EAPOL-4WAY-HS-COMPLETED 9c:9c:1f:89:8a:65
Wed Dec  6 14:33:42 2023 kern.err kernel: [  435.460939] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:43 2023 kern.err kernel: [  436.690962] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:45 2023 kern.err kernel: [  437.930947] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:46 2023 kern.err kernel: [  439.160970] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:46 2023 auth.err sslh[4190]: forward to openvpn failed:connect: Connection refused
Wed Dec  6 14:33:47 2023 kern.err kernel: [  440.420875] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:48 2023 kern.err kernel: [  441.650896] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:33:49 2023 kern.err kernel: [  442.660896] mt76x2e 0000:02:00.0: MAC error detected

skif5, 2023-12-06:
I cannot understand why - mt76x2e 0000:02:00.0: MAC error detected. Try change mac on 5G network - but it's have not result.

Wed Dec  6 14:43:23 2023 daemon.notice hostapd: phy0-ap1: AP-STA-CONNECTED 9c:9c:1f:89:8a:65 auth_alg=open
Wed Dec  6 14:43:23 2023 daemon.info hostapd: phy0-ap1: STA 9c:9c:1f:89:8a:65 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 14:43:23 2023 daemon.notice hostapd: phy0-ap1: EAPOL-4WAY-HS-COMPLETED 9c:9c:1f:89:8a:65
Wed Dec  6 14:43:24 2023 kern.err kernel: [ 1016.867695] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:25 2023 kern.err kernel: [ 1017.887810] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:28 2023 kern.err kernel: [ 1020.777671] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:29 2023 kern.err kernel: [ 1021.787725] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:29 2023 daemon.notice hostapd: phy0-ap1: AP-STA-DISCONNECTED f4:12:fa:06:c3:14
Wed Dec  6 14:43:29 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: disassociated
Wed Dec  6 14:43:29 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: authenticated
Wed Dec  6 14:43:29 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 IEEE 802.11: associated (aid 2)
Wed Dec  6 14:43:29 2023 daemon.notice hostapd: phy0-ap1: AP-STA-CONNECTED f4:12:fa:06:c3:14 auth_alg=open
Wed Dec  6 14:43:29 2023 daemon.info hostapd: phy0-ap1: STA f4:12:fa:06:c3:14 WPA: pairwise key handshake completed (RSN)
Wed Dec  6 14:43:29 2023 daemon.notice hostapd: phy0-ap1: EAPOL-4WAY-HS-COMPLETED f4:12:fa:06:c3:14
Wed Dec  6 14:43:30 2023 kern.err kernel: [ 1023.017759] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:31 2023 kern.err kernel: [ 1024.267671] mt76x2e 0000:02:00.0: MAC error detected
Wed Dec  6 14:43:32 2023 kern.err kernel: [ 1025.497706] mt76x2e 0000:02:00.0: MAC error detected

DragonBluep, 2023-12-06:
It's a well-known MT7612 issue. Removing USB peripheral should have some help.

skif5, 2023-12-06:
> It's a well-known MT7612 issue. Removing USB peripheral should have some help.

Yes, 5G network is up after disconnect usb device. But I need usb storage. How make them work together

DragonBluep, 2023-12-06:
> > It's a well-known MT7612 issue. Removing USB peripheral should have some help.
> 
> Yes, 5G network is up after disconnect usb device. But I need usb storage. How make them work together

You can have a try:

1. Use USB2.0 cable.
2. Keep the USB peripheral away from the router.
3. Cover the data cable and USB device with tin foil.

brada4, 2023-12-06:
Set channel to auto, it may be able to escape interference.
Xiaomi is known for production variety, like shielding components for one quarter, then not then back again.
You need to switch USB devices on one at a time and tinfoil the bad ones https://www.usb.org/sites/default/files/327216.pdf (ff to page 14 for entertaining comic strip)

LynxChaus, 2023-12-07:
> page 14 for entertaining comic strip

But in pdf measure noise impact on 2.4GHz, but there 5GHz not working.

brada4, 2023-12-07:
You see unshielded back of usb3 port? 
https://i.ytimg.com/vi/ERAKolFLYxk/maxresdefault.jpg
Basically highest spread spectrum of USB3 ends before 5GHz, and lowest is double of 2.4GHz CH14.
Tinfoil works for both bands. 
First bet is to glue coin at back of USB port, then go with wrapping if no luck.
1010101 pattern breaks lowest part of 5GHz and 110011001100 breaks upper side of 2.4GHz with very bad luck.

LynxChaus, 2023-12-07:
> You see unshielded back of usb3 port?
> https://i.ytimg.com/vi/ERAKolFLYxk/maxresdefault.jpg

You show MI3 router, without G. Mi3G have shielded USB connector: https://ixbt.online/live/images/original/00/05/45/2017/10/02/e33fcf1be3.jpg https://ixbt.online/live/images/original/00/05/45/2017/10/03/960053518f.jpg

And i own MI3G with USB3.0 Seagate drive connected to it for serving videos. No problem with WiFi.

LynxChaus, 2023-12-07:
For reference https://github.com/openwrt/mt76/issues/666

LynxChaus, 2023-12-07:
> Yes, 5G network is up after disconnect usb device. But I need usb storage. How make them work together

Just curious - USB storage powered from USB port or from it's own PSU?

skif5, 2023-12-15:
Thanks, problem was with usb3, may be with power. After replace to usb2 cable problem not repeated. May be ticket can be closed.

closed 2023-12-15 by skif5