openwrt/openwrt #14161: Boot loop/crash for Buffalo WHR-600D after installation of 23.05.2 and enabling wifi - incompatible wireless driver for this hardware

tmittelstaedt, 2023-12-07:
### Describe the bug

Wireless driver was changed and the changed driver now DOES list the 5Ghz interface whereas the older driver only listed the 2.4Gkz interface, however enabling the wireless crashes the router

### OpenWrt version

23.05.2

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

23.05.2

### Device

Buffalo WHR-600D

### Image kind

Official downloaded image

### Steps to reproduce

I had a Buffalo WHR-600D running 19.07.5 for 4 months setup as an access point and it was running fine however it only showed the 802.11bgn wireless interface.  I loaded  https://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7620/openwrt-23.05.2-ramips-mt7620-buffalo_whr-600d-squashfs-sysupgrade.bin from the LuCi webinterface and selected to NOT maintain settings.  After rebooting twice the router came up fine displaying 23.05.2 and I logged in and set the root password.  I then went to the Network, Wireless interface and now both 2.4 and 5 Ghz wireless interfaces were showing but were in disabled state. I enabled the a/b/g/n interface.  The router immediately rebooted and then continued to reboot over and over.  Finally, after about 6 or 7 reboots it came back up and I was able to login.  The a/b/g/n interface was enabled but I could NOT see a wireless advertisement from the router, even on a PC located about a foot from the router.  I then attempted to downgrade to the 19.07.10 firmware not saving settings and the router refused to take the firmware even after forcing it and went back through the reboot cycle of multiple reboots ending up on 23.05.2 once more with the wireless interface enabled but no actual advertisement from the device.   I then attempted to downgrade back to 19.07.5 from the LuCi interface, not saving settings, and this was successful even though the interface complained that the checksum was invalid.  The router went back to normal without multiple reboots.  I then updated to 19.07.6 from LuCi with no problems but the router was once more again showing only the single 2.4Ghz wireless interface.  I noted that the size of the firmware image jumped past 4MB going from version 19.07.6 to 19.07.7 and I have to conclude a change in the wifi driver was made, which made the firmware larger, but which is invalid for this hardware.

Please update https://openwrt.org/toh/buffalo/whr-600d with a link to this bug until such time that the firmware for this router is corrected, thank you!

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-07:
Obviously if you got mismatched checksum after upload you do not have openwrt.
Try 100x if necessary, but that checksum should match the firmware-selector.

tmittelstaedt, 2023-12-08:
> Hi! can you help to test 22.03 and snapshot versions? I think snapshot should be okay.

Uh, you want me to go backwards from the released version????

tmittelstaedt, 2023-12-08:
> It is 8MB flash device, you can try reduced image:
> https://firmware-selector.openwrt.org/?version=23.05.2&target=ramips%2Fmt7620&id=buffalo_whr-600d
> Expand package selection, by default it has no luci, now generate smaller sysupgrade image.
> jffs needs 3 free eraseblocks to perform own garbage collection, might be first boot scripts detect wifi and overlay is almost full at that point.
> You need to check after flash with config reset - /proc/mtd and df -k to assess if overlay is full and causes problems vefore getting into bringing up wifi with uci command line.

There is no reduced image listed on that link.  only kernel and sysupgrade.  sysupgrade on that device is only 5.6MB in size.  How many MB free flash would be needed for 3 eraseblocks?

Incidentally the dd-wrt project had a big problem with this hardware also.

Is there a reduced image snapshot build somewhere?

brada4, 2023-12-08:
Could be large OEM partitions like bootloader, recovery. calibration data.
v22 measures 5.8MB which means it is less likely to finish first boot hardware detections.
neither reduces final image size after removing luci via firmware selector (just click package list and generate and luci will be gone)
To diagnose problem:
Check `cat /proc/mtd` -> what is the erase block size?
Check `df -h` -> how much of eraseblocks are available in /overlay
Unlikely but possible improvements
- find if flash chip on router can work (likely emulating at double wear) with smaller eraseblocks. That would give that bit of missing space
- try imagebuilder before time-consuming rebuild: https://openwrt.org/supported_devices/864_warning

tmittelstaedt, 2023-12-09:
Actually, I'm also interested in porting OpenWRT to the Netgear R6050 so I went ahead and did setup a build environment for doing that on Debian 12.  That router is a 128MB ram device with a Ralink SoC and looks like a prime candidate for a port as the R6020 is similar and is supported.  So I should be able to build a reduced size image for the WHR-600D and see how that works out, once I figure out how to use the build system, LOL.

brada4, 2023-12-09:
imagebuilder is just the last part of buildiing - stuffing existing kernel and packages into img.