openwrt/openwrt #14163: Linksys WRT1900ACS yesterdays nightly image 6th December does not boot

bittin, 2023-12-07:
### Describe the bug

Hey!

Tried to update to yesterdays NIGHTLY 6th December but it does not work so had to revert to 5th December

### OpenWrt version

-

### OpenWrt release

-

### OpenWrt target/subtarget

mvebu

### Device

Linksys WRT1900ACS

### Image kind

Official downloaded image

### Steps to reproduce

Flash the https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-linksys_wrt1900acs-squashfs-sysupgrade.bin image via luci and the router does not start

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-12-07:
Check which commits have been made from one version to the next.

fuqiang03, 2023-12-08:
Take this position

The questions are as follows

upgrade: Image metadata not present
Image check failed but --force given - will update anyway!
upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: ubus call system sysupgrade { "prefix": "\/tmp\/root", "path": "\/tmp\/openwrt-mvebu-cortexa9-linksys_wrt1900acs-squashfs-factory.img", "force": true, "command": "\/lib\/upgrade\/do_stage2", "options": { "save_partitions": 1 } } (Connection failed)

openwrt/main---Branch

bittin, 2023-12-09:
the nightly image from 9th December works as it should so could update my router now, so closing this issue

closed 2023-12-09 by bittin