openwrt/openwrt #14168: After upgrading to 23.05.2, had to delete /etc/uhttpd.crt and /etc/uhttpd.key to get uhttpd to run

dreamlayers, 2023-12-07:
### Describe the bug

I upgraded my Buffalo WZR-600DHP from 22.03.5 to 23.05.2 using Luci. Afterwards, it started properly, performed its routing functions and wireless worked. However, I couldn't use Luci. Connections to it via http redirected to https, but https connections were refused. I could still log in using my ssh key. Running logread, I saw errors like "daemon.err uhttpd[8712]: Failed to load certificate/key files". After deleting /etc/uhttpd.crt and /etc/uhttpd.key and then doing 'service uhttpd start', those files got regenerated, and Luci access via https worked again.

These were self signed certificates that were automatically generated by some earlier version of OpenWrt. 

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

Buffalo WZR-600DHP

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dreamlayers, 2023-12-08:
I was using it in 22.03.5 and then a few minutes later it didn't work in 23.05.2. It is highly unlikely it expired in that short interval. But wow, looking in the backup I made before the upgrade, Windows 10 says /etc/uhttpd.crt expired in 2022. So I guess that expired certificate was okay for 22.03.5 but not okay for 23.05.2?

I'm not sure how to properly view the contents of /etc/uhttpd.key. I don't suppose there is any expiration in there. Both of the old uhttpd.crt and uhttpd.key are well under 2 KB file size,  but the currently used working ones are even smaller.

The file command in Ubuntu says the old uhttpd.key is "DER Encoded Key Pair, 2048 bits" and the new one currently being used is "DER Encoded Key Pair, 1024 bits".

dreamlayers, 2023-12-08:
This is a bug though. An upgrade shouldn't break web admin access even if the key is expired.

brada4, 2023-12-08:
your configured redirect denied web admin access. the bug is in v22 loading expired certificate.

dreamlayers, 2023-12-08:
Sure, you can say loading an expired certificate is a bug. But when certificate expiration prevents the web server from working, OpenWrt needs to handle certificate expiration. Users should not simply lose web admin access and have to fix things via ssh. Without a fix for this, I expect the exact same thing to happen when my new certificate expires. Working with an expired certificate seems like less of a problem than loss of web admin access.

I did attempt this while I had the expired certificate:
```
uci set uhttpd.main.redirect_https=0
uci commit uhttpd
service uhttpd reload
```
It turned off the redirect, but did not help me regain web admin access via http.

brada4, 2023-12-08:
It will be a new certificate in browsers perception even if you self-sign using same private key. If it is "handled" automatically you need to accept any key at any time thus gaining no edge over plain http.

Currently new key generation is in luci-app-uhttpd that does not show expiry in UI. Probably you can submit your wish for enhancement right over there
https://github.com/openwrt/luci

mbedtls validates key by default, wolfssl did not.
https://os.mbed.com/teams/sandbox/code/mbedtls/docs/tip/pk_8h.html#af5a0ae67100c005638bb74f575ee4ad6

Another would be to parse crt using available of 3 frameworks and regenerate self-signed cert if it is expired. That goes to uhttpd packages' startup script. Note you need to add a lot of logic to pre-validate cert.

jow-, 2023-12-08:
@dreamlayers - can you provide the failing cert+key pair here? I assume it is related to the changed default SSL library and we need to relax some checks.

dreamlayers, 2023-12-10:
@jow-  I have attached the failing uhttpd.crt and uhttpd.key in
[uhttpd_crt_key.tar.gz](https://github.com/openwrt/openwrt/files/13628970/uhttpd_crt_key.tar.gz). These are from the backup I made immediately before the upgrade.

brada4, 2023-12-10:
@dreamlayers no news there, pair is cryptographically matching, but not valid for time of being

this validation can strike if network glitches and device is started with manufacture date before openwrt first start.

Shine-, 2023-12-12:
Your certificate is a SHA1 signed certificate. Allowing SHA1 certificates was disabled by default starting with mbedTLS 2.5.1 and completely removed as of mbedTLS 2.28.0, see [change log](https://github.com/Mbed-TLS/mbedtls/blob/development/ChangeLog).
So yes, it's a regression due to the SSL library change. But it looks like it's not possible to re-enable SHA1 support in mbedTLS anymore (and it's undesired anyway from a security point of view).

anjandev, 2024-09-28:
I ran into this issue on v22.03.6. My openwrt box just booted up one day and I could not access the webgui. I sshed in and found the uhttpd service was logging that it failed to load the certificates. I  removed /etc/uhttpd.crt and /etc/uhttpd.key files, restarted uhttpd, and uhttpd started up successfully.

venzislav, 2024-11-07:
I had the same issue and lost few hours until find what is the problem.
I upgraded from 20 to OpenWrt 23.05.5 r24106-10cc5fcd00 and cannot open luci anymore.
I tried to uninstall all uhttpd and luci packages and the same problem at the end I delete both .cer and .key files and 
after this I restarted the uhttpd daemon and it generate new keys and run ok now :

`root@venzihomeWrt:/etc# mv uhttpd.crt uhttpd.crt.org
root@venzihomeWrt:/etc# mv uhttpd.key uhttpd.key.org
root@venzihomeWrt:/etc# /etc/init.d/uhttpd restart
4+0 records in
4+0 records out
Generating RSA private key, 2048 bit long modulus

Generating selfsigned certificate with subject 'C=ZZ,ST=Somewhere,L=Unknown,O=OpenWrt822c3fd8,CN=OpenWrt,' and validity 20241107094313-20261107094313
root@venzihomeWrt:/etc# 
`

GGG-KILLER, 2025-01-21:
Just ran into this. Trying to make LUCI available over HTTPS with P-256 and curiously enough, it doesn't work with PEM, but works with DER.

I need PEM so I can load the entire certificate chain though.

Shine-, 2025-01-21:
It works fine with P-256 certificates in PEM format and also with certificate chains. I'm using this on all my devices.

But it's as I wrote above - unfortunately there's no hint as to **why** `libmbedtls` won't accept a certificate or chain, so you're on your own to figure out what's wrong with your file.