openwrt/openwrt #14170: Build fails with `/usr/include/x86_64-linux-gnu/c++/12/bits/os_defines.h:44:19: error: missing binary operator before token "("`

paulmenzel, 2023-12-07:
### Describe the bug

Build OpenWrt on Debian sid/unstable fails:

```
$ more /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
$ git log --oneline --no-decorate -1
230a2b8a5e tools/cmake: update to 3.27.9
```

```
$ make -j1 V=sc
[…]
mkdir -p -- target/.deps
/usr/bin/env bash ./../mkinstalldirs nat/.deps
mkdir -p -- nat/.deps
/home/build/src/openwrt/staging_dir/host/bin/g++ -x c++    -I. -I. -I./config -DLOCALEDIR="\"/home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/share/locale\"" -DHAVE_CONFIG_H -I./../include/opcode -I../bfd -I./../bfd -I./../include -I./../readline/readline/.. -I./../zlib -I/home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/include -pthread -I../libdecnumber -I./../libdecnumber  -I./../gnulib/import -I../gnulib/import -I./.. -I.. -I./../libbacktrace/ -I../libbacktrace/  -DTUI=1 -I/home/build/src/openwrt/staging_dir/host/include    -I./.. -pthread  -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-variable -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-error=maybe-uninitialized -Wno-mismatched-tags -Wsuggest-override -Wimplicit-fallthrough=3 -Wduplicated-cond -Wshadow=local -Wdeprecated-copy -Wdeprecated-copy-dtor -Wredundant-move -Wmissing-declarations -Wstrict-null-sentinel -Wformat -Wformat-nonliteral   -c -o gdb.o -MT gdb.o -MMD -MP -MF ./.deps/gdb.Tpo gdb.c
In file included from <command-line>:
/home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/include/stdc-predef.h:10: warning: "__STDC_UTF_16__" redefined
   10 | #define __STDC_UTF_16__ 1
      |
<built-in>: note: this is the location of the previous definition
/home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/include/stdc-predef.h:11: warning: "__STDC_UTF_32__" redefined
   11 | #define __STDC_UTF_32__ 1
      |
<built-in>: note: this is the location of the previous definition
In file included from /home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/include/stdio.h:8,
                 from ./../gdbsupport/common-defs.h:88,
                 from defs.h:28,
                 from gdb.c:19:
/home/build/src/openwrt/staging_dir/toolchain-mips_4kec_gcc-12.3.0_musl/include/features.h:33: warning: "_Noreturn" redefined
   33 | #define _Noreturn __attribute__((__noreturn__))
      |
In file included from ./../gdbsupport/common-defs.h:31:
./../gnulib/config.h:1518: note: this is the location of the previous definition
 1518 | #  define _Noreturn __attribute__ ((__noreturn__))
      |
In file included from /usr/include/x86_64-linux-gnu/c++/12/bits/c++config.h:655,
                 from /usr/include/c++/12/cstdlib:41,
                 from ./../gdbsupport/common-defs.h:92:
/usr/include/x86_64-linux-gnu/c++/12/bits/os_defines.h:44:19: error: missing binary operator before token "("
   44 | #if __GLIBC_PREREQ(2,15) && defined(_GNU_SOURCE)
      |                   ^
/usr/include/x86_64-linux-gnu/c++/12/bits/os_defines.h:53:20: error: missing binary operator before token "("
   53 | # if __GLIBC_PREREQ(2, 27)
      |                    ^
/usr/include/x86_64-linux-gnu/c++/12/bits/os_defines.h:65:20: error: missing binary operator before token "("
   65 | # if __GLIBC_PREREQ(2, 34)
      |                    ^
In file included from /usr/include/stdlib.h:26,
                 from /usr/include/c++/12/cstdlib:75:
/usr/include/x86_64-linux-gnu/bits/libc-header-start.h:74:17: error: missing binary operator before token "("
   74 | #if __GLIBC_USE (IEC_60559_BFP_EXT) || __GLIBC_USE (ISOC2X)
      |                 ^
/usr/include/x86_64-linux-gnu/bits/libc-header-start.h:80:17: error: missing binary operator before token "("
   80 | #if __GLIBC_USE (IEC_60559_BFP_EXT) || defined __STDC_WANT_IEC_60559_EXT__
      |                 ^
/usr/include/x86_64-linux-gnu/bits/libc-header-start.h:97:17: error: missing binary operator before token "("
   97 | #if __GLIBC_USE (IEC_60559_FUNCS_EXT) || __GLIBC_USE (ISOC2X)
      |                 ^
In file included from /usr/include/stdlib.h:56:
/usr/include/x86_64-linux-gnu/bits/floatn.h:30:22: error: missing binary operator before token "("
   30 |      ? __GNUC_PREREQ (4, 3)                                             \
      |                      ^
In file included from /usr/include/x86_64-linux-gnu/bits/floatn.h:120:
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:69:19: error: missing binary operator before token "("
   69 | #if __GNUC_PREREQ (7, 0) && !defined __cplusplus
      |                   ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:90:22: error: missing binary operator before token "("
   90 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:98:22: error: missing binary operator before token "("
   98 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:110:22: error: missing binary operator before token "("
  110 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:118:22: error: missing binary operator before token "("
  118 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:148:22: error: missing binary operator before token "("
  148 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:156:22: error: missing binary operator before token "("
  156 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:168:22: error: missing binary operator before token "("
  168 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:176:22: error: missing binary operator before token "("
  176 | #  if !__GNUC_PREREQ (7, 0) || (defined __cplusplus && !__GNUC_PREREQ (13, 0))
      |                      ^
/usr/include/x86_64-linux-gnu/bits/floatn-common.h:213:22: error: missing binary operator before token "("
[…]
/usr/include/stdlib.h:134:35: error: missing binary operator before token "("
  134 | #if __HAVE_FLOAT16 && __GLIBC_USE (IEC_60559_TYPES_EXT)
      |                                   ^
[…]
```

```
$ dpkg -S /usr/include/x86_64-linux-gnu/c++/12/bits/c++config.h
libstdc++-12-dev:amd64: /usr/include/x86_64-linux-gnu/c++/12/bits/c++config.h
$ dpkg -l libstdc++-12-dev | grep ^ii
ii  libstdc++-12-dev:amd64 12.2.0-14    amd64        GNU Standard C++ Library v3 (development files)
```

### OpenWrt version

230a2b8a5e (tools/cmake: update to 3.27.9)

### OpenWrt release

230a2b8a5e (tools/cmake: update to 3.27.9)

### OpenWrt target/subtarget

CONFIG_TARGET_realtek_rtl838x_DEVICE_zyxel_gs1900-24e=y

### Device

Zyxel GS1900HP

### Image kind

Self-built image

### Steps to reproduce

    $ make -j1 V=sc

### Actual behaviour

Build fails.

### Expected behaviour

No failure.

### Diffconfig

```text
$ ./scripts/diffconfig.sh
CONFIG_TARGET_realtek=y
CONFIG_TARGET_realtek_rtl838x=y
CONFIG_TARGET_realtek_rtl838x_DEVICE_zyxel_gs1900-24e=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
# CONFIG_BUSYBOX_CONFIG_RESIZE is not set
# CONFIG_BUSYBOX_CONFIG_TASKSET is not set
CONFIG_KERNEL_IO_URING=y
# CONFIG_PACKAGE_libmbedtls is not set
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-wolfssl=y
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_zyxel-bootconfig=m
CONFIG_WOLFSSL_HAS_NO_HW=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-08:
yor host compiler is broken, verify the erring file, not some random other file. You need to fix host compiler to build cross-compilers or download sdk.

paulmenzel, 2023-12-08:
Sorry, I forgot to mention, that it’s a regression. `bin/targets/realtek/rtl838x/version.buildinfo` says that it build fine with r24328-91169898ce.

    $ cat bin/targets/realtek/rtl838x/config.buildinfo
    CONFIG_TARGET_realtek=y
    CONFIG_TARGET_realtek_rtl838x=y
    CONFIG_TARGET_realtek_rtl838x_DEVICE_zyxel_gs1900-24e=y
    CONFIG_BUSYBOX_CUSTOM=y
    # CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
    # CONFIG_BUSYBOX_CONFIG_RESIZE is not set
    # CONFIG_BUSYBOX_CONFIG_TASKSET is not set
    CONFIG_KERNEL_IO_URING=y
    CONFIG_PACKAGE_kmod-crypto-acompress=y
    CONFIG_PACKAGE_kmod-lib-lzo=y
    # CONFIG_PACKAGE_libmbedtls is not set
    # CONFIG_PACKAGE_libustream-mbedtls is not set
    CONFIG_PACKAGE_libustream-wolfssl=y
    CONFIG_PACKAGE_libwolfssl=y
    CONFIG_PACKAGE_zyxel-bootconfig=m
    CONFIG_WOLFSSL_HAS_NO_HW=y
    CONFIG_ZRAM_DEF_COMP_LZORLE=y
    CONFIG_PACKAGE_kmod-crypto-acompress=y
    CONFIG_PACKAGE_kmod-lib-lzo=y
    # CONFIG_TARGET_realtek_rtl838x_DEVICE_hpe_1920-8g-poe is not set
    # CONFIG_ZRAM_DEF_COMP_LZ4 is not set
    # CONFIG_ZRAM_DEF_COMP_LZ4HC is not set
    # CONFIG_ZRAM_DEF_COMP_LZO is not set
    CONFIG_ZRAM_DEF_COMP_LZORLE=y
    # CONFIG_ZRAM_DEF_COMP_ZSTD is not set

brada4, 2023-12-08:
broken file in title does not regeress from openwrt.

robimarko, 2023-12-08:
I have seen similar issues when toolchain components in OpenWrt get updated, and the whole toolchain needs to be rebuilt

closed 2023-12-08 by paulmenzel