openwrt/openwrt #14186: 001-bootimage-path-option.patch.orig seems to be outdated when building the SNAPSHOT source for ipq40xx/mikrotik

janos666, 2023-12-09:
### Describe the bug

make[3]: Entering directory '/home/openwrt/openwrt/tools/kernel2minor'
. /home/openwrt/openwrt/include/shell.sh; xzcat /home/openwrt/openwrt/dl/kernel2minor-0.25.tar.xz | tar -C /home/openwrt/openwrt/build_dir/host/kernel2minor-0.25/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/openwrt/openwrt/build_dir/host/kernel2minor-0.25

Applying /home/openwrt/openwrt/tools/kernel2minor/patches/001-bootimage-path-option.patch using plaintext:
patching file kernel2minor.c

Applying /home/openwrt/openwrt/tools/kernel2minor/patches/001-bootimage-path-option.patch.orig using plaintext:
patching file kernel2minor.c
Hunk #1 FAILED at 38.
Hunk #2 FAILED at 56.
Hunk #3 FAILED at 83.
Hunk #4 FAILED at 365.
Hunk #5 FAILED at 592.
Hunk #6 FAILED at 601.
6 out of 6 hunks FAILED -- saving rejects to file kernel2minor.c.rej
Patch failed!  Please fix /home/openwrt/openwrt/tools/kernel2minor/patches/001-bootimage-path-option.patch.orig!
make[3]: *** [Makefile:35: /home/openwrt/openwrt/build_dir/host/kernel2minor-0.25/.prepared52c835d19d5914062e7c171ea8cbaf4c_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/openwrt/openwrt/tools/kernel2minor'
time: tools/kernel2minor/compile#0.03#0.03#0.07
    ERROR: tools/kernel2minor failed to build.
make[2]: *** [tools/Makefile:228: tools/kernel2minor/compile] Error 1
make[2]: Leaving directory '/home/openwrt/openwrt'
make[1]: *** [tools/Makefile:224: /home/openwrt/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynyynyynnynnyyyyyynyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/openwrt/openwrt'
make: *** [/home/openwrt/openwrt/include/toplevel.mk:232: world] Error 2

### OpenWrt version

r24537-889a50e266

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

mikrotik_hap-ac2

### Image kind

Self-built image

### Steps to reproduce

Use the attached diffconfig. Other target like ath79/generic (for tp-link or Mikrotik) and /tiny (for Netgear) build successfully with very similar .config (besides the target/device settings).

### Additional info

I copied the version of a running OpenWRT that was built successfully for some other target device. I tried to update the sources today but the afformentioned target won't be built still.

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_mikrotik=y
CONFIG_TARGET_ipq40xx_mikrotik_DEVICE_mikrotik_hap-ac2=y
CONFIG_DEVEL=y
CONFIG_ATH10K_LEDS=y
CONFIG_BUILD_PATENTED=y
CONFIG_CLEAN_IPKG=y
# CONFIG_DROPBEAR_ED25519 is not set
CONFIG_EXTRA_OPTIMIZATION=""
CONFIG_IMAGEOPT=y
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_AIO is not set
CONFIG_KERNEL_CC_STACKPROTECTOR_NONE=y
# CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_FANOTIFY is not set
# CONFIG_KERNEL_FHANDLE is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_SECCOMP is not set
CONFIG_KERNEL_SQUASHFS_FRAGMENT_CACHE_SIZE=1
# CONFIG_KERNEL_STACKPROTECTOR is not set
# CONFIG_KERNEL_SWAP is not set
CONFIG_LUCI_SRCDIET=y
CONFIG_MOLD=y
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
CONFIG_PACKAGE_ath10k-firmware-qca4019=y
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_dawn=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_hostapd-mbedtls=y
# CONFIG_PACKAGE_jansson is not set
# CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers is not set
CONFIG_PACKAGE_kmod-ath10k-smallbuffers=y
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
# CONFIG_PACKAGE_kmod-hwmon-core is not set
# CONFIG_PACKAGE_kmod-leds-gpio is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-lib-crc32c is not set
# CONFIG_PACKAGE_kmod-nf-conntrack is not set
# CONFIG_PACKAGE_kmod-nf-conntrack6 is not set
# CONFIG_PACKAGE_kmod-nf-flow is not set
# CONFIG_PACKAGE_kmod-nf-log is not set
# CONFIG_PACKAGE_kmod-nf-log6 is not set
# CONFIG_PACKAGE_kmod-nf-nat is not set
# CONFIG_PACKAGE_kmod-nf-reject is not set
# CONFIG_PACKAGE_kmod-nf-reject6 is not set
# CONFIG_PACKAGE_kmod-nfnetlink is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-random-core=y
# CONFIG_PACKAGE_kmod-usb3 is not set
CONFIG_PACKAGE_libgcrypt=y
CONFIG_PACKAGE_libgpg-error=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmnl is not set
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libubus-lua=y
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-dawn=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_ntpclient=y
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_swconfig=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_umdns=y
# CONFIG_PACKAGE_usign is not set
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PKG_ASLR_PIE_NONE=y
# CONFIG_PKG_ASLR_PIE_REGULAR is not set
CONFIG_PKG_CC_STACKPROTECTOR_NONE=y
# CONFIG_PKG_CC_STACKPROTECTOR_REGULAR is not set
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_NONE=y
# CONFIG_PKG_RELRO_FULL is not set
CONFIG_PKG_RELRO_NONE=y
CONFIG_PREINITOPT=y
# CONFIG_SECCOMP is not set
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
CONFIG_TARGET_DEFAULT_LAN_IP_FROM_PREINIT=y
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_PREINIT_IP="192.168.1.2"
CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=1024
CONFIG_USE_GC_SECTIONS=y
CONFIG_MOLD=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
# CONFIG_USE_MOLD is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-09:
That patch-orig is from where?

janos666, 2023-12-10:
> That patch-orig is from where?

Oh, sorry! My bad! I though I removed all custom files after I got this error (I haven't used Git for some 5+ years) but it's something I added manually (https://patch-diff.githubusercontent.com/raw/john-tho/openwrt/pull/7.diff).

closed 2023-12-10 by janos666