openwrt/openwrt #14188: opkg: /usr/lib/opkg/status - broken Packages: line causes segfault

voltagex, 2023-12-10:
### Describe the bug

When my router's root partition ran out of space mid install, `/usr/lib/opkg/status` was corrupted.
This caused opkg update and opkg install (and probably others) to break.

### OpenWrt version

23.05.0 and 23.05.2

### OpenWrt release

r23497-6637af95aa and r23630-842932a63d

### OpenWrt target/subtarget

armsr/armv8 and x86/64

### Device

Traverse Ten64, reproduced on x86_64 VM

### Image kind

Official downloaded image

### Steps to reproduce

To break opkg:

`printf '\nPackage:\n' >> /usr/lib/opkg/status`

### Actual behaviour

```
root@OpenWrt:~# opkg --verbosity=4 update
opkg_conf_parse_file: Loading conf file /etc/opkg.conf.
opkg_conf_parse_file: Loading conf file /etc/opkg/customfeeds.conf.
opkg_conf_parse_file: Loading conf file /etc/opkg/distfeeds.conf.
pkg_hash_load_status_files:
flag_related_packages: propagating pkg flag to dependent abpkg libc
pkg_vec_insert_merge: Adding new pkg=terminfo version=6.4 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=liblucihttp0 version=2023-03-15-9b5b683f arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg libubox20230523
flag_related_packages: propagating pkg flag to dependent abpkg libubus20230605
pkg_vec_insert_merge: Adding new pkg=cgi-io version=2022-08-10-901b0f04 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg libgcc
pkg_vec_insert_merge: Adding new pkg=libc version=1.2.4 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg libf2fs6
flag_related_packages: propagating pkg flag to conflicting abpkg mkf2fs-selinux
pkg_vec_insert_merge: Adding new pkg=mkf2fs version=1.16.0 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg uclient-fetch
flag_related_packages: propagating pkg flag to dependent abpkg libpthread
pkg_vec_insert_merge: Adding new pkg=opkg version=2022-02-24-d038e5b6 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg kernel
flag_related_packages: propagating pkg flag to dependent abpkg kmod-nft-core
flag_related_packages: propagating pkg flag to dependent abpkg kmod-nf-ipt
pkg_vec_insert_merge: Adding new pkg=kmod-nft-compat version=5.15.134 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg opkg
pkg_vec_insert_merge: Adding new pkg=luci-app-opkg version=git-23.311.75635 arch=all.
flag_related_packages: propagating pkg flag to dependent abpkg libblobmsg-json20230523
flag_related_packages: propagating pkg flag to dependent abpkg ubusd
pkg_vec_insert_merge: Adding new pkg=ubus version=2023-06-05-f787c97b arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg libgcc1
pkg_vec_insert_merge: Adding new pkg=libpthread version=1.2.4 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=libuci20130104 version=2023-08-10-5781664d arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg kmod-mdio
flag_related_packages: propagating pkg flag to dependent abpkg kmod-ptp
flag_related_packages: propagating pkg flag to dependent abpkg kmod-hwmon-core
flag_related_packages: propagating pkg flag to dependent abpkg kmod-libphy
flag_related_packages: propagating pkg flag to dependent abpkg kmod-mdio-devres
[... 277 lines cut ...]
pkg_vec_insert_merge: Adding new pkg=kmod-igb version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=kmod-igc version=5.15.134 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg bnx2-firmware
pkg_vec_insert_merge: Adding new pkg=kmod-bnx2 version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=kmod-i2c-algo-bit version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=kmod-mdio-devres version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=urngd version=2023-11-01-44365eb1 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=kmod-slhc version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=rpcd-mod-rrdns version=20170710 arch=x86_64.
flag_related_packages: propagating pkg flag to dependent abpkg kmod-pppoe
pkg_vec_insert_merge: Adding new pkg=ppp-mod-pppoe version=2.4.9.git-2021-01-04 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=kmod-nf-tproxy version=5.15.134 arch=x86_64.
pkg_vec_insert_merge: Adding new pkg=libevent2-core7 version=2.1.12 arch=x86_64.
Segmentation fault
```

### Expected behaviour

opkg to either:
* Warn about the broken file and fix it
* Warn about the broken file and instruct the user to fix it

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

voltagex, 2023-12-10:
OK that bot isn't that helpful

brada4, 2023-12-10:
How to fix:
`opkg --verbose=4 list-installed` to identify offending package
remove files pertaining that package and re-install it.

voltagex, 2023-12-10:
@brada4 not really - if list-installed also segfaults. If I have time I will work out how to build a debug version of opkg and throw it into gdb.

brada4, 2023-12-10:
You wont be able to install it without fixing package metadata.....

ynezz, 2023-12-21:
> If I have time I will work out how to build a debug version of opkg and throw it into gdb.

https://openwrt.org/docs/guide-developer/debugging

voltagex, 2024-01-04:
Gotcha.

```
(gdb) !
/usr/lib/opkg # printf '\nPackage:\n' >> /usr/lib/opkg/status
/usr/lib/opkg # 
(gdb) r update
Starting program: /usr/bin/opkg update
warning: Error disabling address space randomization: Operation not permitted

Program received signal SIGSEGV, Segmentation fault.
djb2_hash (str=str@entry=0x0) at /src/opkg-lede/libopkg/hash_table.c:29
warning: 29	/src/opkg-lede/libopkg/hash_table.c: No such file or directory
(gdb) bt
#0  djb2_hash (str=str@entry=0x0) at /src/opkg-lede/libopkg/hash_table.c:29
#1  hash_index (hash=hash@entry=0x534288 <_conf+328>, key=key@entry=0x0)
    at /src/opkg-lede/libopkg/hash_table.c:36
#2  0x000000000041051a in hash_table_get (hash=0x534288 <_conf+328>, key=0x0)
    at /src/opkg-lede/libopkg/hash_table.c:102
#3  0x000000000040ea5d in abstract_pkg_fetch_by_name (pkg_name=<optimized out>)
    at /src/opkg-lede/libopkg/pkg_hash.h:42
#4  pkg_parse_line (ptr=0x119aaf0, line=0x1197880 "Package:", mask=4286578623)
    at /src/opkg-lede/libopkg/pkg_parse.c:286
#5  0x00000000004109d7 in parse_from_stream_nomalloc (
    parse_line=0x40e483 <pkg_parse_line>, item=item@entry=0x119aaf0, 
    fp=fp@entry=0x1191830, mask=mask@entry=0, buf0=buf0@entry=0x7ffe2e168578, 
    buf0len=buf0len@entry=4096) at /src/opkg-lede/libopkg/parse_util.c:162
#6  0x000000000040df2d in pkg_hash_add_from_file (file_name=<optimized out>, 
    src=src@entry=0x0, dest=dest@entry=0x1196a80, 
    is_status_file=is_status_file@entry=1, state_flags=state_flags@entry=512, 
    cb=cb@entry=0x0, priv=0x0) at /src/opkg-lede/libopkg/pkg_hash.c:104
#7  0x000000000040e1fa in pkg_hash_load_status_files (cb=cb@entry=0x0, 
    priv=priv@entry=0x0) at /src/opkg-lede/libopkg/pkg_hash.c:204
#8  0x000000000040227f in main (argc=2, argv=0x7ffe2e1688e8)
    at /src/opkg-lede/src/opkg-cl.c:455

```

brada4, 2024-01-07:
Hash table returns null value?

laozhoubuluo, 2025-02-04:
This issue is confirmed to be reproducible in the following versions:

# OpenWrt version
23.05.0

# OpenWrt release
r23497-6637af95aa

# OpenWrt target/subtarget
x86/64

# Device
x86_64 VM

# Image kind
Official downloaded image

KanjiMonster, 2025-02-04:
The issue is that `parse_simple("Package", line)` returns a NULL pointer (if the string behind "Package:" is empty), which is then passed on to `abstract_pkg_fetch_by_name()`, which then eventually causes the segfault.

The question is if it is worth the effort to fix it, especially since this likely isn't the only place where it will break with empty/missing fields.

This is essentially a fatal, non-recoverable situation. There is no way to regenerate the package index, or to know what is missing in the index, and what is left in the filesystem. The only way to get in a known, good state is resetting the overlay, and that will also fix the status again.

laozhoubuluo, 2025-02-04:
> The issue is that `parse_simple("Package", line)` returns a NULL pointer (if the string behind "Package:" is empty), which is then passed on to `abstract_pkg_fetch_by_name()`, which then eventually [...]

It can be confirmed that the issue pointed out in the issue is not the only one that causes the problem. The last line of the fault I investigated today was Auto-Installed field not written completely due to insufficient disk, causing segfault.

Normal file end:
> Package: ip6tables-zz-legacy
> Version: 1.8.8-1
> Depends: libc, kmod-ip6tables, xtables-legacy
> Provides: ip6tables, ip6tables-legacy
> Status: install ok installed
> Architecture: x86_64
> Installed-Time: 1721554486
> Auto-Installed: yes

End of file that causes segfault:
> Package: ip6tables-zz-legacy
> Version: 1.8.8-1
> Depends: libc, kmod-ip6tables, xtables-legacy
> Provides: ip6tables, ip6tables-legacy
> Status: install ok installed
> Architecture: x86_64
> Installed-Time: 1721554486
> Auto-Installed: