openwrt/openwrt #14190: rockhip nanopi r4s: sysupgrade - Image metadata not present

kmarzic, 2023-12-10:
### Describe the bug

Hi,
Sysupgrade is throwing the following error:

- FriendlyElec NanoPi R4S:

```
root@np1:/tmp# sysupgrade -v /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img
Sat Dec  9 19:20:01 CET 2023 upgrade: Image metadata not present
Sat Dec  9 19:20:01 CET 2023 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
Sat Dec  9 19:20:01 CET 2023 upgrade: Reading partition table from bootdisk...
Sat Dec  9 19:20:01 CET 2023 upgrade: Reading partition table from image...
Image check failed.
root@np1:/tmp#
```

```
root@np1:~# ubus call system board
{
        "kernel": "6.1.65",
        "hostname": "np1",
        "system": "ARMv8 Processor rev 4",
        "model": "FriendlyElec NanoPi R4S",
        "board_name": "friendlyarm,nanopi-r4s",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r24600-9291f6025f",
                "target": "rockchip/armv8",
                "description": "OpenWrt SNAPSHOT r24600-9291f6025f"
        }
}
root@np1:~#
```

- FriendlyElec NanoPi R2S:

```
root@np0:~# sysupgrade -v /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img
Wed Dec  6 00:16:16 CET 2023 upgrade: Image metadata not present
Wed Dec  6 00:16:16 CET 2023 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
Wed Dec  6 00:16:16 CET 2023 upgrade: Reading partition table from bootdisk...
Wed Dec  6 00:16:17 CET 2023 upgrade: Reading partition table from image...
Image check failed.
root@np0:~#
```

```
root@np0:~# ubus call system board
{
        "kernel": "6.1.65",
        "hostname": "np0",
        "system": "ARMv8 Processor rev 4",
        "model": "FriendlyElec NanoPi R2S",
        "board_name": "friendlyarm,nanopi-r2s",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r24586-f3cdc9f988",
                "target": "rockchip/armv8",
                "description": "OpenWrt SNAPSHOT r24586-f3cdc9f988"
        }
}
root@np0:~#
```

Images have valid sha256 checksum.

### OpenWrt version

r24600-9291f6025f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Self-built image

### Steps to reproduce

Self build image:

- FriendlyElec NanoPi R4S:

```
$ wget https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz
$ tar -Jxf openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz
$ cd openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/
$ sed -i "s/CONFIG_TARGET_ROOTFS_PARTSIZE=.*/CONFIG_TARGET_ROOTFS_PARTSIZE=1024/g" .config
$ sed -i "s/CONFIG_TARGET_KERNEL_PARTSIZE=.*/CONFIG_TARGET_KERNEL_PARTSIZE=32/g" .config
$ make image PROFILE=friendlyarm_nanopi-r4s PACKAGES="\
6in4 adblock banip block-mount bridge bzip2 comgt curl ddns-scripts-cloudflare ddns-scripts-freedns ddns-scripts-noip \
dnscrypt-proxy2 dmesg dropbear e2fsprogs gzip htop ifstat iperf3 ip-bridge ip-full \
kmod-fs-autofs4 kmod-fs-ext4 kmod-fs-msdos kmod-fs-ntfs kmod-tun kmod-usb-storage-uas kmod-usb2 kmod-usb3 \
lm-sensors ncat nmap nping less liblzo2 \
luci luci-ssl luci-app-adblock luci-app-advanced-reboot luci-app-banip luci-app-bcp38 luci-app-ddns luci-app-openvpn luci-app-sqm \
luci-app-statistics luci-app-vnstat2 luci-proto-wireguard \
mkf2fs mailsend netdata netperf ntfs-3g openvpn-openssl openssl-util siproxd sqm-scripts sqm-scripts-extra stubby \
tcpdump transmission-cli transmission-daemon transmission-remote unrar unzip vim vim-runtime vnstat2 wireguard-tools wget-ssl xz-utils"
$ scp -O bin/targets/rockchip/armv8/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz root@np1:/tmp
```

```
root@np1:~# cd /tmp && gzip -d openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz
root@np1:~# sysupgrade -v /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img
```

- FriendlyElec NanoPi R2S:

```
$ wget https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz
$ tar -Jxf openwrt-imagebuilder-rockchip-armv8.Linux-x86_64.tar.xz
$ cd openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/
$ sed -i "s/CONFIG_TARGET_ROOTFS_PARTSIZE=.*/CONFIG_TARGET_ROOTFS_PARTSIZE=256/g" .config
$ sed -i "s/CONFIG_TARGET_KERNEL_PARTSIZE=.*/CONFIG_TARGET_KERNEL_PARTSIZE=32/g" .config
$ make image PROFILE=friendlyarm_nanopi-r2s PACKAGES="\
6in4 adblock banip block-mount bridge bzip2 comgt curl ddns-scripts-cloudflare ddns-scripts-freedns ddns-scripts-noip \
dnscrypt-proxy2 dmesg dropbear e2fsprogs gzip htop ifstat iperf3 ip-bridge ip-full \
kmod-fs-autofs4 kmod-fs-ext4 kmod-fs-msdos kmod-fs-ntfs kmod-tun kmod-usb-storage-uas kmod-usb2 kmod-usb3 \
lm-sensors ncat nmap nping less liblzo2 \
luci luci-ssl luci-app-adblock luci-app-advanced-reboot luci-app-banip luci-app-bcp38 luci-app-ddns luci-app-openvpn luci-app-sqm \
luci-app-statistics luci-app-vnstat2 luci-proto-wireguard \
mkf2fs mailsend netdata netperf ntfs-3g openvpn-openssl openssl-util siproxd sqm-scripts sqm-scripts-extra stubby \
tcpdump transmission-cli transmission-daemon transmission-remote unrar unzip vim vim-runtime vnstat2 wireguard-tools wget-ssl xz-utils"
$ scp -O bin/targets/rockchip/armv8/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz root@np0:/tmp
```

```
root@np0:~# cd /tmp && gzip -d openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
root@np0:~# sysupgrade -v /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img
```

### Additional info

Problem is noted after 2023-12-01, after revision r24550-f909059b74.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

blocktrron, 2023-12-10:
@kmarzic @brada4 This is not a bug, please call `sysupgrade` with the gzipped image and don't extract it before upgrading. Extracting will omit the image metadata.

Please report back if this fixes your problem.

kmarzic, 2023-12-11:
Hi,
Yes, `sysupgrade` using  `gz` files (without extracting files on device) is working.
Thanks for help!

```
root@np0:~# sysupgrade -i -v --test /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img
Wed Dec  6 00:17:03 CET 2023 upgrade: Image metadata not present
Wed Dec  6 00:17:03 CET 2023 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
Wed Dec  6 00:17:03 CET 2023 upgrade: Reading partition table from bootdisk...
Wed Dec  6 00:17:03 CET 2023 upgrade: Reading partition table from image...
Image check failed.
root@np0:~#
```

```
root@np0:~# sysupgrade -v --test /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
Wed Dec  6 00:18:53 CET 2023 upgrade: Reading partition table from bootdisk...
Wed Dec  6 00:18:53 CET 2023 upgrade: Reading partition table from image...
root@np0:~#
```

```
root@np0:~# sysupgrade -v /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
Wed Dec  6 00:19:15 CET 2023 upgrade: Reading partition table from bootdisk...
Wed Dec  6 00:19:16 CET 2023 upgrade: Reading partition table from image...
Wed Dec  6 00:19:16 CET 2023 upgrade: Saving config files...
etc/adblock/adblock.blacklist
etc/adblock/adblock.whitelist
etc/banip/banip.allowlist
etc/collectd.conf
etc/config/adblock
...
etc/uhttpd.key
etc/vnstat_backup.tar.gz
Wed Dec  6 00:19:17 CET 2023 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
root@np0:~# Connection to 10.254.200.1 closed by remote host.
Connection to 10.254.200.1 closed.
```

closed 2023-12-11 by kmarzic