openwrt/openwrt #14192: eth1 "Link is up/Link is down" continuously on TUX-AX4200

FunFR, 2023-12-10:
### Describe the bug

Hi, I've installed Openwrt 23.05.2 on my new Tuf-AX4200 and have a problem with the eth1 device disconnecting unexpectedly.

`mtk_soc_eth 15100000.ethernet eth1: Link is Down`

I didn't have this behavior on my previous Openwrt router.
I've tried to replace the cable but with no more success.
I have activated debug logs on netifd but I have no more info.
I also tried to restore factory firmware and then install 23.05.0 but I got the same result.

Only way to fix the issue is to downgrade link from 1Gbps to 100mbps

`ethtool -s eth1 speed 100 duplex full autoneg on`

Here are Kernel logs:

```
[ 3433.229977] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 3437.385533] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3497.641503] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 3501.796984] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3733.470100] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 3736.586957] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3901.770875] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 3905.926832] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4049.293883] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4053.449367] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4138.639051] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4142.795354] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4208.249948] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4212.404848] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4221.763461] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4224.870185] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4377.583964] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4381.749962] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 4467.967129] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 4471.084013] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 5828.916298] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 5832.033868] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6570.686029] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 6574.841515] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6669.390835] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[ 6672.507514] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
```

And system logs:

```
Mon Dec  4 16:51:55 2023 daemon.notice netifd: Network device 'eth1' link is down
Mon Dec  4 16:51:55 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Dec  4 16:51:55 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Mon Dec  4 16:51:55 2023 kern.info kernel: [ 9111.818425] mtk_soc_eth 15100000.ethernet eth1: Link is Down
Mon Dec  4 16:52:02 2023 daemon.notice netifd: Network device 'eth1' link is up
Mon Dec  4 16:52:02 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Dec  4 16:52:02 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Mon Dec  4 16:52:02 2023 kern.info kernel: [ 9119.090311] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX4200

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-10:
what modes you get from
`ethtool eth1`
in the meantime ethtool -s permits to select multiple link modes or just gige for auto negotiation.

FunFR, 2023-12-10:
Here is the output

```command
root@tuf-ax4200:~# ethtool eth1
Settings for eth1:
	Supported ports: [ ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 100Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 6
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yesl
```

brada4, 2023-12-10:
anything from  --show-eee? Lets assume it is on.
ethtool -S eth1 (attach, any errors, overflows,pauses are potential not always imminent trouble)

`ethtool -s eth1 advertise 0x028` i.e leave gig fdx and 100 fdx and just wait.

what I suspect is opposite end enters EEE mode (you know WoL of desktop PC like powersave) and just sends frame at the speed we are not ready to understand.

doc from manual page:
```
                 0x001                       10baseT Half
                  0x002                       10baseT Full
                  0x004                       100baseT Half
                  0x008                       100baseT Full
                  0x010                       1000baseT Half               (not supported by IEEE standards)
                  0x020                       1000baseT Full
                  0x800000000000              2500baseT Full
```
try some other combos, maybe eee works only at single speed though as you observed.

FunFR, 2023-12-10:
"EEE" doesn't seem to be supported:

```command
root@tuf-ax4200:~# ethtool --show-eee eth1
Cannot get EEE settings: Not supported
```

Here are some stats at 100Mbps:

```command
root@tuf-ax4200:~# ethtool -S eth1
NIC statistics:
     tx_bytes: 1520887142
     tx_packets: 6719301
     tx_skip: 0
     tx_collisions: 0
     rx_bytes: 17513189160
     rx_packets: 12298613
     rx_overflow: 0
     rx_fcs_errors: 0
     rx_short_errors: 0
     rx_long_errors: 4
     rx_checksum_errors: 2
     rx_flow_control_packets: 0
     rx_xdp_redirect: 0
     rx_xdp_pass: 0
     rx_xdp_drop: 0
     rx_xdp_tx: 0
     rx_xdp_tx_errors: 0
     tx_xdp_xmit: 0
     tx_xdp_xmit_errors: 0
     rx_pp_alloc_fast: 12967663
     rx_pp_alloc_slow: 18
     rx_pp_alloc_slow_ho: 0
     rx_pp_alloc_empty: 18
     rx_pp_alloc_refill: 205967
     rx_pp_alloc_waive: 0
     rx_pp_recycle_cached: 0
     rx_pp_recycle_cache_full: 0
     rx_pp_recycle_ring: 13172986
     rx_pp_recycle_ring_full: 0
     rx_pp_recycle_released_ref: 150
```

Errors appear again after applying: `ethtool -s eth1 advertise 0x028`

```
[15650.918275] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[15654.053549] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16085.192724] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16088.309385] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16094.543592] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16098.698333] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16125.709974] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16128.837282] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16247.259716] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16254.541849] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16450.892469] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16455.038799] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16663.856829] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16668.013577] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16714.762687] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16718.928502] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16897.606764] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16901.762221] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16911.121420] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16915.267924] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[16987.997966] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[16991.115796] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[17269.540891] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[17272.648098] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[17654.961001] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[17658.078059] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[17766.122229] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[17770.278446] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
```

I will try different combinations.

For information, the opposite end is a `Freebox revolution` from a French ISP in `bridge` mode. 

Here are some stats at 1Gbps:

```command
root@tuf-ax4200:~# ethtool -S eth1
NIC statistics:
     tx_bytes: 1724941488
     tx_packets: 7669518
     tx_skip: 0
     tx_collisions: 0
     rx_bytes: 20019141386
     rx_packets: 14126046
     rx_overflow: 0
     rx_fcs_errors: 0
     rx_short_errors: 0
     rx_long_errors: 4
     rx_checksum_errors: 7
     rx_flow_control_packets: 0
     rx_xdp_redirect: 0
     rx_xdp_pass: 0
     rx_xdp_drop: 0
     rx_xdp_tx: 0
     rx_xdp_tx_errors: 0
     tx_xdp_xmit: 0
     tx_xdp_xmit_errors: 0
     rx_pp_alloc_fast: 14909980
     rx_pp_alloc_slow: 509
     rx_pp_alloc_slow_ho: 0
     rx_pp_alloc_empty: 509
     rx_pp_alloc_refill: 240100
     rx_pp_alloc_waive: 0
     rx_pp_recycle_cached: 0
     rx_pp_recycle_cache_full: 0
     rx_pp_recycle_ring: 15118256
     rx_pp_recycle_ring_full: 0
     rx_pp_recycle_released_ref: 31821
```

brada4, 2023-12-10:
The driver does not support eee status check or change
Just speculation that phy may not be able to receive 10mbps frame from that box, I am trying to mask out those that phy will not be able (does not claim to be...) receive.
(thats dsa port on mtk soc, that indeed permits eee control and it is off)
```
root@ax53u:~# ethtool --show-eee lan1
EEE settings for lan1:
        EEE status: disabled
        Tx LPI: 30 (us)
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
```

brada4, 2023-12-10:
In statistics - long packet means mtu > 1500 - if you are using pppoe then check providers' documentation on mtu sizes.
Checksum errors on direct cable - should not happen in normal life, but you tried 2 cables, no danger losing 1 frame per hour.

next to try is 0x20 (only gigabit) , you tried 0x8 well.

if you have a chance try to use one lan port for WAN, it is better instrumented than original WAN port.

FunFR, 2023-12-11:
I tried with 0x20 without more success. ISP box connection is simple dhcp so I get the public IP.

I have also enabled ipv6 with a delegation subnet for private network.

Here is my configuration

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdbb:xxxx:xxxx::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config device
	option name 'lan1'
	option macaddr 'aa:bb:cc:dd:ee:ff'

config device
	option name 'lan2'
	option macaddr 'aa:bb:cc:dd:ee:ff'

config device
	option name 'lan3'
	option macaddr 'aa:bb:cc:dd:ee:ff'

config device
	option name 'lan4'
	option macaddr 'aa:bb:cc:dd:ee:ff'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.x.1'
	option netmask '255.255.255.0'
	option ip6assign '64'

config device
	option name 'eth1'
	option macaddr 'aa:bb:cc:dd:ee:ff'

config interface 'wan'
	option device 'eth1'
	option proto 'dhcp'
	option hostname '*'
	option force_link '1'

config interface 'wan6'
	option device 'eth1'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix 'auto'
	option ip6prefix '2a01:xxxx:xxxx:xxxx::/64'
	option force_link '1'

config route6
	option interface 'wan6'
	option target '2000::/3'
	option gateway 'fe80::xxxx:xxxx:xxxx:xxxx'
```

```
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/xxxx.com/'
	option domain 'xxxx.com'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	list server '208.67.222.222'
	list server '208.67.220.220'
	list rebind_domain 'xxxx.xxxx'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra 'server'
	option ra_default '1'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

FunFR, 2023-12-11:
So I switched eth1 with lan4 device to map wan and wan6 interfaces and there is no more disconnection.

Here are some new datas: 

```command
root@tuf-ax4200:~# ethtool lan4
Settings for lan4:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 4
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes
root@tuf-ax4200:~# ethtool -S lan4
NIC statistics:
     tx_packets: 359634
     tx_bytes: 523440004
     rx_packets: 1084536
     rx_bytes: 1463931805
     TxDrop: 0
     TxCrcErr: 0
     TxUnicast: 523780
     TxMulticast: 46
     TxBroadcast: 3
     TxCollision: 0
     TxSingleCollision: 0
     TxMultipleCollision: 0
     TxDeferred: 0
     TxLateCollision: 0
     TxExcessiveCollistion: 0
     TxPause: 0
     TxPktSz64: 3455
     TxPktSz65To127: 126542
     TxPktSz128To255: 34622
     TxPktSz256To511: 15641
     TxPktSz512To1023: 4154
     Tx1024ToMax: 339415
     TxBytes: 539671170
     RxDrop: 0
     RxFiltering: 0
     RxUnicast: 1083115
     RxMulticast: 6
     RxBroadcast: 1415
     RxAlignErr: 0
     RxCrcErr: 0
     RxUnderSizeErr: 0
     RxFragErr: 0
     RxOverSzErr: 0
     RxJabberErr: 0
     RxPause: 0
     RxPktSz64: 4468
     RxPktSz65To127: 72544
     RxPktSz128To255: 26320
     RxPktSz256To511: 6053
     RxPktSz512To1023: 4593
     RxPktSz1024ToMax: 970559
     RxBytes: 1483453569
     RxCtrlDrop: 0
     RxIngressDrop: 0
     RxArlDrop: 0
```

brada4, 2023-12-11:
--show-eee?
Try to enable eee and see if it glitches suddenly.

brada4, 2023-12-11:
Identify different phy-s just that one line on top of ethtool provided info:
```
mii-tool eth0 -v
...
  product info: vendor 00:07:32, model 0 rev 0
...
```

Type for your 2.5Gbps port needs some magic programming to make it function correctly.

@nbd168 is among developers of the driver for both ports.

FunFR, 2023-12-11:
EEE is not supported on lan4

```command
root@tuf-ax4200:~# ethtool --show-eee lan4
EEE Settings for lan4:
	EEE status: not supported
````

Here is the current mii-tool output. I will try to switch to other lanX and also switch back to eth1

```command
root@tuf-ax4200:~# mii-tool lan4 -v
lan4: negotiated 1000baseT-FD flow-control, link ok
  product info: vendor 00:e8:a5, model 4 rev 1
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 1000baseT-HD 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
```

brada4, 2023-12-11:
Only "product info" line is important because it is not shown by ethtool. You dont have to connect anything to flaky port.

brada4, 2023-12-11:
To have non-destructive cable replug create br-wan with those 2 ports, just take care to not loop them eg by not connecting both at once, and bring the dhcp "wan" interfaces to that br-wan.
also dhcp will not "see" the cable up/down at the other end of bridge.

namiltd, 2023-12-11:
https://github.com/openwrt/openwrt/issues/14146

philS1, 2023-12-11:
``mii-tool eth0 -v
SIOCGMIIREG on eth0 failed: Invalid argument
SIOCGMIIREG on eth0 failed: Invalid argument
SIOCGMIIREG on eth0 failed: Invalid argument
eth0: 1000 Mbit, half duplex, link ok
  product info: vendor 19:f2:77, model 33 rev 0
  basic mode:   100 Mbit, half duplex
  basic status: link ok
  capabilities:
  advertising:  1000baseT-HD 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 1000baseT-HD 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
``

brada4, 2023-12-17:
Asus AX53U
`product info: vendor 00:e8:a5, model 1 rev 2`
same type on both dsa and wan ports, works flawlessly

brada4, 2024-02-24:
Phy type with mii-tool -v
ethtool report of negotiation - enough to confirm best speed is agreed
if fixing ethtool -s advertise to 1g fdx helps
wild errors in ethtool -S vs below:
workaround use a vacant an port esp in case phy type is different

PS there is no 100% fix, not even idea under what conditions it flipflops.

brada4, 2024-02-24:
Gigabit is always duplex, mii-tool is just to read phy type.
What phy type is on lan ports?
There are very frw errorsm ideally must be zero.
You need to set gigabit (0x20) and wait if it still sporadically disconnects.
If it does try 0x8 100m fdx, if this is stable then phy cannot sustain eee causing transmissions at other speed.

brada4, 2024-02-28:
You can formally try to disable eee, though seems dependent on ethtool support
https://git.openwrt.org/?p=project/netifd.git;a=commitdiff;h=eee02ccca8c8ba2540108e6a5cff10888bb337c9

TacticalCode, 2024-05-26:
I'm having the same symptoms with a Zyxel NWA50AX-Pro access point: Link goes down and up again, no further messages in dmesg.
It's a 2.5GbE port, connected to my Sophos SG 1GbE (also running OpenWRT). I can reproduce it with iperf from a wifi client (connected to the Zyxel) to the Sophos firewall, but not when running iperf from client to zyxel or from zyxel to sophos.
Workaround is to set the link to 1GBit Half duplex and autonegotiation off: `ethtool -s eth0 speed 1000 duplex half autoneg off`
This also disables Flow Control and the link stays up. `ethtool -a` is not supported on my hardware.

Before:

> Sun May 26 14:35:49 2024 kern.info kernel: [241782.606958] mtk_soc_eth 15100000.ethernet eth0: Link is Down
> Sun May 26 14:35:49 2024 kern.info kernel: [241782.613681] br-lan: port 1(eth0) entered disabled state
> Sun May 26 14:35:49 2024 daemon.notice netifd: Network device 'eth0' link is down
> Sun May 26 14:35:53 2024 kern.info kernel: [241786.766111] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
> Sun May 26 14:35:53 2024 kern.info kernel: [241786.774671] br-lan: port 1(eth0) entered blocking state
> Sun May 26 14:35:53 2024 kern.info kernel: [241786.779994] br-lan: port 1(eth0) entered forwarding state
> Sun May 26 14:35:53 2024 daemon.notice netifd: Network device 'eth0' link is up
> Sun May 26 14:36:51 2024 kern.info kernel: [241844.991341] mtk_soc_eth 15100000.ethernet eth0: Link is Down
> Sun May 26 14:36:51 2024 kern.info kernel: [241844.997954] br-lan: port 1(eth0) entered disabled state
> Sun May 26 14:36:51 2024 daemon.notice netifd: Network device 'eth0' link is down

With Half-Duplex (autoneg off, flow control off) set on the Zyxel device:

> Sun May 26 14:36:55 2024 kern.info kernel: [241849.150475] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
> Sun May 26 14:36:55 2024 kern.info kernel: [241849.159031] br-lan: port 1(eth0) entered blocking state
> Sun May 26 14:36:55 2024 kern.info kernel: [241849.164351] br-lan: port 1(eth0) entered forwarding state
> Sun May 26 14:36:55 2024 daemon.notice netifd: Network device 'eth0' link is up

Should I piggy-back on this issue, or #14437 or open a separate one?

kurlberg, 2025-01-15:
I had similar problems (tricky to find, basically only sign was lag spikes and my squeebox having lots of rebuffering issues) with a recently purchased Tuf-AX4200. After running 

ethtool -s eth1 advertise 0x028 
ethtool -s eth1 speed 1000 duplex full autoneg off

(I think I needed *both* of them) the problem went away. Knock on wood...  It was extremely bad just prior to the fix:

Wed Jan 15 19:47:59 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 19:55:47 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 19:56:09 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 19:58:21 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 19:59:22 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 20:00:17 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 20:00:32 2025 daemon.notice netifd: Network device 'eth1' link is down
Wed Jan 15 20:02:36 2025 daemon.notice netifd: Network device 'eth1' link is down

But now it's been running without link down for a couple of hours.

Very happy, thank you very much!

kurlberg, 2025-01-15:
PS: I'd be happy to run ethtool / mii-tool and paste the output here in case it might be helpful.

brada4, 2025-01-15:
1/ identify device and OpenWRT version (as seen in `ubus call system board`) 
2/ mii-tool -v lan5 (the phy identifiiers)
3/ ethtool lan5 / ethtool --show-eee lan5 (the link modes)

If you feel strong about your clicking skills - repurpose one of gigabit lan ports as wan as a temporary fixup.

kurlberg, 2025-01-16:
I don't seem to have lan5, so I tried it with eth1 and lan2; lan2 is the
port I connect to my lan; and the only lan having any traffic.

root@OpenWrt:~# ubus call system board
{
	"kernel": "5.15.134",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "ASUS TUF-AX4200",
	"board_name": "asus,tuf-ax4200",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.0",
		"revision": "r23497-6637af95aa",
		"target": "mediatek/filogic",
		"description": "OpenWrt 23.05.0 r23497-6637af95aa"
	}
}

root@OpenWrt:~# mii-tool -v lan5
SIOCGMIIPHY on 'lan5' failed: No such device

root@OpenWrt:~# mii-tool -v eth1
SIOCGMIIREG on eth1 failed: Invalid argument
SIOCGMIIREG on eth1 failed: Invalid argument
SIOCGMIIREG on eth1 failed: Invalid argument
eth1: 1000 Mbit, half duplex, link ok
  product info: vendor 19:f2:77, model 33 rev 0
  basic mode:   10 Mbit, half duplex
  basic status: link ok
  capabilities:
  advertising:  100baseTx-FD
  link partner: 1000baseT-HD 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD

root@OpenWrt:~# mii-tool -v lan2
lan2: negotiated 1000baseT-FD flow-control, link ok
  product info: vendor 00:e8:a5, model 4 rev 1
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 1000baseT-HD 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control

root@OpenWrt:~# ethtool lan5
netlink error: no device matches name (offset 24)
netlink error: No such device
netlink error: no device matches name (offset 24)
netlink error: No such device
netlink error: no device matches name (offset 24)
netlink error: No such device
netlink error: no device matches name (offset 24)
netlink error: No such device
netlink error: no device matches name (offset 24)
netlink error: No such device
No data available

root@OpenWrt:~# ethtool lan2
Settings for lan2:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: slave
	Port: Twisted Pair
	PHYAD: 2
	Transceiver: external
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes

root@OpenWrt:~# ethtool --show-eee lan2
EEE settings for lan2:
	EEE status: not supported

kurlberg, 2025-01-23:
The suggested workaround works very well, not a single problem since implementing it. Thank you *very* much!

BTW, tuf-ax4200 is mentioned in a couple of forums as an awesome openwrt router (which is why I bought it) - perhaps worth telling people about this issue (+ easy fix) in the install instructions? E.g. the one at https://openwrt.org/toh/asus/tuf-ax4200

brada4, 2025-01-23:
Which of 3 offloads is wrong for the rmtk_soc_eth? If there was a problem with kernel bridges it would have been noticed long ago with all dockers crashing.

kurlberg, 2025-01-23:
I'm sorry, I have no idea. If you give me some commands to run to get further info I'd be happy to run them. There was some earlier discussion (see above) that it might be related to the isp side dropping down to 10mbs as some power saving measure, and that the ax4200 does not recover. I've got the wan port hooked up to the isp, and I think it's 2.5 gbps. brada4 wrote above "i am trying to understand if 2.5gbit port is glitching, vs more common gigabit ones."

I have not yet tried to hook up my wan to one of the "internal" lan ports. If you think this data point is important I can give it a try.

kurlberg, 2025-01-23:
PS: about the isp issue: if this is the case, it could be that me and the other guy are just unlucky with a "bad isp"/ax4200 incompatibility issue. So if very rare, perhaps not worth saying something on the wiki. But a link to this thread would have helped me. It was a bit subtle to spot: the  squeezebox is very sensitive to the connection going down briefly (i.e., watching youtube was no problem); the main other issue was bad lag spikes when gaming. In retrospect I should have checked router logs faster.

blocktrron, 2025-04-07:
@TacticalCode Which ethernet controlelr does the Sophos appliance use? I did not find reliable information online about this.

To the others: Which device did you experience these symptoms with as a link partner to the GPY211? I ddi not yet encounter one that i own which produces these symptoms, but I'd like to get them reproduced.

Another question which i did not find the answer to in this thread - Does the problem occur with the stock firmware from ZyXEL / Asus also?

Anteus, 2025-04-07:
I have the issue on Zyxel EX5601-T1 with the media converter Inteno XG6846-R1 as WAN uplink. Below solves it.

```
ethtool -s eth1 advertise 0x028
ethtool -s eth1 speed 1000 duplex full autoneg off
```

blocktrron, 2025-04-08:
@vae-21 just to ensure I got this right - the issue is reproducible when connecting two ax4200 with each other on the wan port?

vae-21, 2025-04-09:
> [@vae-21](https://github.com/vae-21) just to ensure I got this right - the issue is reproducible when connecting two ax4200 with each other on the wan port?

@blocktrron I think I may have missed some part in the discussion above or something, but that is not something I've tried. I only have the one device and it's connected to the wall-hole/internets (via the wan port).

blocktrron, 2025-04-10:
I did have a look today at the datasheet for EEE /  Power related options.

There is a option for disabling the Packet Manager. I don't entirely get it's functions apart from handling flow-control in the 2.5G SGMII fixed mode. I'm unsure whether it affects PTP operation (we don't support currently, so worth a shot).

[image]

There are also options for low-poer modes in Autonegotiation state as well as Link-Quality dependent power consumption scaling settings which are enabled by default. 

[image]

I've created a branch which disables all these options. I was only able to verify nothing broke by this, however i was not able to reproduce these issues before so i can not assess if any of my changes made a difference.

I have prebuilt images with LuCI and ethtool-full installed here: https://files.darmstadt.freifunk.net/s/A36k6Cii3HfQiwK

A branch with  the changes is here: https://github.com/blocktrron/openwrt/tree/pr-nwa50axpro-link

Looking forward to feedback.

blocktrron, 2025-04-25:
I've tested various switches over the last days and had no dice reproducing the issue.

Just to exclude we have a variance in the firmware on the OTP of the chip, please post the Firmware version which can be found in dmesg as well as the link-partner you encounter this issue with.

```
root@OpenWrt:~# dmesg  | grep GPY
[    2.639506] Maxlinear Ethernet GPY211C mdio-bus:06: Firmware Version: 8.111 (0x886F)
[   14.359812] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:06] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
```

Anteus, 2025-04-25:
Zyxel EX5601-T1 with the media converter Inteno XG6846-R1 as WAN uplink
```
 OpenWrt 24.10.1, r28597-0425664679
 -----------------------------------------------------
root@OpenWrt:~# dmesg  | grep GPY
[    7.831654] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.111 (0x886F)
[    7.841107] Maxlinear Ethernet GPY211C mdio-bus:06: Firmware Version: 8.111 (0x886F)
[    8.154892] mt7530-mdio mdio-bus:1f lan1 (uninitialized): PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
[   19.343614] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:06] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
```

blocktrron, 2025-04-27:
I was sent another Bootlog from an NWA50AX Pro which can not even establish a link to a Cisco Router.

```
[    2.969900] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.153 (0x8899)
[    2.973227] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc081780000, irq 75
```

So it seems we need to take this into account moving forward.

gaernik, 2025-05-15:
> I have the issue on Zyxel EX5601-T1 with the media converter Inteno XG6846-R1 as WAN uplink. Below solves it.
> 
> ```
> ethtool -s eth1 advertise 0x028
> ethtool -s eth1 speed 1000 duplex full autoneg off
> ```

This fixed link down/up between rb5009 and nwa50ax pro

blocktrron, 2025-05-15:
@gaernik which device is the link partner of your ZyXEL AP?

gaernik, 2025-05-22:
> [@gaernik](https://github.com/gaernik) which device is the link partner of your ZyXEL AP?

it's a mikrotik RB5009UG+S+IN

blocktrron, 2025-05-22:
@gaernik Thanks this is very helpful. How often / how fast after the initial boot / Link establishment begins the Link to flap? Cna be a rough estimation (10 minutes / 1 hour / 2 days / ...)

blocktrron, 2025-05-24:
@gaernik thanks for the info. I have a test-setup with an older inteno switch but the link stays idle.

What all problematic setups seems to have in common is the use of a marvell ETH phy / switch.

pgaufillet, 2025-08-14:
Very annoying anomaly indeed. I am running 4 Asus TUF AX4200 with the vanilla openwrt 24.10.2 in different Ethernet environments (Freebox, Livebox, Netgear GS108T), and they all behave incorrectly, switching randomly between 100M and 1000M modes on the 2.5G connector.

peterwillcn, 2025-08-27:
ping @dangowrt

FunFR, 2025-08-27:
> Hi, [@FunFR](https://github.com/FunFR) Did you find a solution? I am experiencing the same issue, on Asus TUF AX4200 which acts as a Bridged AP with configured trunk. The device is on OpenWRT 24.10.2. The upsteram device is Mikrotik RB5009UG+S+IN.
> 
> For now I am using one of gig ports but I would prefer to use 2.5G interface as interconnector.
> 
> [@blocktrron](https://github.com/blocktrron) In my case, the interface flaps every few minutes.

No solution. I tried again a few months ago with new firmware, but without success, so I'm still using the workaround, which involves not using the 2.5G connector.

Timinator0815, 2025-10-24:
Did this bug get fixed in the new 24.10.3 release? Not finding anything about it in the release notes.

blocktrron, 2026-03-19:
Hi, so did you do something manually after installing the linked build?

Also what is your link partner?

svfrd, 2026-03-19:
The build is based on v25.12 branch with the applied change on top.

```
--- a/target/linux/mediatek/dts/mt7981b-zyxel-nwa50ax-pro.dts
+++ b/target/linux/mediatek/dts/mt7981b-zyxel-nwa50ax-pro.dts
@@ -85,6 +85,9 @@
                reg = <5>;
                compatible = "ethernet-phy-ieee802.3-c45";

+               eee-broken-100tx;
+               eee-broken-1000t;
+
                leds {
                        #address-cells = <1>;
                        #size-cells = <0>;
```

No additional 'ethtool' settings. Plain network config.
Link partner is a clearfog pro device running 24.10 with a builtin mv88e6085 switch.

blocktrron, 2026-03-19:
The proposed fix in the branch contains this commit also: https://github.com/blocktrron/openwrt/commit/fa63a753bd6230c2c6ba2d381f1a31a11a3028a2

Did you apply these as well?

gladiac, 2026-08-16:
Hello everybody, I also have issues with one of my 4 NWA50AX Pros. 3 of them are older and are running perfectly for years. However, the newest one which also has a way newer serial number (old: S230xxxxxxxx, new: S250xxxxxxxx) has the link flapping problem. I tested many workarounds in this thread. Currently, it seems like disabling EEE TX LPI via
```
# ethtool --set-eee eth0 tx-lpi off
# ethtool --show-eee eth0
EEE Settings for eth0:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
```
fixes the problem and the interface is stable on a 2.5G link. I will monitor the stability for the coming days.

The PHY on my problematic device has firmware version 8.153 (0x8899)
```
# dmesg|grep GPY|grep Firmware\ Version
[    2.981494] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.153 (0x8899)
```
All my non-problematic APs have the same firmware version - so the PHY firmware seems not to be a factor for this problem.

andreas5232, 2026-10-23:
I've been just running into the same issues using a Zyxel WX5600-T0. 

Unfortunately the LAN interfaces are currently unusable, because of a permanent ~90% packet loss due to the flaky MaxLinear GPY211. I was hoping that this had already addressed in the newest OpenWRT Snapshot version, but obviously there is no clean and reliable fix yet.

EDIT: 
The EEE fix does unfortunately not work for my GPY211C (FW 8.111) NIC.
```
root@OpenWrt:~# dmesg|grep GPY|grep Firmware\ Version
[    2.738092] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.111 (0x886F)

root@OpenWrt:~# ethtool --set-eee lan1 tx-lpi off
Cannot get EEE settings: Not supported

root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: lan2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN qlen 1000
    link/ether 90:9f:22:b5:28:b6 brd ff:ff:ff:ff:ff:ff
3: lan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP qlen 1000
    link/ether 90:9f:22:b5:28:b3 brd ff:ff:ff:ff:ff:ff
```

gladiac, 2026-10-24:
> 27\. I've been just running into the same issues using a Zyxel WX5600-T0.
>          Unfortunately the LAN interfaces are currently unusable, because of a permanent ~90% packet loss due to the flaky MaxLinear GPY211. I was hoping that this had already addressed in the newest OpenWRT Snapshot version, but obviously there is no clean and reliable fix yet.

You can try changing the LAN cable. Weirdly enough, what worked for me was changing a half meter cat8.1 cable to a half meter cat6. I have not had a single link instability since then. I do not know why that is though.

andreas5232, 2026-10-25:
>          You can try changing the LAN cable. Weirdly enough, what worked for me was changing a half meter cat8.1 cable to a half meter cat6. I have not had a single link instability since then. I do not know why that is though.

Thanks for the tip! Unfortunately I cannot measure any difference by just changing the cable. I've tried different lengths of cat5e, cat6 and cat7. 

Also changing the speed to 100M makes no real difference. I'm somewhat stuck between 90% and 40% loss at "best" settings. Using a different 2.5G or 1G Switch also didn't help. But when looking at dmesg and the logs it must be something weird with this GPY nic, as all my other devices work flawless with 2.5G and 1G mixed at the same switch.

dmbnex, 2026-10-28:
Using zyxel ex5700 (Telenor) - same issue on 'wan' on both 25.12.5 and 25.12.6. lan4(2.5) not tested.
Lan1-3 work fine

disabling eee and tx-lpi using ethtiol fixes the issue(until reboot)

dmbnex, 2026-10-29:
> 27\. Using zyxel ex5700 (Telenor) - same issue on 'wan' on both 25.12.5 and 25.12.6. lan4(2.5) not tested.
>      Lan1-3 work fine
>      disabling eee and tx-lpi using ethtiol fixes the issue(until reboot)

1G infrastructure
tests are done on 25.12.6
**[    1.121881] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.111 (0x886F)
 [    1.131146] Maxlinear Ethernet GPY211C mdio-bus:06: Firmware Version: 8.111 (0x886F)
 [    2.517121] mt7530-mdio mdio-bus:1f lan4 (uninitialized): PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
 [   15.329566] mtk_soc_eth 15100000.ethernet wan: PHY [mdio-bus:06] driver [Maxlinear Ethernet GPY211C] (irq=POLL)** 

most dropouts are when there is traffic on the **wan** interface.
**wan** @ 100M (forced on switch-side) works without any workarounds

tested **lan4** (@ 1G), is ok without any additional workarounds

Gig-led on **wan** does not work correctly, it only shortly blink when plugged in. 100M led works

henriknil, 2026-10-30:
> Using zyxel ex5700 (Telenor) - same issue on 'wan' on both 25.12.5 and 25.12.6. lan4(2.5) not tested.
> Lan1-3 work fine
> 
> disabling eee and tx-lpi using ethtiol fixes the issue(until reboot)
> 

Is it enough for you to turn off tx-lpi only?
E.g.:
```
ethtool --set-eee wan tx-lpi off
ethtool --show-eee wan

```

dmbnex, 2026-10-30:
> 27\. > Using zyxel ex5700 (Telenor) - same issue on 'wan' on both 25.12.5 and 25.12.6. lan4(2.5) not tested.
>      > Lan1-3 work fine
>      > disabling eee and tx-lpi using ethtiol fixes the issue(until reboot)
>      
>      
>      Is it enough for you to turn off tx-lpi only?
>      E.g.:
>      ```
>      ethtool --set-eee wan tx-lpi off
>      ethtool --show-eee wan
>      ```

No, both have to be disabled.

henriknil, 2026-10-30:
I had the same issue on my GL MT-3000 and have been using Claude Opus to debug it. However I changed my WAN device and the issue is now gone, probably due to the new device not supporting EEE. 
According to Claude the following can pinpoint the issue (and it sounds to me it can):

 ```
apk add mdio-tools
mdio mdio-bus mmd 6:30 raw 12 0x0000/0xfffe
mdio mdio-bus mmd 6:30 raw 12
ethtool -r wan
```

dmbnex, 2026-10-30:
> 27\. I had the same issue on my GL MT-3000 and have been using Claude Opus to debug it. However I changed my WAN device and the issue is now gone, probably due to the new device not supporting EEE.
>      According to Claude the following can pinpoint the issue (and it sounds to me it can):
>      ```
>      apk add mdio-tools
>      mdio mdio-bus mmd 6:30 raw 12 0x0000/0xfffe
>      mdio mdio-bus mmd 6:30 raw 12
>      ethtool -r wan
>      ```

I'm not even close to understand what those change and on which device that would be.
you gave me an idea, I disabled eee on uplink switch port and **wan** became stable, sadly we not always have that possibility

I reverted the change and tried the above commands;
those got through but when I rebooted the router, it gave some kind of call trace, which I had not seen before,
after reboot both **eee** and **tx-lpi** were disabled.

trace call:
**[   20.813314] Call trace:
[   20.813316]  0xffffffc080019294
[   20.813321]  0xffffffc0800192dc
[   20.813323]  0xffffffc080951598
[   20.813326]  0xffffffc0809515c8
[   20.813328]  0xffffffc080280fe0
[   20.813330]  0xffffffc080280b58
[   20.813333]  0xffffffc080281b3c
[   20.813335]  0xffffffc080281d14
[   20.813337]  0xffffffc0804e803c
[   20.813340]  0xffffffc08056e914
[   20.813342]  0xffffffc080280124
[   20.813344]  0xffffffc08027e988
[   20.813347]  0xffffffc0801f39d8
[   20.813349]  0xffffffc0801f3d1c
[   20.813352]  0xffffffc0801f3db8
[   20.813354]  0xffffffc08001eec8
[   20.813357]  0xffffffc08001ef6c
[   20.813359]  0xffffffc0809519d8
[   20.813361]  0xffffffc0809526e0
[   20.813364]  0xffffffc080011530
[   20.900943] br-lan: port 5(phy1-ap0) entered blocking state
[   20.906545] br-lan: port 5(phy1-ap0) entered disabled state
[   20.912160] mt7915e 0000:01:00.0 phy1-ap0: entered allmulticast mode
[   20.918635] mt7915e 0000:01:00.0 phy1-ap0: entered promiscuous mode
[   21.133429] br-lan: port 6(phy2-ap0) entered blocking state
[   21.139013] br-lan: port 6(phy2-ap0) entered disabled state
[   21.144723] mt798x-wmac 18000000.wifi phy2-ap0: entered allmulticast mode
[   21.151785] mt798x-wmac 18000000.wifi phy2-ap0: entered promiscuous mode
[   21.158699] br-lan: port 6(phy2-ap0) entered blocking state
[   21.164301] br-lan: port 6(phy2-ap0) entered forwarding state
[   21.180615] br-lan: port 6(phy2-ap0) entered disabled state
[   22.647552] Ethtool ioctl interface doesn't support passing EEE linkmodes beyond bit 32
[   22.851222] mtk_soc_eth 15100000.ethernet wan: Link is Down
[   27.012072] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control off
[   27.678322] Ethtool ioctl interface doesn't support passing EEE linkmodes beyond bit 32**

the command output:
**root@OpenWrt:~# mdio mdio-bus mmd 6:30 raw 12
0x0003**

**root@OpenWrt:~# mdio mdio-bus mmd 6:30 raw 12 0x0000/0xfffe
root@OpenWrt:~#
root@OpenWrt:~# mdio mdio-bus mmd 6:30 raw 12
0x0002**

...which means it changes something

I went ahead and soft defaulted the router via UCI, reloaded
no trace call
**root@OpenWrt:~# mdio mdio-bus mmd 6:30 raw 12
0x0003**
and see that **eee** is disabled on **wan**, **tx-lpi** is still enabled
it may mean that **eee** was always disabled(Im preatty sure it was enabled when after first flash) or that the command did something on the hardware level

I flashed 25.12.5:
the eee is on tx-lpi is off
23.05.06:
wan does not support eee, the reset problem is still there, but the LED works fine (tested on 1G only)
all 4 lan port have eee disabled and tx-lpi enabled

dmbnex, 2026-10-30:
tested 24.12.8:
wan:
no eee support on wan, LED works, the problem with instability is there.
so clearly something has been changed that killed the LED in 25.12(.5 and .6 at least)

lan:
eee is off and tx-lpi is enabled on all lan1-lan3 and lan4(2.5G)

25.12.6:
wan: no LED, eee is supported and is disabled, tx-lpi is enabled and has to be disabled to avoid the instability, or disable eee on uplink
lan: both eee and tx-lpi are disabled by default on all lan1-lan4 interfaces

dmbnex, 2026-10-30:
on snapshot: SNAPSHOT (r37908-5700101d7c)
**wan** LED does not work, EEE is not supported *cry*
BUT no more link instability on **wan** with same tests as done when failing..

root@OpenWrt:~# ethtool --show-eee wan
Cannot get EEE settings: Not supported
root@OpenWrt:~# ethtool --show-eee lan1
EEE Settings for lan1:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
root@OpenWrt:~# ethtool --show-eee lan2
EEE Settings for lan2:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
root@OpenWrt:~# ethtool --show-eee lan3
EEE Settings for lan3:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  Not reported
root@OpenWrt:~# ethtool --show-eee lan4
EEE Settings for lan4:
        EEE status: enabled - inactive
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  100baseT/Full 
                                    1000baseT/Full 
        Link partner advertised EEE link modes:  Not reported