openwrt/openwrt #14195: Bridge master interface not reachable after enabling VLAN filtering on Banana R3

ffainelli, 2023-12-10:
### Describe the bug

After enable VLAN filtering manually or using Luci, the router's bridge IP address is unreachable from the stations in the network.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Create a br-lan bridge with the following configuration:

`config device
	option type 'bridge'
	option name 'br-lan'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan1:u*'
	list ports 'lan2:u*'
	list ports 'lan3:u*'
	list ports 'lan4:u*'
`

### Actual behaviour

br-lan is created, all of the LAN ports defined are configured as PVID 1 untagged but stations are unable to reach the router's IP address configured on the bridge interface (192.168.1.254/24). The "br-lan" interface as reported by the "bridge v" command shows:

`root@router:~# bridge v
port              vlan-id
eth1              1 PVID Egress Untagged
wan               1 PVID Egress Untagged
lan1              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan4              1 PVID Egress Untagged
sfp2              1 PVID Egress Untagged
br-lan            1
br-wan            1 PVID Egress Untagged
phy1-ap0          1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
`

Using a bridge without VLAN filtering, or using VLAN uppers (e.g.: lan1.<VID>) works as intended.

### Expected behaviour

br-lan with the assigned IP address should be able to ping machines in the same segment which do not expect 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-10:
because br-lan.1 has no ip address, no dhcp,no internet.

ffainelli, 2023-12-10:
My expectation is that we should be making br-lan PVID 1 untagged as well that would be the most natural way of switching a non-VLAN filtering enabled bridge to a VLAN filtering enabled one.

ffainelli, 2023-12-10:
Also some time in 5.15 we did have the ability for DSA drivers to automatically pop the VLAN tag:

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=1dc0408cdf3ca

brada4, 2023-12-10:
Yes, those are bridged to br-lan.1 now. No change in capabilities.

ffainelli, 2023-12-11:
This is really going against the principle of least surprise, as soon as someone clicks "Enable VLAN filtering" you would expect that the current configuration which is with VLAN filtering disabled continues to work, and that there should not be a need to explicitly configure a "br-lan.1" device. This is a major usability issue IMHO.

ffainelli, 2023-12-11:
For instance when toggling "Enable VLAN filtering" in Luci we have the following:

[image]

and it would be a good idea that by default; in the absence of any previously defined VLANs that we automatically get this:

[image]

in addition to making "br-lan" PVID egress untagged in the same one and only VLAN defined because otherwise it gets really confusing. If people want to have the bridge master device VLAN tagged, let them have that through more advanced settings.

brada4, 2023-12-11:
You made dual configuration and are rolling dice which one would apply in part or as a whole.
- old one, utilising VLAN ID 1 and linking untagged ports together
- new one with physical ports untagged re-utilizing VLAN ID 1 but this time bridging physical ports with br-lan.1
If you state what result you are trying to achieve I might be able to help, but that does not involve configuring VLAN ID 0 or 1 if you want to isolate networks for good. Once those are isolated you need to instrument each with like dhcp and at least some IOT-restricted NAT.

ffainelli, 2023-12-11:
The end goal for me is that vlan_filtering is set to 1 under /sys/class/net/br-lan/bridge/vlan_filtering and essentially continuing to have this configuration:

```
root@router:~# bridge v
port              vlan-id
eth1              1 PVID Egress Untagged
wan               1 PVID Egress Untagged
lan1              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan4              1 PVID Egress Untagged
sfp2              1 PVID Egress Untagged
br-lan            1 PVID Egress Untagged
br-wan            1 PVID Egress Untagged
phy1-ap0          1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
```

because as an user of the Luci interface that is what you would expect when you click "Enable VLAN filtering", anything different than that goes against the principle of least surprise. Whether that means that Luci should automatically dispaly a configuration row for the VLAN ID corresponding to "default_pvid" is up for debate I suppose.

It is not IMHO reasonable to expect users to know that they have to create a "br-lan.1" interface to pop/push the VID 1 VLAN tag, even less so when the device that is being configured to enable VLAN filtering is "br-lan". This is really asking too much knowledge from our users. It is also an important detail that is completely missed out in the guide:

https://openwrt.org/docs/guide-user/network/dsa/converting-to-dsa

there is a display of br0.1 but no mention of how to configure it, and nowhere is it to be found in the example /etc/config/network.

And finally, this is completely unnecessary because br-lan can be made PVID egress untagged in its default VLAN!

brada4, 2023-12-11:
It is not much different from swconfig, you have to now make br-lan.vlan in place of eth0.vlan in one place and wire physical ports in other place. Indeed pictured re-use of VLAN1 leads to confusion like yours that one has no idea on which of 2 configs is active to what part.

Basic step-by-step:
create interface br-lan.2
[image]
instrument br-lan.2 with dhcp , forwarding etc, and configure VLAN filtering in br-lan (order of 2 items is not important)

Same amount of clicks as with swconfig, only in different places.

ffainelli, 2023-12-11:
It is not different from swconfig, that I agree with, but it is not necessary at all and because of that it is IMHO confusing.

There are ways to coerce even the most "bogus" DSA switch configurations into popping the bridge master VLAN tag if they can only be configured as egress tagged (with untag_bridge_pvid). I really think we should make it a tad simpler for people to use VLAN filtering with as little configuration as possible.

ffainelli, 2023-12-11:
I suppose my question is whether netifd can create a bridge that is "PVID egress untagged" in its default_pvid? Is that even supported without resorting to a manual configuration?

brada4, 2023-12-11:
If you throw 2 VLAN-s at it it is not possible physically.

ffainelli, 2023-12-11:
If someone wants to be unable to differentiate traffic by throwing 2 "untagged" VLANs that is their problem and they should be allowed to do that, because the bridge configuration allows you to do that as long as the CPU port of the switch allows you to do that, which most switches allow.

It does not look like you see this as a big usability issue, while I really do.

brada4, 2023-12-11:
I think it is order of parsing network file, you add untagged physical port to bridge, then override it with untagged other vlan port. If you swap sections it goes in reverse.... (keep one port out of experiments leaving to br-lan only)

ffainelli, 2024-01-16:
From the web interface, how would an user even remotely know that their 'lan' interface physical device must now be 'br-lan.1'? There are no hints being provided at all.

Also, why not default to VLAN filtering enabled right off the bat?

brada4, 2024-01-16:
Not all adapters offload VLAN-s (ethtool -k), so in some cases that is slower (like add memcpy in and memcpy out) than nothing.

ffainelli, 2024-01-17:
OK, but the default configuration prior to DSA was to use VLAN tags unconditionally, so it would not be massively different. Anyway, I still don't understand how this can be so difficult to achieve out of the box, it is a steep learning curve for most users and it can trip people who incidentally even worked on the DSA framework itself, so something should be done about it IMHO. I suppose the next step would be to submit patches rather than continue this discussion.

brada4, 2024-01-17:
Vlans are supposedly isolated. No idea what you intend to stich on it.

ffainelli, 2024-01-17:
Let's start over again since what is being offered via the web interface is insufficient and confusing and there is a documentation gap in the wiki.

User expectation: you click on "Enable VLAN filtering", define a default VLAN with all of your ports untagged. Click Save & apply and it works.

Result: it does not. You wait 90 seconds for a configuration roll back because the router stops being reachable from the client. Thankfully there is this is rollback via the web interface, not so much via the CLI AFAICT.

Reason: the user had zero idea or hints being given that they should have been setting a "br-lan.1" network interface to terminate the bridge's default VLAN ID because the default VID is tagged on the CPU port. This is written nowhere and the documentation on the wiki barely mentions it, it seems to assume people will just know. This is unacceptable since OpenWrt is supposed to be largely user friendly and this goes against the principle of least surprise.

Expectation: if migrating from a bridge with VLAN filtering disabled to a bridge with VLAN filtering enabled, just make sure that the CPU port is also VLAN untagged in that VLAN ID. If there are more than one VLAN being defined in the configuration: warn the user that adequate "br-lan.VID" interfaces must be created if the router must be present in that VLAN. The reason is because the CPU port would end up being VLAN tagged in all VLANs, therefore terminating traffic requires a VLAN upper to pop/push the VLAN tag.

Heck, just allow the CPU port/bridge master interface to have its VLAN configuration altered, that way users can realize any configuration they want.

brada4, 2024-01-17:
You have to blindly instrument br-lan.1234 before assigning real ports to it. Wifi or serial helps with mistakes.

ffainelli, 2024-01-17:
Yes they help, but they are not reasonable IMHO, maybe you are right that most people connect over Wi-Fi these days, and only a minority uses Ethernet (like me).

brada4, 2024-01-17:
You can keep wifi in br-lan while you configure br-lan.1234. 
alternative is to make non-trivial consistent changes in /etc/config and restart with fingers crossed

jbsky, 2024-02-28:
Hello,

I am exactly in this situation where I create a bridge with WDS STA <=> AP and I connect VLANx brigdes with access points.

The fact of wanting to create a WDS is in the case where the router is only used as a WIFI repeater (without the possibility of connecting with ethernet).

Either I remove the VLAN filtering on the 2 WDS routers, which is mentioned in this issue, or I find another solution (perhaps a WDS for each VLAN).

Best regards,

brada4, 2024-02-28:
@jbsky completely different issue, you need to rise wifi mtu via hotplug to transfer frame plus vlan tag. You do not need wds, just bridge ends of wifi ap-sta or mesh connection.

jow-, 2024-02-28:
3addr STA interfaces cannot be bridged, WDS (4addr mode) *is* needed.

sbusceti, 2024-03-26:
Hi, I have the same problem: if I  turn on "Enable VLAN filtering" router becomes unreachable.
How do I enable VLAN filtering on the br-lan interface?

Tehvan, 2025-03-22:
I want to add that this . . . lack of documentation at the very least caused me to give up on VLAN implementation for six months. Because how was I supposed to intuit that I need to simultaneously change the lan interface to use br-lan.1 ? It makes sense after I know it must be done, but that isn't really helpful when there is no documentation or warning to the user saying it must be done. So I agree with the author

> From the web interface, how would an user even remotely know that their 'lan' interface physical device must now be 'br-lan.1'? There are no hints being provided at all.
> 
> Also, why not default to VLAN filtering enabled right off the bat?

KanjiMonster, 2025-03-22:
> The end goal for me is that vlan_filtering is set to 1 under /sys/class/net/br-lan/bridge/vlan_filtering and essentially continuing to have this configuration:
>
> ```
> root@router:~# bridge v [...]

While I agree that enabling `vlan_filtering` should not break things, I'm not sure if setting `default_pvid` is the right way. `default_pvid` affects all ports on the bridge, not just the bridge itself. Something like `(bridge_)pvid` for the bridge device might make more sense, which then gets translated to a pvid untagged vlan on the bridge device.