openwrt/openwrt #14199: D-link DIR-620 flash drive is not detected

Vorotila2002, 2023-12-11:
### Describe the bug

I have a D-link DIR-620 A1 (Ralink RT3052 id:1 rev:3) and I needed to connect a flash drive.
However, the flash drive is not detected in the system.
Sda or sda1 does not appear in the /dev directory
The flash drive is not detected in the system.

This is observed in versions 17, 18, 19.
In version 15, flash drive detection occurs normally.

It appears that as of version 17 and later, USB support is not included in the kernel for this type of processor.

### OpenWrt version

15-19

### OpenWrt release

19.07.8

### OpenWrt target/subtarget

ramips/rt305x

### Device

D-Link DIR-620 A1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Vorotila2002, 2023-12-11:
This problem is observed from 17 to 19 versions

DragonBluep, 2023-12-11:
You need to install usb package `kmod-usb2` and filesystem support packages such as `kmod-fs-exfat`.

brada4, 2023-12-11:
Use supported version (last for that matter)
https://firmware-selector.openwrt.org/?version=22.03.6&target=ramips%2Frt305x&id=dlink_dir-620-a1

device page says usb controller is kmod-usb-dwc2 , not ehci.

Vorotila2002, 2023-12-11:
> Use supported version (last for that matter) https://firmware-selector.openwrt.org/?version=22.03.6&target=ramips%2Frt305x&id=dlink_dir-620-a1
> device page says usb controller is kmod-usb-dwc2 , not ehci.

Thanks. 
I'll try version 22, but I'm sure the result will be the same.
In any case, this did not work in versions 17-19.

Vorotila2002, 2023-12-12:
Thanks.
I am forced to use this device because I have limitations on power consumption.
I am powered by a solar panel and every watt is very important to me.

brada4, 2023-12-12:
if you change your mind and want some internet one day ...
https://store-eu.gl-inet.com/en-jp/products/eu-local-delivery-beryl-gl-mt1300

brada4, 2023-12-12:
Check a bit older/newer models too around their site, you know watt less may be worth it.

DragonBluep, 2023-12-12:
If you can build the OpenWrt source code, I can provide you a patch. I've caught the issue. However, v19.07 is EOL now, this device won't get the official fix.

brada4, 2023-12-12:
https://github.com/openwrt/openwrt/blob/05f74354bd43efe9d08c0ce0db3fb0ef37ada94e/target/linux/ramips/dts/rt3050_dlink_dir-620-a1.dts#L126
imho it is just dwc2 kmod needed, still on time to fix if it does not work.

Vorotila2002, 2023-12-12:
> If you can build the OpenWrt source code, I can provide you a patch. I've caught the issue. However, v19.07 is EOL now, this device won't get the official fix.

This is amazing.
I am very happy.

But I wrote here about the error not to solve my problem, but so that the problem would be solved for everyone.

If this bug disappears in future versions, then that is my goal.
I spent more than three days trying to connect the flash drive, but nothing worked, although I did everything correctly.

I would like other people not to have this problem.
In addition, this will help improve the wonderful OpenWRT product

Unfortunately, I did not try to compile the kernel.
Now I don’t have the opportunity to learn.

brada4, 2023-12-12:
Try v21 or v22, USB port driver is dwc2, then you need usb storage driver and a filesystem driver and mount tool
Outline is here https://openwrt.org/docs/guide-user/storage/usb-drives-quickstart, since your flash space is very limited - install reeally minimal set of USB drivers and check if the usb controller is detected ie install dwc2 kmod and check that something appeared in /proc/bus/usb/devices. If not call out the bug.