openwrt/openwrt #14207: mediatek: Linksys E8450/Belking RT3200/D-Link M32: Wireless not working in LuCI after clean install of snapshot image

RolandoMagico, 2023-12-12:
### Describe the bug

After flashing the current snapshot image, wireless is not working in LuCI. The wireless device names are shown as "Generic Unknown":
[image]

Additionally, wireless scanning doesn't work (shows "No Data"):
[image]

### OpenWrt version

r24627-3eca9ba53f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Download image (e.g. use sysupgrade) and deselect "Keep settings and retain the current configuration"

### Actual behaviour

After flashing the current snapshot image, wireless is not working in LuCI. The wireless device names are shown as "Generic Unknown":
[image]

Additionally, wireless scanning doesn't work (shows "No Data"):
[image]

### Expected behaviour

LuCI shows the correct wireless device names and wireless scanning is possible.

### Additional info

The issue was introduced with this commit: https://github.com/openwrt/openwrt/commit/b993a00b82b1bdd682199e44eb81a70dee78e5c9

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-12-12:
OpenWrt is configured to have Wi-Fi disabled by default.

RolandoMagico, 2023-12-13:
> OpenWrt is configured to have Wi-Fi disabled by default.

Yes, that’s ok. I don’t expect to see a Wi-Fi network on my clients after setting up the router.
But I expect rhat the WiFi device name is shown correctly in the LuCI web interface. I also expect that I’m able to scan for WiFi networks in LuCI although wireless is disabled.

It was working on MT7622 before the mentioned commit and it is working on other non-MT7622 devices I have.

bananos, 2024-02-25:
@RolandoMagico Have you managed to resolve the issue?  I'm experiencing very similar issue with 5Ghz wifi being disabled after fresh install of `23.05.2 r23630-842932a63d` on my Linksys E8450(UBI). 

So far, I've found out that it fails to properly load drivers

```
dmesg | grep "mt79"
[    7.223391] mt7915e 0000:01:00.0: assign IRQ: got 146
[    7.228510] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    7.234638] mt7915e 0000:01:00.0: enabling bus mastering
[    7.467569] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220929104113a
[    7.607541] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[    7.660930] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205
[    7.778511] mt7915e 0000:01:00.0: eeprom load fail, use default bin
[    7.784976] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_eeprom.bin failed with error -2
[    7.794665] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_eeprom.bin
[    7.810938] mt7915e: probe of 0000:01:00.0 failed with error -12

```
And `mediatek/mt7915_eeprom.bin` is missing in `/lib/firmware/mediatek/`

```
ls -la /lib/firmware/mediatek/
drwxr-xr-x    1 root     root           384 Feb 26 00:08 .
drwxr-xr-x    1 root     root           232 Aug 26  2023 ..
-rw-r--r--    1 root     root        300376 Nov 14 15:38 mt7622_n9.bin
-rw-r--r--    1 root     root         82110 Nov 14 15:38 mt7622_rom_patch.bin
-rw-r--r--    1 root     root        144544 Aug 26  2023 mt7915_rom_patch.bin
-rw-r--r--    1 root     root        116160 Aug 26  2023 mt7915_wa.bin
-rw-r--r--    1 root     root       1260960 Aug 26  2023 mt7915_wm.bin

```
Reinstall of driver + firmware opkg packages didn't help

bananos, 2024-02-26:
For anyone looking for the solution -- I've just installed the most recent OpenWRT snapshot and 5Ghz wifi was up and running:

```
cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25325-c2537f3c8f'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25325-c2537f3c8f'
DISTRIB_TAINTS=''

```

rofra, 2024-11-04:
Dear @RolandoMagico,

You can use the following file for a default configuration with unactivated OpenWRT with no password:
```txt
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wmac'
	option channel 'auto'
	option band '2g'
	option htmode 'HT40'
	option txpower '20'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWRT'
	option encryption 'none'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option cell_density '0'
	option country 'US'
	option htmode 'HE40'
	option band '5g'
	option channel 'auto'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWRT'
	option encryption 'none'
	option disabled '1'
``

RolandoMagico, 2024-11-05:
And this configuration file fixes the reported issue? If this is the case, why isn’t this the default configuration after a clean install?

rofra, 2024-11-05:
@RolandoMagico : No, the issue is till relevant. I just posted you an example configuration file for you and the openwrt community

wfgriffin88, 2025-07-20:
> @RolandoMagico   @ Hi [@bananos](https://github.com/bananos) , for me, the issue is still not resolved. I tried the latest snapshot (OpenWrt SNAPSHOT r25331-0c18c5679a / LuCI Master git-24.052.52717-06c0fbb), the wireless overview still shows "Generic unknown" after initial boot.

I have also encountered the "Generic unknown" device issue after freshly installing OpenWRT 24.10.2 (UBI) on my Linksys E8450 router.    The root cause at least for my case appears to be related to the file  **/etc/board.json**  .

Please see my post [here](https://github.com/dangowrt/owrt-ubi-installer/issues/137#issuecomment-3094675754) for a possible solution devised with help from @kratanat .

RolandoMagico, 2025-09-13:
Hi @maurerle,
I tried on a R32 with OpenWrt 24.10.2. After flashing **without** keeping the settings, I still see the "Generic Unknown" entries in the Wifi section -> Problem not yet solved from my point of view.
Then I entered the commands you provided and the wireless section in LuCI looks good. Why are these commands required? Never needed them on other devices.
I also tried to flash the latest Snapshot. Unfortunately the device doesn't boot at all. Neither with sysupgrade nor with recovery image. But I don't have the possibility to investigate further at the moment.

maurerle, 2025-09-14:
Thanks for checking this again!
Sad that it does not work on your device.. I might check a WAX206 in the future as well.

Setting the device up with uci triggers the renaming of the phy0/phy1 to wl0/wl1, which is why it works afterwards.
I thought that there are already some fallbacks implemented for this, but then we still have to wait for #17821 or a successor.
Related IRC discussion from april: https://oftc.catirclogs.org/openwrt-devel/2025-04-04#

rotanid, 2025-10-22:
on the main branch, there's fix now:
https://github.com/openwrt/openwrt/commit/4b7323e3bfa82ed8279f1e757f82d4961d8e3ae7
but it seems it's not backportable to 24.10

rotanid, 2025-12-27:
might be related, not sure: https://github.com/openwrt/openwrt/commit/e4d5e76af059d9c7fe295bbd144a39a77fbb5491