openwrt/openwrt #14214: Syntax error from dnsmasq when nftset directive is triggered

cbz, 2023-12-13:
### Describe the bug

When using dnsmasq-full as DNS server in the latest release and - in this case - specifying an nftset directive, the following error is seen when the directive is triggered via a lookup:

Dec 13 17:43:53 OpenWRT dnsmasq[1]: nftset ip bulk Error: syntax error, unexpected '{', expecting string or last

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys EA6350v3

### Image kind

Self-built image

### Steps to reproduce

Specify an ipset in LuCI and verify that it appears in /etc/config/firewall as the following:

`
config ipset
        option name 'bulk'
        option family 'ipv4'
        list match 'dest_ip'
        option timeout '600'
`

Include the following line in /etc/dnsmasq.conf:

`nftset=/windowsupdate.com/update.microsoft.com/ip#bulk`

Looking up update.microsoft.com via nslookup should result in the IP address being added to the `bulk` nftset

### Actual behaviour

Can verify that this file is included in the generated /var/etc/dnsmasq.conf.**** file.  When looking up update.microsoft.com via:

`nslookup update.microsoft.com 127.0.0.1`

I see the error log above, and I can look in the nftset and verify that no address was added:

`root@gyre:~# nft list set inet fw4 bulk
table inet fw4 {
        set bulk {
                type ipv4_addr
                timeout 10m
        }
}
`

### Expected behaviour

Looking up update.microsoft.com via nslookup should result in the IP address being added to the `bulk` nftset

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-12-13:
`family` should be `inet`, not `ipv4`

dave14305, 2023-12-13:
If you’ve added the nftset directly to dnsmasq.conf (which is wrong), it should be:
```
nftset=/windowsupdate.com/update.microsoft.com/4#inet#fw4#bulk
```

cbz, 2023-12-13:
> If you’ve added the nftset directly to dnsmasq.conf (which is wrong), it should be:
> 
> ```
> nftset=/windowsupdate.com/update.microsoft.com/4#inet#fw4#bulk
> ```

It's supported via the generated file in /var/etc/dnsmasq.conf always containing the line:
`conf-file=/etc/dnsmasq.conf`

In principle it isn't different from configuring collectd via alt_config_file

In any case the suggested alternate throws the same error, I suspect this is down to an intermediate command being constructed incorrectly:

`Dec 13 18:34:46 OpenWRT dnsmasq[1]: nftset inet bulk Error: syntax error, unexpected '{', expecting string or last`

The syntax I used originally was from the source below, which I presume should be correct as it's a mirror of the dnsmasq repo:

https://github.com/imp/dnsmasq/blob/master/dnsmasq.conf.example#L88C65-L88C65

[I have a separate issue with ipset in /etc/config/dhcp not generating the correct syntax and thought it best to split things this way]

dave14305, 2023-12-13:
`ip#bulk` is wrong because your bulk set is in a table of type inet and the table is fw4. `inet#fw4#bulk` would also work.

cbz, 2023-12-13:
> `ip#bulk` is wrong because your bulk set is in a table of type inet and the table is fw4. `inet#fw4#bulk` would also work.

Okay, that works now. I'll close the ticket.

dave14305, 2023-12-14:
> I'll close the ticket.

Still open. :-)

closed 2023-12-14 by cbz