openwrt/openwrt #14215: TP-Link RE 305 v1 resets settings to factory after reboot

fshahin90, 2023-12-13:
### Describe the bug

When flashing the latest version of OpenWrt to TP-Link RE 305 v1 then making modifications (e.g. Change Root Password or Change Device IP) then restarting the device (either by System->Restart or unplug and re-plug the devise) all the modifications will be reverted to factory settings (e.g Root will not have a password, IP returned to 192.168.1.1)

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link RE305 v1

### Image kind

Official downloaded image

### Steps to reproduce

1- flash latest version of OpenWrt to TP-Link RE305 v1
2- Make changes to the device Change root password and change the device IP address
3- Reboot the device (either using System -> Reboot or Unplug and Replug the device)
4- try to access the device using the changed IP address

### Actual behaviour

After flashing the latest version of openwrt and making changes to the device (e.g setting root password, changing the device ip address) then rebooting the device, after the reboot when trying to access the device using the ip I set I get the page can't be reached message but when I try to access using the default ip `192.168.1.1` i get the login page..

### Expected behaviour

After flashing the latest version of openwrt and making changes to the device (e.g setting root password, changing the device ip address) then rebooting the device, when i access the device using the ip address I set i see the LUCI login page with the root have the password i set

### Additional info

The max working version that doent have this bug is `21.02.7` after this version I got the solution from the solved post in this forum thread
https://forum.openwrt.org/t/tp-link-re305v1-save-doesnt-work-reboot-and-theyre-gone/157897

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-12-13:
Please check the latest version 23.05.2.

fshahin90, 2023-12-13:
> Please check the latest version 23.05.2.

I did the still the same thing.

brada4, 2023-12-13:
If you make image with imagebuilder you get:
file-system partition too big (more than 3670016 bytes): Success

You can deselect packages like luci and uhttpd  in imagebuilder to see if image is reduced in size, and no silenced warnings in build output.

Further compaction means re-building kernel and leaving luci out, not much fun setting wifi from uci cli (guide is there in wiki, it works)

brada4, 2023-12-14:
what happens is that firmware grew too big and there is not enough space for jffs2 overlay anymore.

brada4, 2023-12-14:
https://openwrt.org/supported_devices/432_warning
tldr - check /proc/mtd and logread after first boot. If you do not have 3 free eraseblocks , usually 5 eraseblocks of 64k each total you can not save anything more.

fshahin90, 2023-12-15:
what are the minimum packages (including Luci) required to make the device as dumb wireless extender?

tmn505, 2023-12-26:
Recently I stumbled on same problem when porting RE365 V1. To solve it I used mtd-concat driver so I could use unpartitioned 1.6 MiB space unused by vendor firmware ([ramips: mt76x8: add support for TP-Link RE365 v1](https://patchwork.ozlabs.org/project/openwrt/patch/20231222104421.1823-1-tmn505@terefe.re)). If You have access to a serial console on the device (I can't guarantee 100% that this will work), You can try out this patch for 23.05.2 Image Builder:
```diff
--- a/include/image.mk
+++ b/include/image.mk
@@ -182,7 +182,7 @@ endef
 # $(3) extra CPP flags
 # $(4) extra DTC flags
 define Image/BuildDTB/sub
-	$(TARGET_CROSS)cpp -nostdinc -x assembler-with-cpp \
+	$(if IB, cpp, $(TARGET_CROSS)cpp) -nostdinc -x assembler-with-cpp \
 		$(DTS_CPPFLAGS) \
 		-I$(DTS_DIR) \
 		-I$(DTS_DIR)/include \
@@ -568,7 +568,6 @@ define Device/Build/compile
 
 endef
 
-ifndef IB
 define Device/Build/dtb
   ifndef BUILD_DTS_$(1)
   BUILD_DTS_$(1) := 1
@@ -590,7 +589,6 @@ define Device/Build/dtbo
   endif
 
 endef
-endif
 
 define Device/Build/kernel
   $$(eval $$(foreach dts,$$(DEVICE_DTS), \
@@ -611,15 +609,10 @@ define Device/Build/kernel
   $(call Device/Export,$$(KDIR_KERNEL_IMAGE),$(1))
   $(BIN_DIR)/$$(KERNEL_IMAGE): $$(KDIR_KERNEL_IMAGE)
 	cp $$^ $$@
-  ifndef IB
-    ifdef CONFIG_IB
-      install: $$(KDIR_KERNEL_IMAGE)
-    endif
     $$(KDIR_KERNEL_IMAGE): $(KDIR)/$$(KERNEL_NAME) $(CURDIR)/Makefile $$(KERNEL_DEPENDS) image_prepare
 	@rm -f $$@
 	$$(call concat_cmd,$$(KERNEL))
 	$$(if $$(KERNEL_SIZE),$$(call Build/check-size,$$(KERNEL_SIZE)))
-  endif
 endef
 
 define Device/Build/image
--- a/target/linux/ramips/dts/mt7628an_tplink_re305-v1.dts
+++ b/target/linux/ramips/dts/mt7628an_tplink_re305-v1.dts
@@ -5,6 +5,23 @@
 / {
 	compatible = "tplink,re305-v1", "mediatek,mt7628an-soc";
 	model = "TP-Link RE305 v1";
+
+	virtual_flash {
+		compatible = "mtd-concat";
+		devices = <&fwconcat0 &fwconcat1>;
+
+		partitions {
+			compatible = "fixed-partitions";
+			#address-cells = <1>;
+			#size-cells = <1>;
+
+			partition@0 {
+				compatible = "tplink,firmware";
+				label = "firmware";
+				reg = <0x0 0x0>;
+			};
+		};
+	};
 };
 
 &spi0 {
@@ -26,8 +43,7 @@
 				read-only;
 			};
 
-			partition@20000 {
-				compatible = "tplink,firmware";
+			fwconcat0: partition@20000 {
 				label = "firmware";
 				reg = <0x20000 0x5e0000>;
 			};
@@ -38,10 +54,10 @@
 				read-only;
 			};
 
-			/*
-				The flash space between 0x650000 and 0x7f0000 is blank in the
-				stock firmware so it is left out as well.
-			*/
+			fwconcat1: partition@650000 {
+				label = "fwconcat1";
+				reg = <0x650000 0x1a0000>;
+			};
 
 			radio: partition@7f0000 {
 				label = "radio";
--- a/target/linux/ramips/image/mt76x8.mk
+++ b/target/linux/ramips/image/mt76x8.mk
@@ -640,6 +640,8 @@ define Device/tplink_re305-v1
   DEVICE_VARIANT := v1
   DEVICE_PACKAGES := kmod-mt76x2
   TPLINK_BOARD_ID := RE305-V1
+  IMAGE/sysupgrade.bin := append-kernel | append-rootfs | pad-rootfs | check-size | append-metadata
+  IMAGE_SIZE := 7680k
 endef
 TARGET_DEVICES += tplink_re305-v1
 
```

You'll need to put `include` directory and its contents from 5.15.137 kernel source (https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/linux-5.15.137.tar.xz) inside `build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-5.15.137` directory of Image Builder. Then run `make image PROFILE="tplink_re305-v1" PACKAGES="luci"`. Use created sysupgrade image. After upgrade You should have enough space for overlay. For convenience I'm attaching mentioned patch  [ib-re305v1.txt](https://github.com/openwrt/openwrt/files/13773183/ib-re305v1.txt) (extension renamed because of stupid GitHub interface).

brada4, 2023-12-26:
you could use just that partition for rootfs-data. There is no obligation that data follows squash

tmn505, 2023-12-27:
Yes, there is no obligation, but that's temporary remedy which would be difficult to migrate from when upgrading. There would be a time, when kernel+squashfs would grow beyond what's assigned to them, while space which could be used is statically assigned to rootfs_data and the code responsible for upgrade, residing on the device, is not accommodated to that situation. This would need some intermediary process to fix it, complicating it for users.

svanheule, 2024-01-23:
Snapshot builds are now `small_flash` and 10% smaller (~500kB) compared to 23.05 images. Hopefully that frees up enough flash to resolve this issue. Thanks, @tmn505!

closed 2024-01-23 by svanheule