openwrt/openwrt #14219: netifd: segfault

schm0, 2023-12-14:
### Describe the bug

netifd segfaults while bringing up bridge ports
[image]

### OpenWrt version

r24660+19-c22aa0be3e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Custom x86/64 box

### Image kind

Self-built image

### Steps to reproduce

Build latest image, flash, see netifd segfaulting

### Actual behaviour

netifd segfaults while bringing up bridge ports

### Expected behaviour

netifd doesn't segfaults while bringing up bridge ports

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2023-12-14:
Please post your /etc/config/network, so that I can try to reproduce this

schm0, 2023-12-14:
```

config globals 'globals'
	option ula_prefix 'fdx:x:x::/48'

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device
	option name 'br-lan'
	option type 'bridge'
	option ageing_time '14410'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr 'x.x.x.x'
	option netmask '255.255.255.0'
	option ip6assign '64'
	option ip6ifaceid '::x:x:x:x'

config interface 'wan'
	option device 'eth1'
	option proto 'dhcp'
	option peerdns '0'
	option hostname '*'
	option norelease '1'
	option delegate '0'

config interface 'wan6'
	option device 'eth1'
	option proto 'dhcpv6'
	option iface_dslite '0'
	option peerdns '0'
	option reqaddress 'force'
	option forceprefix '1'
	option reqprefix '56'
	option norelease '1'
	option sourcefilter '0'

config interface 'modem'
	option device 'eth1'
	option proto 'static'
	option ipaddr 'x.x.x.x'
	option delegate '0'
	option defaultroute '0'
	option netmask '255.255.255.254'

config interface 'wgc0'
	option proto 'wireguard'
	option private_key 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
	list addresses 'x.x.x.x/x'
	list addresses 'xxx:xxxx:xxxx:xxxx::x:xxxx/128'
	list dns 'x.x.x.x'
	option nohostroute '1'
	option mtu '1280'
	option auto '0'

config wireguard_wgc0
	option public_key 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
	option endpoint_host 'xx.xx.xx.xx'
	option endpoint_port 'x'
	option route_allowed_ips '0'
	list allowed_ips '0.0.0.0/0'
	list allowed_ips '::/0'

config route
	option interface 'wgc0'
	option target 'x.x.x.x/32'

config route
	option interface 'wgc0'
	option target 'x.x.x.x/32'

config route
	option interface 'wgc0'
	option target 'x.x.xx/16'

config route6
	option interface 'wgc0'
	option target '::/0'
	option metric '256'
	option table 'pbr_wgc0'

```

mikma, 2023-12-15:
I can confirm an issue with netifd 2023-12-05-cc9e928f. I run a custom build with glibc on x86_64, and I can trigger the bug with a config containing only the loopback interface:

```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
```

Valgrind reports an invalid read, and a segfault. In GDB I get some odd log output before it segfaults. There seems to be a problem related to processing the variable arguments.

```
==5753== Memcheck, a memory error detector
==5753== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==5753== Using Valgrind-3.22.0 and LibVEX; rerun with -h for copyright info
==5753== Command: /root/netifd
==5753== 
killall: meshd-nl80211: no process killed
./mac80211.sh: eval: line 417: drv_mac80211_init_vlan_config: not found
./mac80211.sh: eval: line 422: drv_mac80211_init_station_config: not found
==5753== Invalid read of size 1
==5753==    at 0x48472C2: strlen (vg_replace_strmem.c:502)
==5753==    by 0x4920480: __printf_buffer (vfprintf-process-arg.c:435)
==5753==    by 0x493F59E: __vsnprintf_internal (vsnprintf.c:96)
==5753==    by 0x49BD7DA: __vsyslog_internal (syslog.c:193)
==5753==    by 0x406A7B: netifd_log_message (main.c:130)
==5753==    by 0x40E567: UnknownInlinedFun (interface.c:399)
==5753==    by 0x40E567: interface_main_dev_cb (interface.c:444)
==5753==    by 0x40FA76: device_broadcast_cb (device.c:608)
==5753==    by 0x4857AAE: safe_list_for_each (safe_list.c:74)
==5753==    by 0x411766: device_broadcast_event (device.c:627)
==5753==    by 0x413722: device_claim (device.c:704)
==5753==    by 0x40CD04: UnknownInlinedFun (interface.c:1137)
==5753==    by 0x40CD04: interface_set_up (interface.c:1123)
==5753==    by 0x40FA0B: UnknownInlinedFun (interface.c:1183)
==5753==    by 0x40FA0B: config_init_all (config.c:822)
==5753==  Address 0xf is not stack'd, malloc'd or (recently) free'd
==5753== 
==5753== 
==5753== Process terminating with default action of signal 11 (SIGSEGV)
==5753==  Access not within mapped region at address 0xF
==5753==    at 0x48472C2: strlen (vg_replace_strmem.c:502)
==5753==    by 0x4920480: __printf_buffer (vfprintf-process-arg.c:435)
==5753==    by 0x493F59E: __vsnprintf_internal (vsnprintf.c:96)
==5753==    by 0x49BD7DA: __vsyslog_internal (syslog.c:193)
==5753==    by 0x406A7B: netifd_log_message (main.c:130)
==5753==    by 0x40E567: UnknownInlinedFun (interface.c:399)
==5753==    by 0x40E567: interface_main_dev_cb (interface.c:444)
==5753==    by 0x40FA76: device_broadcast_cb (device.c:608)
==5753==    by 0x4857AAE: safe_list_for_each (safe_list.c:74)
==5753==    by 0x411766: device_broadcast_event (device.c:627)
==5753==    by 0x413722: device_claim (device.c:704)
==5753==    by 0x40CD04: UnknownInlinedFun (interface.c:1137)
==5753==    by 0x40CD04: interface_set_up (interface.c:1123)
==5753==    by 0x40FA0B: UnknownInlinedFun (interface.c:1183)
==5753==    by 0x40FA0B: config_init_all (config.c:822)
==5753==  If you believe this happened as a result of a stack
==5753==  overflow in your program's main thread (unlikely but
==5753==  possible), you can try to increase the size of the
==5753==  main thread stack using the --main-stacksize= flag.
==5753==  The main thread stack size used in this run was 8388608.
==5753== 
==5753== HEAP SUMMARY:
==5753==     in use at exit: 93,960 bytes in 122 blocks
==5753==   total heap usage: 3,304 allocs, 3,182 frees, 869,124 bytes allocated
==5753== 
==5753== LEAK SUMMARY:
==5753==    definitely lost: 0 bytes in 0 blocks
==5753==    indirectly lost: 0 bytes in 0 blocks
==5753==      possibly lost: 5,336 bytes in 28 blocks
==5753==    still reachable: 88,624 bytes in 94 blocks
==5753==         suppressed: 0 bytes in 0 blocks
==5753== Rerun with --leak-check=full to see details of leaked memory
==5753== 
==5753== For lists of detected and suppressed errors, rerun with: -s
==5753== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

```
Reading symbols from /root/netifd...
Starting program: /root/netifd -S
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
[Detaching after vfork from child process 5821]
[Detaching after vfork from child process 5824]
[Detaching after vfork from child process 5827]
[Detaching after vfork from child process 5830]
[Detaching after vfork from child process 5833]
[Detaching after vfork from child process 5836]
[Detaching after vfork from child process 5841]
killall: meshd-nl80211: no process killed
./mac80211.sh: eval: line 417: drv_mac80211_init_vlan_config: not found
./mac80211.sh: eval: line 422: drv_mac80211_init_station_config: not found
Interface 'lo' is s��A
Interface '(null)' is setting up now
[Detaching after fork from child process 5844]
Interface '���A' is now up

Program received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:76
76	../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
```

mikma, 2023-12-15:
I have added a PR (https://github.com/openwrt/netifd/pull/19) to [netifd](https://github.com/openwrt/netifd) which fixes the problem I reported above.

schm0, 2023-12-16:
> I have added a PR ([openwrt/netifd#19](https://github.com/openwrt/netifd/pull/19)) to [netifd](https://github.com/openwrt/netifd) which fixes the problem I reported above.
Seems fixed with this, thank you.

nbd168, 2023-12-18:
Fix pushed to master, thanks

closed 2023-12-18 by nbd168