openwrt/openwrt #14223: Remote syslog is sent with wrong timezone

a-bali, 2023-12-14:
### Describe the bug

I have set the proper local timezone in OpenWrt and have configured logging to a remote syslog host. The messages are sent with UTC-formatted timestamp, but with local timezone i.e. the time and date are not converted to UTC.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

Set local timezone. Configure remote syslog. Check timestamp of incoming messages on remote syslog host.

### Actual behaviour

I have set the proper local timezone in OpenWrt and have configured logging to a remote syslog host. The messages are sent with UTC-formatted timestamp, but with local timezone i.e. the time and date are not converted to UTC.

### Expected behaviour

Timestamp of remote syslog messages should be transformed into UTC.

### Additional info

```
# cat /etc/config/system
config system
	option hostname 'openwrt1'
	option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
	option ttylogin '0'
	option log_size '64'
	option urandom_seed '0'
	option zonename 'Europe/Budapest'
	option log_proto 'tcp'
	option conloglevel '8'
	option cronloglevel '5'
	option log_ip 'sentinel.lan'
	option log_port '6000'
# cat /etc/TZ
CET-1CEST,M3.5.0,M10.5.0/3
# ls -l /etc/localtime
lrwxrwxrwx    1 root     root            14 Oct  9 23:45 /etc/localtime -> /tmp/localtime
# ls -l /tmp/localtime
ls: /tmp/localtime: No such file or directory
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-14:
You mean in central syslog TZ offset is doubled?
Can you get pcap (or wireshark rendering picture thereoff) from
```
logger `date ; echo -n @@@\ ; date -u`
```
Obvious workaround is to set timezone back to UTC

a-bali, 2023-12-14:
Sure:

```
5:30:19.228258 IP (tos 0x0, ttl 64, id 56019, offset 0, flags [DF], proto TCP (6), length 149)
    openwrt1.lan.36340 > sentinel.lan.x11: Flags [P.], cksum 0x6a21 (correct), seq 1235389673:1235389770, ack 1062687835, win 4015, options [nop,nop,TS val 2680583478 ecr 724951368], length 97
	0x0000:  4500 0095 dad3 4000 4006 dc01 c0a8 010b  E.....@.@.......
	0x0010:  c0a8 0132 8df4 1770 49a2 8ce9 3f57 545b  ...2...pI...?WT[
	0x0020:  8018 0faf 6a21 0000 0101 080a 9fc6 7536  ....j!........u6
	0x0030:  2b35 e148 3c31 333e 4465 6320 3134 2031  +5.H<13>Dec.14.1
	0x0040:  363a 3330 3a31 3920 6f70 656e 7772 7431  6:30:19.openwrt1
	0x0050:  2072 6f6f 743a 2054 6875 2044 6563 2031  .root:.Thu.Dec.1
	0x0060:  3420 3136 3a33 303a 3139 2043 4554 2032  4.16:30:19.CET.2
	0x0070:  3032 3320 4040 4020 5468 7520 4465 6320  023.@@@.Thu.Dec.
	0x0080:  3134 2031 353a 3330 3a31 3920 5554 4320  14.15:30:19.UTC.
	0x0090:  3230 3233 0a                             2023.
```

brada4, 2023-12-14:
Is UDP same?

a-bali, 2023-12-14:
```
15:48:15.876175 IP (tos 0x0, ttl 64, id 58251, offset 0, flags [DF], proto UDP (17), length 124)
    openwrt1.lan.51061 > sentinel.lan.6000: [udp sum ok] UDP, length 96
	0x0000:  4500 007c e38b 4000 4011 d357 c0a8 010b  E..|..@.@..W....
	0x0010:  c0a8 0132 c775 1770 0068 5cc5 3c31 333e  ...2.u.p.h\.<13>
	0x0020:  4465 6320 3134 2031 363a 3438 3a31 3520  Dec.14.16:48:15.
	0x0030:  6f70 656e 7772 7431 2072 6f6f 743a 2054  openwrt1.root:.T
	0x0040:  6875 2044 6563 2031 3420 3136 3a34 383a  hu.Dec.14.16:48:
	0x0050:  3135 2043 4554 2032 3032 3320 4040 4020  15.CET.2023.@@@.
	0x0060:  5468 7520 4465 6320 3134 2031 353a 3438  Thu.Dec.14.15:48
	0x0070:  3a31 3520 5554 4320 3230 3233            :15.UTC.2023
```

brada4, 2023-12-14:
@nbd168 can help with fixing busybox sysklog.