openwrt/openwrt #14225: [General Protection Fault] - Getting glibc error "cannot get entropy for arc4random" that forces reboot after 10s

danysc23, 2023-12-14:
### Describe the bug

 I am getting a glibc error "cannot get entropy for arc4random" in runtime that is forcing a reboot every 10s. 

### OpenWrt version

r23630-842932a63d

### OpenWrt release

r23630-842932a63d

### OpenWrt target/subtarget

x86/apu2

### Device

PC Engines APU2

### Image kind

Official downloaded image

### Actual behaviour

```
[    8.327635] mount_root: loading kmods from internal overlay
[    8.398021] kmodloader: loading kernel modules from //etc/modules-boot.d/*
[    8.409498] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
[    8.468357] block: attempting to load /etc/config/fstab
[    8.473980] block: unable to load configuration (fstab: Entry not found)
[    8.480993] block: no usable configuration
[    8.486063] loop0: detected capacity change from 0 to 212992
[    8.549812] loop0: detected capacity change from 212992 to 158720
[    8.576960] EXT4-fs (loop0): recovery complete
[    8.582652] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null). Quota mode: disabled.
[    8.609181] block: attempting to load /etc/config/fstab
[    8.614922] block: unable to load configuration (fstab: Entry not found)
[    8.622177] block: no usable configuration
[    8.627929] mount_root: overlay filesystem has not been fully initialized yet
[    8.638566] mount_root: switching to ext4 overlay
[    8.650173] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
[    8.818025] EXT4-fs (sda1): warning: mounting unchecked fs, running e2fsck is recommended
[    8.829211] EXT4-fs (sda1): mounted filesystem without journal. Opts: (null). Quota mode: disabled.
[    8.848913] urandom-seed: Seed file not found (/etc/urandom.seed)
[    9.174986] procd: - early -
Fatal glibc erro[    9.320837] traps: procd[1] general protection fault ip:7f25702a13c4 sp:7ffe17ebe2e0 error:0 in libc.so.6[7f25702a1000+141000]
r: cannot get entropy for arc4random
[   11.510033] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[   11.550100] reboot: Restarting system
```

### Additional info

 I did a little research and I noticed that the arc4random that glibc is using is different from the libbsd library, so I tried to use that lib ( with CONFIG_PACKAGE_libbsd=y) and this error didn't occur, but I'm still facing forced reboots after 10 min.
In previous version ( 22.03) this was not necessary! Am I missing some config? 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danysc23, 2023-12-14:
> Invalid Release reported. `r23630-842932a63d` Is this from a clean repository?

yes

brada4, 2023-12-14:
Attach 2 diffconfigs.

danysc23, 2023-12-14:
> Attach 2 diffconfigs.

```
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

brada4, 2023-12-14:
You need to run script and attach its produce - one time with problem exhibiting, another time with problem solved.

danysc23, 2023-12-14:
> You need to run script and attach its produce - one time with problem exhibiting, another time with problem solved.

Working: 

  ```
  CONFIG_TARGET_x86=y
    CONFIG_TARGET_x86_apu2=y
    CONFIG_TARGET_x86_apu2_DEVICE_apu2=y
    CONFIG_DEVEL=y
    CONFIG_TOOLCHAINOPTS=y
    CONFIG_PACKAGE_libbz2=y
    CONFIG_PACKAGE_libdb47=y
    CONFIG_PACKAGE_libexpat=y
    CONFIG_PACKAGE_libffi=y
    CONFIG_PACKAGE_libgdbm=y
    CONFIG_PACKAGE_libgmp=m
    CONFIG_PACKAGE_libipset=y
    CONFIG_PACKAGE_liblua=y
    CONFIG_PACKAGE_liblzo=y
    CONFIG_PACKAGE_libmnl=y
    CONFIG_PACKAGE_libncurses=y
    CONFIG_PACKAGE_libnetfilter-conntrack=m
    CONFIG_PACKAGE_libnetfilter-queue=y
    CONFIG_PACKAGE_libnettle=m
    CONFIG_PACKAGE_libnfnetlink=y
    CONFIG_PACKAGE_libopenssl=y
    CONFIG_PACKAGE_libpcap=y
    CONFIG_PACKAGE_libpopt=y
    CONFIG_PACKAGE_libsqlite3=y
    CONFIG_PACKAGE_libssp=y
    CONFIG_PACKAGE_libstdcpp=y
    CONFIG_PACKAGE_libubus-lua=y
    CONFIG_PACKAGE_libuci-lua=y
    CONFIG_PACKAGE_libbsd=y
    CONFIG_PACKAGE_libustream-openssl=y
    CONFIG_PACKAGE_libxml2=y
```

    
   Not working 
   
       ```
    CONFIG_TARGET_x86=y
    CONFIG_TARGET_x86_apu2=y
    CONFIG_TARGET_x86_apu2_DEVICE_apu2=y
    CONFIG_DEVEL=y
    CONFIG_TOOLCHAINOPTS=y
    CONFIG_PACKAGE_libbz2=y
    CONFIG_PACKAGE_libdb47=y
    CONFIG_PACKAGE_libexpat=y
    CONFIG_PACKAGE_libffi=y
    CONFIG_PACKAGE_libgdbm=y
    CONFIG_PACKAGE_libgmp=m
    CONFIG_PACKAGE_libipset=y
    CONFIG_PACKAGE_liblua=y
    CONFIG_PACKAGE_liblzo=y
    CONFIG_PACKAGE_libmnl=y
    CONFIG_PACKAGE_libncurses=y
    CONFIG_PACKAGE_libnetfilter-conntrack=m
    CONFIG_PACKAGE_libnetfilter-queue=y
    CONFIG_PACKAGE_libnettle=m
    CONFIG_PACKAGE_libnfnetlink=y
    CONFIG_PACKAGE_libopenssl=y
    CONFIG_PACKAGE_libpcap=y
    CONFIG_PACKAGE_libpopt=y
    CONFIG_PACKAGE_libsqlite3=y
    CONFIG_PACKAGE_libssp=y
    CONFIG_PACKAGE_libstdcpp=y
    CONFIG_PACKAGE_libubus-lua=y
    CONFIG_PACKAGE_libuci-lua=y
    CONFIG_PACKAGE_libustream-openssl=y
    CONFIG_PACKAGE_libxml2=y
```

brada4, 2024-01-10:
cf https://bugs.gentoo.org/920476

Somehow your kernel is built without getrandom/getentropy system calls, but other library compensates them. 
It is the thing triggering unseeded rng read on other systems at boot.