openwrt/openwrt #14226: tools/elfutils build fails on  libbz2.so.1.0: cannot open shared object file

paulwouters, 2023-12-15:
### Describe the bug

```
make[3]: Entering directory '/home/build/openwrt.git/tools/elfutils'
. /home/build/openwrt.git/include/shell.sh; bzcat /home/build/openwrt.git/dl/elfutils-0.189.tar.bz2 | tar -C /home/build/openwrt.git/build_dir/host/elfutils-0.189/.. -xf -
bzcat: error while loading shared libraries: libbz2.so.1.0: cannot open shared object file: No such file or directory
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors

```

The weird thing is, issuing this manually works:
```
build@lake:~/openwrt.git$ cd /home/build/openwrt.git/tools/elfutils
build@lake:~/openwrt.git/tools/elfutils$ . /home/build/openwrt.git/include/shell.sh; bzcat /home/build/openwrt.git/dl/elfutils-0.189.tar.bz2 | tar -C /home/build/openwrt.git/build_dir/host/elfutils-0.189/.. -xf -
build@lake:~/openwrt.git/tools/elfutils$ 
```

Is it trying to use the wrong bzcat? eg from staging instead of the system one?

### OpenWrt version

head is at c22aa0be3e2df4d46010fa57a1004c17c1288781

### OpenWrt release

head is at c22aa0be3e2df4d46010fa57a1004c17c1288781

### OpenWrt target/subtarget

mvebu_cortexa9_DEVICE_cznic_turris-omnia

### Device

build system

### Image kind

Self-built image

### Steps to reproduce

grab openwrt git, and use .config from https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-imagebuilder-mvebu-cortexa9.Linux-x86_64.tar.xz

### Actual behaviour

```
make[3]: Entering directory '/home/build/openwrt.git/tools/elfutils'
. /home/build/openwrt.git/include/shell.sh; bzcat /home/build/openwrt.git/dl/elfutils-0.189.tar.bz2 | tar -C /home/build/openwrt.git/build_dir/host/elfutils-0.189/.. -xf -
bzcat: error while loading shared libraries: libbz2.so.1.0: cannot open shared object file: No such file or directory
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors

```

### Expected behaviour

elfutils that builds properly

### Diffconfig

```text
build@lake:~/openwrt.git$  ./scripts/diffconfig.sh
CONFIG_TARGET_mvebu=y
CONFIG_TARGET_mvebu_cortexa9=y
CONFIG_TARGET_mvebu_cortexa9_DEVICE_cznic_turris-omnia=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_FEED_luci=y
CONFIG_FEED_packages=y
CONFIG_FEED_routing=y
CONFIG_FEED_telephony=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_KERNEL_WERROR=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_ath10k-board-qca988x=m
CONFIG_PACKAGE_ath10k-firmware-qca988x-ct=m
CONFIG_PACKAGE_bzip2=m
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_hostapd-common=m
CONFIG_PACKAGE_ip-full=m
CONFIG_PACKAGE_iw=m
CONFIG_PACKAGE_iwinfo=m
CONFIG_PACKAGE_kmod-ath=m
CONFIG_PACKAGE_kmod-ath10k-ct=m
CONFIG_PACKAGE_kmod-ath9k=m
CONFIG_PACKAGE_kmod-ath9k-common=m
CONFIG_PACKAGE_kmod-cfg80211=m
CONFIG_PACKAGE_kmod-crypto-aead=m
CONFIG_PACKAGE_kmod-crypto-ccm=m
CONFIG_PACKAGE_kmod-crypto-cmac=m
CONFIG_PACKAGE_kmod-crypto-ctr=m
CONFIG_PACKAGE_kmod-crypto-gcm=m
CONFIG_PACKAGE_kmod-crypto-gf128=m
CONFIG_PACKAGE_kmod-crypto-ghash=m
CONFIG_PACKAGE_kmod-crypto-hmac=m
CONFIG_PACKAGE_kmod-crypto-manager=m
CONFIG_PACKAGE_kmod-crypto-null=m
CONFIG_PACKAGE_kmod-crypto-rng=m
CONFIG_PACKAGE_kmod-crypto-seqiv=m
CONFIG_PACKAGE_kmod-crypto-sha512=m
CONFIG_PACKAGE_kmod-fs-vfat=m
CONFIG_PACKAGE_kmod-hwmon-core=m
CONFIG_PACKAGE_kmod-i2c-core=m
CONFIG_PACKAGE_kmod-i2c-mux=m
CONFIG_PACKAGE_kmod-i2c-mux-pca954x=m
CONFIG_PACKAGE_kmod-leds-turris-omnia=m
CONFIG_PACKAGE_kmod-mac80211=m
CONFIG_PACKAGE_kmod-nls-base=m
CONFIG_PACKAGE_kmod-nls-cp437=m
CONFIG_PACKAGE_kmod-nls-iso8859-1=m
CONFIG_PACKAGE_kmod-nls-utf8=m
CONFIG_PACKAGE_kmod-random-core=m
CONFIG_PACKAGE_libblkid=m
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libbz2=m
CONFIG_PACKAGE_libcomerr=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_libevent2=m
CONFIG_PACKAGE_libevent2-core=m
CONFIG_PACKAGE_libevent2-pthreads=m
CONFIG_PACKAGE_libext2fs=m
CONFIG_PACKAGE_libf2fs=m
CONFIG_PACKAGE_libiwinfo=m
CONFIG_PACKAGE_libiwinfo-data=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_librt=m
CONFIG_PACKAGE_libsmartcols=m
CONFIG_PACKAGE_libss=m
CONFIG_PACKAGE_libuuid=m
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_mdadm=m
CONFIG_PACKAGE_mkf2fs=m
CONFIG_PACKAGE_partx-utils=m
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_ucode-mod-nl80211=m
CONFIG_PACKAGE_ucode-mod-rtnl=m
CONFIG_PACKAGE_ucode-mod-uloop=m
CONFIG_PACKAGE_wireless-regdb=m
CONFIG_PACKAGE_wpad-basic-mbedtls=m
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_xfrm=m
CONFIG_PACKAGE_zlib=m
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_BZIP2=y
# CONFIG_TARGET_INITRAMFS_COMPRESSION_NONE is not set
CONFIG_USE_LLVM_BUILD=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

paulwouters, 2023-12-15:
The bot seems to trigger on automation issues. I do have the right HEAD, and I prob just used a wrong format of the mvebu/cortexa9.

pktpls, 2024-03-19:
Same error on today's main

pktpls, 2024-03-19:
This seems to be a concurrency issue - all host tools build fine with `-j 1`

pktpls, 2024-03-19:
It also occurs later with normal packages. I usually build with `-j 20` and now that all host tools are present, it fails at libmnl and dropbear.

pktpls, 2024-03-20:
I've traced this back to 36150ff6ff cc @Spudz76 @Ansuel 

Reproduce with `CONFIG_SDK=y` and defconfig - the concurrency thing was just confusion.

spmadden, 2024-06-30:
@pktpls Just hit this myself on Fedora 40 same supported target.  Seems to be a LD_LIBRARY_PATH issue.  Running again with `export LD_LIBRARY_PATH=$(pwd)/staging_dir/host/lib` did the trick.

drizt, 2024-08-13:
I fix this for myself with 

``` diff
diff --git a/tools/bzip2/Makefile b/tools/bzip2/Makefile
index 0c5a92849a..d438832cdf 100644
--- a/tools/bzip2/Makefile
+++ b/tools/bzip2/Makefile
@@ -29,9 +29,9 @@ HOST_CFLAGS += $(HOST_FPIC)
 HOST_MAKE_FLAGS+= \
        CFLAGS="$(HOST_CFLAGS)" \
        LDFLAGS="$(HOST_LDFLAGS)" \
-       ENABLE_BIN_SHARED=1 \
-       ENABLE_BIN_STATIC=0 \
-       ENABLE_LIB_SHARED=1 \
+       ENABLE_BIN_SHARED=0 \
+       ENABLE_BIN_STATIC=1 \
+       ENABLE_LIB_SHARED=0 \
        ENABLE_LIB_STATIC=1 \
        ENABLE_DEV=1 \
        ENABLE_DOCS=1 \
```

OpenWrt uses own bzip2 which has libbzip2.so but binary bzip2 tries to use system version of .so file. So I change compiling bzip2 to use static libbzip2.a.

Neustradamus, 2024-11-04:
@ynezz: The @drizt fix is good?