openwrt/openwrt #14228: Platform 2.4G WiFi broken on Archer C59 and A7

kot-begemot-uk, 2023-12-15:
### Describe the bug

Onboard 2.4GHz is detected and configured. The interface is visible to ifconfig and iwinfo. Nothing is transmitted. The 2.4 GHz radio is not visible to other devices.

Tested with both normal and -ct drivers/firmware

This happens with v22.03.6, v23.05.2 and master on C59v1 and with 23.05.2 and master on A7.

### OpenWrt version

r24660-c22aa0be3e

### OpenWrt release

r24660-c22aa0be3e

### OpenWrt target/subtarget

ath79/generic

### Device

TP Link Archer A7

### Image kind

Self-built image

### Steps to reproduce

Build openwrt from v22.03.6, v23.05.2 or master for Archer A7 or C59v1. Install. Test. 2.4GHz does not work. 

### Actual behaviour

2.4GHz interface is present, detected, configured, but does not transmit.

### Expected behaviour

Well, having a working radio would be nice.

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-a7-v5=y
# CONFIG_ATH9K_HWRNG is not set
CONFIG_KERNEL_IO_URING=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_ath10k-board-qca9888=y
CONFIG_PACKAGE_ath10k-firmware-qca9888-ct=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_liblz4=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libusb-1.0=y
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-wolfssl=y
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_openvpn-openssl=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_wpad-basic-mbedtls=m
CONFIG_PACKAGE_wpad-basic-wolfssl=y
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_WPA_WOLFSSL=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kot-begemot-uk, 2023-12-15:
The same is observed on A7 and C59v1. 
The same is observed with most recent 22, 23 and master (23 and master for A7).
The same is observed with ct and normal drivers
The same...

Basically, the 2.4 GHz radio on some TP-Links is totally broken and has been broken for at least a couple of releases.

brada4, 2023-12-15:
c59 is ath79 device, likely misconfiguration,
 please post from /etc/config
. wireless
. network
and from logread while restarting particular radio.
best if using proper release from firmware-selector.

brada4, 2023-12-16:
Also worth trying to scan in Network/Wireless menu.

kot-begemot-uk, 2024-01-08:
option legacy_rates 1

There is a non-wifi 2.4GHz noise source in the area. Without it connections just die. 

With - they are kicked back to very low rate, but recover later. 

IMHO - documentation needed (it presently defaults to 0, set to 1 if 2.4GHz connections are unstable/high noise area).

brada4, 2024-01-08:
can you snap survey for dozen consecutive minutes of:
`while sleep 60 ; do iw dev phy0-ap0 survey dump | tee -a /tmp/survey.out ; done`
-90dBm is likely measurement margin
-70 should be well survivable but already limiting range
You can import it in your excel and find bad channel. (that is data that acs_survey aka channel=auto uses)
You dont show config/wireless? Maybe something overcustom there?