openwrt/openwrt #14231: 64 bit generic-squashfs-combined-efi.img.gz contains only 32bit EFI binary, doesn't boot on 64bit only system

kitor, 2023-12-16:
### Describe the bug

I have Wyse 5070 terminal (Pentium J5005). It has no legacy boot options, only UEFI.
Trying to boot generic-squashfs-combined-efi.img.gz from https://downloads.openwrt.org/releases/23.05.2/targets/x86/generic/ doesn't work, even if I add `efi\boot\bootia32.efi` as a boot option in BIOS.

However openwrt-x86-64-generic-ext4-combined-efi.img contains `\efi\boot\bootx64.efi`. After substituting that file into partition written from squashfs image variant - system boots fine intop OpenWRT

### OpenWrt version

23.05.2

### OpenWrt release

doesn't boot, can't check

### OpenWrt target/subtarget

x86_64

### Device

Dell Wyse 5070

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kitor, 2023-12-16:
Yes, this is from clean repository. I understand requirements to fill software versions - but your check fails when official pre-built image doesn't boot. How you expect reporter to fill this data in such case?

Leo-PL, 2023-12-16:
I think it is the subtarget naming, which is unclear, "64" subtarget has correct file. When 64 subtarget was added, all targets were 32-bit only, "generic" included. Perhaps it could be renamed "i686", "32" or something similar.

kitor, 2023-12-16:
Tbh it is generic as amd64 devices will run 32bit OS no issues. It is just about missing x64 boot binary - they can coexist in the same `boot/efi` directory making image really generic one.

brada4, 2023-12-16:
It is implementation specifics. Does it work if you rename/copy ia32 to x64 loader? At least dell server EFI will not load 32bit PE bootloader at all. Likely they dont want you to drop managed service by reinstalling 32bit windows on "thin client"

kitor, 2023-12-16:
No, it doesn't work when file is renamed. It boots fine when replaced with x64 counterpart from x86-64 image.

brada4, 2023-12-16:
You can change title to ... please include x64 bootloader in ia32 images...
Documentation says to use 64 if compatible.

The later ability to execute ia32 grub basically means that EFI they call BIOS is intentionally rigged to reject 32bit PE file, but later same binary type is supported to get grub loading kernel.

kitor, 2023-12-16:
Sure, updated the title.

ehem, 2024-01-06:
> I think it is the subtarget naming, which is unclear, "64" subtarget has correct file, and works. When `64` subtarget was added, all other subtargets were 32-bit only, "generic" included. Perhaps it could be renamed "i686", "32" or something similar.

I suppose one might rename it "pentium4".  Several people seem interested in reducing the number of x86 targets, in which case it seems very likely "x86/generic" will be nuked.

What seems more plausible is to rename "x86/64" to "generic64".

brada4, 2024-01-06:
wont boot on particular system. it needs EFI (PE) x64 bootloader to call PE ia32 grub.

ehem, 2024-01-06:
Looking at information to make sure.  [Pentium J5005](https://www.intel.com/content/www/us/en/products/sku/128984/intel-pentium-silver-j5005-processor-4m-cache-up-to-2-80-ghz/specifications.html) "Instruction Set", "64-bit"; which means this is amd64.  [Dell Wyse 5070](https://www.dell.com/support/manuals/en-us/wyse-5070-thin-client/5070_std_ug/memory?guid=guid-abdc8f78-ea1b-4ed3-9781-a6279f2372c0&lang=en-us), "Minimum memory": "4 GB", "Maximum memory": "8 GB".

Given the specifications, you want "64" not "generic".  (I've been complaining about "generic" being the _most specialized_ of the x86 targets)

brada4, 2024-01-06:
Yes, it is more of archaeological curiosity than a practical long term usage. Not debian nor SuSE rolling supplies x64 loader in 32bit images.

kitor, 2024-01-06:
> Given the specifications, you want "64" not "generic". (I've been complaining about "generic" being the _most specialized_ of the x86 targets)

Yes, I know, however - as you mentioned, the issue is about "generic" not being generic at all.

> Yes, it is more of archaeological curiosity than a practical long term usage. Not debian nor SuSE rolling supplies x64 loader in 32bit images.

Sure, but they name it "i386" and "amd64" not "generic" "x64" "geode" and so on. OpenWRT naming, as above - suggests generic is the one that should run everywhere, while other ones are specialized. This is highly confusing.

> What seems more plausible is to rename "x86/64" to "generic64".

I still think it would be slightly confusing as it doesn't indicate "generic" is 32 bit only.

> Several people seem interested in reducing the number of x86 targets, in which case it seems very likely "x86/generic" will be nuked.

That would obviously solve the issue too :)

brada4, 2024-01-06:
but generic runs everywhere since 1993. Rest is how you deal with bios novelties.

HunterDG, 2024-03-18:
To be clear, does a "generic" Openwrt build exist that boots 32 bit UEFI/64 bit CPU combos (like older Intel atoms)?

brada4, 2024-03-18:
You have to measure on the system. Full 32bit chain boots. 64bit subject to EFI supporting 64bit PE grub (or better if you have BIOS to boot either)