openwrt/openwrt #14232: Kernel memory leak and illegal config to disable color mode

SpectreDev1, 2023-12-16:
### Describe the bug

Theres a memory leak that seems to eat up the ram after a few it's Todo with BSS color, and if you try to disable BSS color it becomes an illegal config which hostapd will complain and it will kill off WiFi 

unreferenced object 0xffffff8016737300 (size 128):
  comm "hostapd", pid 1852, jiffies 4294963209 (age 665.710s)
  hex dump (first 32 bytes):
    01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  backtrace:
    [<000000008be796bb>] __kmem_cache_alloc_node+0x2b0/0x414
    [<00000000bdeeba9b>] kmalloc_trace+0x28/0x34
    [<00000000ece5f103>] ieee80211_obss_color_collision_notify+0x760/0x7d0 [mac80211]
    [<00000000fb169c5d>] cfg80211_vendor_cmd_reply+0x2a8/0x4924 [cfg80211]
    [<00000000ce5577a6>] genl_family_rcv_msg_doit+0xb8/0x11c
    [<000000000f9220af>] genl_rcv_msg+0x108/0x230
    [<00000000b962df06>] netlink_rcv_skb+0x5c/0x12c
    [<00000000bb8ab304>] genl_rcv+0x38/0x50
    [<0000000056412cce>] netlink_unicast+0x1e8/0x2d4
    [<000000005b10df87>] netlink_sendmsg+0x1a0/0x3d0
    [<000000009356cb3c>] ____sys_sendmsg+0x1c8/0x270
    [<00000000f6a0402e>] ___sys_sendmsg+0x7c/0xc0
    [<000000007050f128>] __sys_sendmsg+0x48/0xb0
    [<00000000fbf02fb5>] __arm64_sys_sendmsg+0x24/0x30
    [<00000000b806496c>] invoke_syscall.constprop.0+0x5c/0x104
    [<00000000a1c9480e>] do_el0_svc+0x58/0x17c

### OpenWrt version

Snapshot

### OpenWrt release

Snapshot 

### OpenWrt target/subtarget

Latest Snapshot

### Device

Linksys MX4200v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2023-12-16:
@SpectreDev1 can you provide the config for hostapd to repro?

SpectreDev1, 2023-12-16:
> @SpectreDev1 can you provide the config for hostapd to repro?
Option he_bss_

brada4, 2023-12-16:
Strange option
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n836