openwrt/openwrt #14233: Wifi stops working for some clients until they are forced to reconnect

micw, 2023-12-16:
### Describe the bug

I have several OpenWRT routers. One is configured as router, the other as dump AP. I use VLANs for multiple Wifi SSIDs. I have a bunch of clients which are always connected. Most are ESP8266 devices, connected to 2.4 GHz WiFi.

It happens quite often that a Client looses Wifi connection and is not accessible until it restarts. The client is connected to the WiFi but is not reachable.

The issue occurs on the Router as well as on the APs. I'm debugging it on the Router.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6 v3

### Image kind

Official downloaded image

### Steps to reproduce

Connect a lot if wifi clients and wait a few days ;-)

### Actual behaviour

Wifi Client gets unreachable. I am debugging a particular client with MAC 18:FE:34:CC:3E:DA.

```
#> tcpdump -i wifi-wiot-2g ether host 18:FE:34:CC:3E:DA -v -n
17:35:59.210727 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.3.1 tell 169.254.219.62, length 42
```

* The message repeats every second
* wifi-wiot-2g is the wifi device where the client is connected to
* 192.168.3.1 is the router's IP on VLAN 1003. The SSID is on this VLAN
* No ARP Answer is received on the device

```
#> tcpdump -i br-lan ether host 18:FE:34:CC:3E:DA -v -n
17:38:09.221641 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.3.1 tell 169.254.219.62, length 42
17:38:09.221976 ARP, Ethernet (len 6), IPv4 (len 4), Reply 192.168.3.1 is-at 10:27:f5:b0:65:70, length 28
```

* On the bridge interface, the ARP requests are shown along with the ARP answer

```
#> arp -n | grep -i -e "18:FE:34:CC:3E:DA"
192.168.3.162    0x1         0x0         18:fe:34:cc:3e:da     *        br-lan.1003
169.254.219.62   0x1         0x2         18:fe:34:cc:3e:da     *        br-lan.1003
```

* The device is known with 2 IP addresses because it lost it's DHCP ip since DHCP responses also don't reach the device

```
#> bridge -statistics fdb show | grep -i -e "18:FE:34:CC:3E:DA"
18:fe:34:cc:3e:da dev lan1 vlan 1003 used 1935/0 master br-lan 
18:fe:34:cc:3e:da dev lan1 vlan 1003 self 
```

This looks a bit like `https://github.com/openwrt/openwrt/issues/9555`. Hoever, I have applied several workarounds for this. Most important is `brctl setageing br-lan 3` and `multicast_to_unicast_all`. Issue remains.

Also a working client looks the same here:
```
#> bridge -statistics fdb show | grep -i -e "18:FE:34:CC:3E:DA"
18:fe:34:cc:3e:da dev lan1 vlan 1003 used 1935/0 master br-lan 
18:fe:34:cc:3e:da dev lan1 vlan 1003 self 
```

### Expected behaviour

Client should be reachable.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-16:
Please turn on hostapd logging as per https://openwrt.org/docs/guide-developer/debugging#wireless to see what happens at lower network levels.

brada4, 2023-12-17:
'not reconnect' - do they reconnect after iot ap is restarted or flips to different channel? If not you need to program microcontroller to reconnect, openwrt cannot pull them in if they stop radio.

regae, 2023-12-17:
did you use dynamic VLAN?
i was having the same issue with wpa_psk_file and freeradius if dynamic VLAN enabled.
and it only happened with mt7621/mt7615N devices (ea7500 v2, ea8100 v1, re650 v1), ath10k devices seems normal.
if using wpa_psk_file and freeradius without dynamic VLAN, no problem.
either i had to reconnect or just wait for a view minutes (sometime seconds) to gain access again.
tested on 22.03 - 23.05 - SNAPSHOTS.

maybe @nbd168 has some insight?

micw, 2023-12-17:
> Please turn on hostapd logging as per https://openwrt.org/docs/guide-developer/debugging#wireless to see what happens at lower network levels.

Did not work. I changed it and  restarted WiFi but there is no debug output. I will reboot later to see if it helps

micw, 2023-12-17:
> 'not reconnect' - do they reconnect after iot ap is restarted or flips to different channel? If not you need to program microcontroller to reconnect, openwrt cannot pull them in if they stop radio.

The clients reconnect if they are not connected. Problem is that they are connected but do not receive IP packets.
Problem also occurs on other clients, I choose the esp clients to narrow down the issue.

micw, 2023-12-17:
> did you use dynamic VLAN?
> i was having the same issue with wpa_psk_file and freeradius if dynamic VLAN enabled.
> and it only happened with mt7621/mt7615N devices (ea7500 v2, ea8100 v1, re650 v1), [...]

No dynamic vlan involved.

brada4, 2023-12-17:
> reconnect if they are not connected

do they reconnect if IoT AP is restarted?
Any custom settings in /etc/config/wireless? (just erase IP/mac/passwords)

What appears in `logread` with higher log level may give some hint on what happens to association before semi-disconnect. Obviously if nothing is felt by AP it cannot punch uC

filimonic, 2024-07-15:
Same issue for networks with roaming enabled and switch has multiple SSIDs with different VLANs (no dynamic).

Caught problem only on this wireless type

danpawlik, 2024-07-15:
@filimonic in your case, `dawn` or `usteer` will help you (of course if you configure it properly).

alexf2015, 2024-11-22:
> I have several OpenWRT routers. One is configured as router, the other as dump AP. I use VLANs for multiple Wifi SSIDs. I have > a bunch of clients which are always connected. Most are ESP8266 devices, connected to 2.4 GHz WiFi.

> It happens quite often that a Client looses Wifi connection and is not accessible until it restarts. The client is connected to the WiFi but is not reachable.

I am observing the same issue with my setup. IT only affects IoT devices, mainly Shelly switches but sometimes also my Reolink cameras. Hitting the "Disconnect" button on the AP UI fixes the problem for some time.

raburton, 2024-12-03:
> Same issue for networks with roaming enabled and switch has multiple SSIDs with different VLANs (no dynamic).

This is exactly my scenario. Two SSIDs (each replicated on 2.4 & 5ghz), on different tagged VLANs (each has an unmanaged openwrt interface), and a third untagged vlan with a static interface, used for luci/ssh (not used for wifi). Whole setup replicated on multiple APs. Works fine for a variable amount of time, then traffic stops. Usually a reconnect works, but sometimes even then it gets stuck waiting trying to obtain IP address (probably still no traffic going anywhere).

The same setup used to work fine before the switch to DSA, so I assume it's related to that. Periodically I will try a new release, just been through this again over the last week, and always end up going back to v19.07 before the move to DSA. Would love to move to a newer release, but it just can't handle this semi-complex setup anymore.

bubibubi, 2025-01-30:
Same issue here.

**OpenWRT 23.05 on mt7628an (TP-Link TL-WR841N v13) configured as Access Point**
1 lan (no vlan) connected to lan port
1 wireless with 1 SSID protected with WPA2 PSK
1 ESP8266 device connected to wireless

**After a while ESP8266 "loose" connection**
I still see the device in OpenWRT Associated stations but I can't reach the device
To reconnect the device I need to restart OpenWRT or restart the ESP8266

**Workaround**
Install 19.07.10

brada4, 2025-01-30:
So it is some modern capability introduced between 19 and 21, most likely in hostap?
can you compare generated hostapd.conf in /var/run/ between them?

raburton, 2025-01-31:
> > That PR is from 2024, but this problem* has been been in OpenWrt since 2021.
> 
> This fix is for the old bug.

Right, sorry, I had that backwards! The problem isn't limited to MT76 though - there are a couple of different devices using ath9k mentioned in the thread.

brada4, 2025-12-11:
Did you save v21 generated hostapd.conf ? Unless you figure out specific option causing interop problems with specific client it is hard to guess what is missing.

melektron, 2026-09-05:
I have a similar issue with OpenWrt 24.10.5 on lantiq/xrx200 (AVM FRITZ!Box 7360 V2), with Atheros AR9287-BL1A wifi hardware (ath9k).

In my case when a client connects (e.g. an iPhone) as soon as it produces a significant amount of traffic, it enters this pseudo "disconnected" state. Still shows up as associated, the client device also still "thinks" it's associated. I get arp frames just as described above, but no further traffic seems to pass through, not even a ping to the ap+router in question.

Interestingly it does not affect all devices, some just show very poor performance, and other's can't connect at all, though that might be a client side issue.

tapper82, 2026-09-06:
Is this bug still on 25.xx OpenWrt? There is no point of looking at a bug if it has bin fixt in a new build of OpenWrt! 23 and 24 builds of openwrt are end of life now.

github-k8n, 2026-09-06:
> Is this bug still on 25.xx OpenWrt? There is no point of looking at a bug if it has bin fixt in a new build of OpenWrt! 23 and 24 builds of openwrt are end of life now.

Yes, the issue still occurs. 2 openwrt devices, one as WiFi client, one as AP. For the  TP-Link Archer C7 v2 ( the client), it regularly lost connection while still shown as connected on the FritzBox Repeater 3000 (the AP). I now switched the client to another device and that seems stable so far (few days now), so most likely related to the chipset.

github-k8n, 2026-09-09:
So i had the issue occur again at first even with the new device.
I now did make sure that I have exactly the same Channel settings on the client device for the wifi client wireless network as I have on my AP (Channel 100 , 80MHz ..)  (I had it on "auto" on the wireless client previously)
I additionally checked and added the BSSID on the wifi client to make sure I am connecting to this specific AP with this specific Radio only (there are three radios on the AP with two of them in the 5GHz range).
For now it seems to be stable, but given the random nature of this issue, only time will tell...