openwrt/openwrt #14236: Copy paste bug in `realtek/files-5.15/drivers/net/phy/rtl83xx-phy.c`

qarmin, 2023-12-17:
### Describe the bug

https://github.com/openwrt/openwrt/blob/8815a3114ffb788ca037e634042811a9fed1562d/target/linux/realtek/files-5.15/drivers/net/phy/rtl83xx-phy.c#L2010-L2011

```
dcvs_coef_bin
```
is set twice

Problably this should look like 

https://github.com/openwrt/openwrt/blob/8815a3114ffb788ca037e634042811a9fed1562d/target/linux/realtek/files-5.15/drivers/net/phy/rtl83xx-phy.c#L2000-L2001

### OpenWrt version

Master git branch

### OpenWrt release

Master git branch

### OpenWrt target/subtarget

N/A

### Device

N/A

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-12-24:
Indeed seems wrong; @svanheule?

svanheule, 2023-12-25:
Seems fishy in multiple ways. There's the copy-paste error, the fact that `rtl9300_sds_rxcal_dcvs_get()` doesn't line up with `rtl9300_sds_rxcal_dcvs_manual()`, and `dcvs_manual` isn't assigned consistently (although that shouldn't matter in this case). All that and there are no calls to be found to `rtl9300_sds_rxca_dcvs_{get,manual}()`, making them dead code...