openwrt/openwrt #14238: mt7921u: Access point doesn't start in 6 GHz band

csharper2005, 2023-12-17:
### Describe the bug

SNAPSHOT:
1. Comfast CF-953AX WiFi 6 USB adapter (mt7921au) doesn't start in AP mode (6 GHz band). `Frequency 6075 (primary) not allowed for AP mode, flags: 0x10001`
2. No 6 GHz band in Luci. 

2.4 & 5 GHz work fine.

OpenWrt 23.05.2:
1. AP (6 GHz) starts
2. 6 GHz band in Luci is in place.

### OpenWrt version

r24665-8815a3114f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Beeline SmartBox GIGA

### Image kind

Official downloaded image

### Steps to reproduce

Configure 6g band:
```
config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/1e1c0000.xhci/usb2/2-1/2-1:1.3'
	option channel '25'
	option band '6g'
	option htmode 'HE80'
	option country 'RU'
	option cell_density '0'

config wifi-iface 'default_radio2'
	option device 'radio2'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt-6g'
	option encryption 'sae'
	option key '***********************'

```

### Actual behaviour

AP doesn't start:
```
Sun Dec 17 19:38:09 2023 daemon.notice hostapd: Set new config for phy phy2:
Sun Dec 17 19:38:10 2023 daemon.notice hostapd: Set new config for phy phy2:
Sun Dec 17 19:38:10 2023 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Sun Dec 17 19:38:10 2023 daemon.notice hostapd: Restart interface for phy phy2
Sun Dec 17 19:38:10 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=RU ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=25 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] op_class=133 ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=23 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=23 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_mu_edca_ac_be_aci=0 he_mu_edca_ac_be_ecwmin=9 he_mu_edca_ac_be_ecwmax=10 he_
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: phy2-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: Frequency 6075 (primary) not allowed for AP mode, flags: 0x10001
Sun Dec 17 19:38:11 2023 daemon.err hostapd: Primary frequency not allowed
Sun Dec 17 19:38:11 2023 daemon.warn hostapd: phy2-ap0: IEEE 802.11 Configured channel (25) or frequency (6075) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
Sun Dec 17 19:38:11 2023 daemon.warn hostapd: phy2-ap0: IEEE 802.11 Hardware does not support configured channel
Sun Dec 17 19:38:11 2023 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: phy2-ap0: interface state COUNTRY_UPDATE->DISABLED
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: phy2-ap0: AP-DISABLED
Sun Dec 17 19:38:11 2023 daemon.err hostapd: phy2-ap0: Unable to setup interface.
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: nl80211: deinit ifname=phy2-ap0 disabled_11b_rates=0
Sun Dec 17 19:38:11 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: phy2-ap0: CTRL-EVENT-TERMINATING
Sun Dec 17 19:38:11 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy2-ap0 wasn't started
Sun Dec 17 19:38:11 2023 daemon.notice hostapd: hostapd.add_iface failed for phy phy2 ifname=phy2-ap0
```

### Expected behaviour

AP starts. Example from OpenWrt 23.05.2:
```
Sun Dec 17 19:50:07 2023 daemon.notice hostapd: Set new config for phy phy2:
Sun Dec 17 19:50:08 2023 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Sun Dec 17 19:50:08 2023 daemon.notice hostapd: Restart interface for phy phy2
Sun Dec 17 19:50:08 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=RU ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=25 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] op_class=133 ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=23 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=23 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_mu_edca_ac_be_aci=0 he_mu_edca_ac_be_ecwmin=9 he_mu_edca_ac_be_ecwmax=10 he_
Sun Dec 17 19:50:09 2023 daemon.notice hostapd: phy2-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Dec 17 19:50:10 2023 daemon.notice hostapd: phy2-ap0: interface state COUNTRY_UPDATE->ENABLED
Sun Dec 17 19:50:10 2023 daemon.notice hostapd: phy2-ap0: AP-ENABLED
```

### Additional info

```
iw reg get
global
country RU: DFS-UNSET
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-17:
Set auto channel? Base channels are 17 and 31 for HE80.

DragonBluep, 2023-12-18:
It doesn't support 6 GHz.
http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=182
[image]

csharper2005, 2023-12-18:
> Set auto channel? Base channels are 17 and 31 for HE80.

Behavior doesn't change (snapshot and 23.05.2 both). 

> It doesn't support 6 GHz. 

Officially. Or in practice too?

brada4, 2023-12-18:
You can check `iw list` to see which channels are permitted across all bands.
Post one from 23.05.2 and snapshot.

6GHz supporting chip is 7921**a**u, rather strange you were able to select unannounced band from generation older adapter via LuCI. Though there have been reports on some chips you can select unsupported 160MHz channel and get the bad adapter response. Or LuCI will visualise anything written in conf file, that disappears once you select something supported.

DragonBluep, 2023-12-18:
It's hardware limit. The MT7921 family only support 5 GHz + BW80. MT7922 series support wave2 features.
https://www.mediatek.com/products/broadband-wifi/mediatek-filogic-320

csharper2005, 2023-12-18:
> It's hardware limit. The MT7921 family only support 5 GHz + BW80. MT7922 series support wave2 features. https://www.mediatek.com/products/broadband-wifi/mediatek-filogic-320

Hmm... Why Netgear A8000 based on mt7921 supports WiFi 6e? 
https://www.netgear.com/de/home/wifi/adapters/a8000/
https://www.spinics.net/lists/linux-wireless/msg235747.html

brada4, 2023-12-18:
It is ....21AU chipset....

csharper2005, 2023-12-18:
`iw list` from 23.05.2. There is no 6 GHz in SNAPSHOT.
```
Wiphy phy2
        wiphy index: 2
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce120dc0b306423f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
                                Ng = 16 SU Feedback
                                Ng = 16 MU Feedback
                                Codebook Size SU Feedback
                                Codebook Size MU Feedback
                                Triggered CQI Feedback
                                Partial Bandwidth Extended Range
                                PPE Threshold Present
                                Power Boost Factor ar
                                HE SU PPDU & HE PPDU 4x HE-LTF 0.8us GI
                                20MHz in 40MHz HE PPDU 2.4GHz
                                DCM Max BW: 1
[... 470 lines cut ...]
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: 65535
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)

```

csharper2005, 2023-12-18:
> It is ....21AU chipset....

Yes, this is written about in the 1st message.

> This patch need to be imported [torvalds/linux@7a934b5](https://github.com/torvalds/linux/commit/7a934b5cc3f452df6f9a4903450fc103dee98ee8) I can comfim that fixed the issue on Comfast CF-951AX.

Thanks!

antnyfls, 2023-12-18:
On the chip of the CF-951AX is printed **MT7921AUN**

[image]

csharper2005, 2023-12-18:
6 GHz in windows.
[image]

brada4, 2023-12-18:
Thats like hundred bux grew in your hand? Fair enough, chips sare same size and assembly line stamped new ones, just did not tell web editor to update web page.

csharper2005, 2023-12-19:
@antnyfls probably this is a fix: https://github.com/openwrt/openwrt/commit/1a57d758f229a1e8d9085cd4775ebe7e1ec5d9b2

csharper2005, 2023-12-19:
> @csharper2005 Yes that is the fix [openwrt/mt76@bebd9cf](https://github.com/openwrt/mt76/commit/bebd9cffc2aeb2cecb40aadbb8c6eab3bdf7971b)

I tested `OpenWrt SNAPSHOT, r24685`. 6 GHz band works. So, I close the issue.
[image]

closed 2023-12-19 by csharper2005