openwrt/openwrt #14253: DFS start_dfs_cac() failed - wifi not enabled after 30 minute delay.

piojan, 2023-12-20:
### Describe the bug

Problem 1: 30 minutes after detecting interference on an DFS channel the wifi is not enabling the channel back. A wifi restart is needed.
Problem 2: There is no option in Luci to configure a fallback channel to switch too in case of interference.

```
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: phy1-ap0: DFS-RADAR-DETECTED freq=5300 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finish
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sat Dec 16 13:30:39 2023 kern.info kernel: [1414731.215870] device phy1-ap0 left promiscuous mode
Sat Dec 16 13:30:39 2023 kern.info kernel: [1414731.225950] br-lan: port 5(phy1-ap0) entered disabled state
Sat Dec 16 13:30:39 2023 daemon.notice netifd: Network device 'phy1-ap0' link is down
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Sat Dec 16 13:30:39 2023 kern.info kernel: [1414731.356999] br-lan: port 5(phy1-ap0) entered blocking state
Sat Dec 16 13:30:39 2023 kern.info kernel: [1414731.368626] br-lan: port 5(phy1-ap0) entered disabled state
Sat Dec 16 13:30:39 2023 kern.info kernel: [1414731.380796] device phy1-ap0 entered promiscuous mode
Sat Dec 16 13:30:39 2023 daemon.warn hostapd: nl80211: Failed to add interface phy1-ap0 into bridge br-lan: Resource busy
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: phy1-ap0: interface state DISABLED->COUNTRY_UPDATE
Sat Dec 16 13:30:39 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Sat Dec 16 13:30:40 2023 daemon.err hostapd: could not get valid channel
Sat Dec 16 13:30:40 2023 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5260 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5260 cf2=0
Sat Dec 16 14:01:03 2023 daemon.err hostapd: could not get valid channel
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5280 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5280 cf2=0
Sat Dec 16 14:01:03 2023 daemon.err hostapd: could not get valid channel
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5300 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5300 cf2=0
Sat Dec 16 14:01:03 2023 daemon.err hostapd: could not get valid channel
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5320 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5320 cf2=0
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Sat Dec 16 14:01:03 2023 daemon.notice hostapd: phy1-ap0: DFS-CAC-START freq=5300 chan=60 sec_chan=1, width=0, seg0=58, seg1=0, cac_time=60s
Sat Dec 16 14:01:03 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5310) not in sync
Sat Dec 16 14:01:03 2023 daemon.err hostapd: Can't set freq params
Sat Dec 16 14:01:03 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear R6220

### Image kind

Official downloaded image

### Expected behaviour

1. bringing back up the channel after 30 min - this is not working - wifi restart required
2. automatic transition to a non-DFS channel, as an option in Luci or by default.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-20:
ETSI standard kind of implies that you move your transmissions out of channel and care not about returning for 30 minutes.
Set channel to auto and avoid low rank channels:
`list hostapd_options 'min_tx_power=20'`
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n240
Directly setting it as "option" does not make an error, not conveys to `/var/run/hostapd*conf`

brada4, 2023-12-20:
The "center" messages sort of describe legit 40MHz configuration that should work given AUTO-BW regulatory statement.
Maybe it even does if you set base channel 52....

lukasz1992, 2023-12-21:
What is the device? Maybe it just does not support background scan?

brada4, 2023-12-21:
it is fg scan yielding auto-bw valid upper half of 80mhz channel.

brada4, 2023-12-21:
it is mistake in validation in hostapd.c that expects center frequency of original wide channel to be contained in  half-channels, essentially defeating auto-bw-s purpose by denying all but one part-channel coming out of auto-bw.

piojan, 2024-08-23:
I can still see this issue on: 23.05.4 
Device R6220
Any thoughts how this validation in hostapd.c could be corrected?

`Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: DFS-RADAR-DETECTED freq=5300 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finish
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Tue Aug 20 15:55:51 2024 kern.info kernel: [1135362.387572] device phy1-ap0 left promiscuous mode
Tue Aug 20 15:55:51 2024 kern.info kernel: [1135362.397780] br-lan: port 5(phy1-ap0) entered disabled state
Tue Aug 20 15:55:51 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Tue Aug 20 15:55:51 2024 kern.info kernel: [1135362.549432] br-lan: port 5(phy1-ap0) entered blocking state
Tue Aug 20 15:55:51 2024 kern.info kernel: [1135362.560999] br-lan: port 5(phy1-ap0) entered disabled state
Tue Aug 20 15:55:51 2024 kern.info kernel: [1135362.573124] device phy1-ap0 entered promiscuous mode
Tue Aug 20 15:55:51 2024 daemon.warn hostapd: nl80211: Failed to add interface phy1-ap0 into bridge br-lan: Resource busy
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: interface state DISABLED->COUNTRY_UPDATE
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Tue Aug 20 15:55:51 2024 daemon.err hostapd: could not get valid channel
Tue Aug 20 15:55:51 2024 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5260 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5260 cf2=0
Tue Aug 20 16:25:57 2024 daemon.err hostapd: could not get valid channel
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5280 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5280 cf2=0
Tue Aug 20 16:25:57 2024 daemon.err hostapd: could not get valid channel
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5300 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5300 cf2=0
Tue Aug 20 16:25:57 2024 daemon.err hostapd: could not get valid channel
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: DFS-NOP-FINISHED freq=5320 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5320 cf2=0
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: interface state DFS->DFS
Tue Aug 20 16:25:57 2024 daemon.notice hostapd: phy1-ap0: DFS-CAC-START freq=5300 chan=60 sec_chan=1, width=0, seg0=58, seg1=0, cac_time=60s
Tue Aug 20 16:25:57 2024 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5310) not in sync
Tue Aug 20 16:25:57 2024 daemon.err hostapd: Can't set freq params
Tue Aug 20 16:25:57 2024 daemon.err hostapd: DFS start_dfs_cac() failed, -1
`

brada4, 2024-08-23:
Set `option channels 'desired channel' option channel 'auto'` to permit auto-bw reduce fixed channel width. Obviously dont jump off the ship when it works but confirm.