openwrt/openwrt #14257: JCG Q30 PRO is a variant of JCG Q30

cherinyy, 2023-12-21:
### Describe the bug

`JCG Q30 PRO` is a variant of `JCG Q30`, and only `JCG Q30 PRO` was added to OpenWrt.

https://github.com/openwrt/openwrt/blob/4a2ff73177992abcb30f164358b3692ac9fbdc52/target/linux/mediatek/image/filogic.mk#L419-L441

It's better to have `Q30` as the main version and `Q30 PRO` as the alternative version.

Example: https://github.com/immortalwrt/immortalwrt/blob/d60186f9a9d5e61a27e3665a157e18a3aa34a37c/target/linux/mediatek/image/filogic.mk#L606-L636

### OpenWrt version

r24683-4cc6e7192f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

JCG Q30 PRO

### Image kind

Official downloaded image

### Steps to reproduce

N/A

### Actual behaviour

N/A

### Expected behaviour

N/A

### Additional info

- https://github.com/openwrt/openwrt/pull/13440

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aiamadeus, 2023-12-21:
This is incorrect, PRO is the name not the variant.
DEVICE_VARIANT is usually Version 1.0, 2.0, etc.
BTW I have never seen one without pro on the label.
If it is exactly the same hardware, you should use DEVICE_ALT0_xxx.
If most of the hardware is same but only a few parts are different,
you should add a new device and share the device tree.

cherinyy, 2023-12-22:
I don't have a non-PRO device either, just pointing out the situation.

aiamadeus, 2023-12-31:
So, you are reporting an issue from an unsupported fork.
It is not a bug that new devices are not supported.

closed 2023-12-31 by cherinyy